moveit/moveit #1351: universal_robot package expects missing KDL plugins (gone in melodic)

jwhendy, 2019-02-17:
The universal_robot package tries to include to moveit_kinematics libraries that don't seem to exist on melodic.

The [two lines](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_kinematics/include/ur_kinematics/ur_moveit_plugin.h#L97):
```
#include <moveit/kdl_kinematics_plugin/chainiksolver_pos_nr_jl_mimic.hpp>
#include <moveit/kdl_kinematics_plugin/chainiksolver_vel_pinv_mimic.hpp>
```

These [existed](https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin) in kinetic, but not [anymore](https://github.com/ros-planning/moveit/tree/melodic-devel/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin).

I was able to successfully build by just commenting out those lines. Is this sufficient, or is there a new import that these were rolled into? I did a brief test on a previous package and I'm able to bringup a UR10. This is [cross-posted](https://github.com/ros-industrial/universal_robot/issues/403). If this is about commenting out these libraries or including something else, I'm happy to do a PR over there to get this resolved.

Thanks!

rhaschke, 2019-02-17:
These headers should not be public API and thus they were removed in Melodic.
As you pointed out, UR kinematics also builds without these includes. Please file a PR on their site.

closed 2019-02-17 by rhaschke

jwhendy, 2019-02-17:
[Done](https://github.com/ros-industrial/universal_robot/pull/404). Thanks for the quick clarification.