moveit/moveit #1355: Link is not known to URDF. Cannot disable collisons.

kctoayo88, 2019-02-18:
Hello,
I use Ubuntu 16.04 with ROS Kinetic. 
I created a xacro file, then convert it to urdf file of my robot. 
And I load it on the MoveIt! Setup Assistant, it works without any problem. 
But, when I open the demo.launch of moveit.
It shows these error message:
[ERROR] : Virtual joint does not attach to a link on the robot (link 'world' is not known).
[ WARN] : Link 'table' is not known to URDF. Cannot disable collisons.
[ WARN] : Link 'eef_base_link' is not known to URDF. Cannot disable collisons.
[ WARN] : Link 'lwheel_link' is not known to URDF. Cannot disable collisons.
[ WARN] : Link 'rwheel_link' is not known to URDF. Cannot disable collisons.

Here is my xacro file:
<robot name="motoman_mh5" xmlns:xacro="http://ros.org/wiki/xacro">

<xacro:include filename="$(find motoman_mh5_support)/urdf/mh5_macro.xacro" />
<xacro:motoman_mh5 prefix=""/>

<xacro:include filename="$(find eef)/urdf/eef.urdf" />

<xacro:include filename="/home/cyc/catkin_ws/src/motoman/motoman_mh5_support/urdf/env_macro.xacro" />
<xacro:env prefix=""/>
<link name="world" />
<joint name="world_table_joint" type="fixed">
	<parent link="world" />
	<child link = "table" />
	<origin xyz="0.0 0.0 0.4" rpy="0.0 0.0 0.0" />
</joint>
<joint name="table_robot_joint" type="fixed">
	<parent link="table" />
	<child link = "base_link" />
	<origin xyz="-0.6 0.0 0.4" rpy="0.0 0.0 0.0" />
</joint>
<joint name="robot_eef_joint" type="continuous">
	<parent link="link_t" />
	<child link = "eef_base_link" />
	<origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0" />
</joint>
</robot>

If you could help me solve it, it would be nice to me.
Thanks.

felixvd, 2019-02-18:
Your SRDF file tries to disable collisions for links that are not defined in your URDF file. Maybe you ran the Self-Collisions part of the setup assistant with an older version of the URDF file.

Try regenerating the self-collision matrix via the Self-Collisions tab in the setup assistant and see if the error persists.

kctoayo88, 2019-02-18:
Thank @felixvd for your reply.
I had tried regenerating the self-collision matrix three times,but the error is still existed.
Does it have any other solution to use? 

PS. my eef.urdf is generated by SW 2016

felixvd, 2019-02-18:
Sounds strange, but that would probably have only affected the warnings anyway, not the error.

The link might need to be defined before the macros. Try moving the `<link name='world' />` line up to line 3.

If that is not it, it might be something in the macro files. Please post them as code, not as screenshots.

kctoayo88, 2019-02-18:
Sorry for bad news, and here is my xacro code:

<?xml version="1.0" ?>
<robot name="motoman_mh5" xmlns:xacro="http://ros.org/wiki/xacro">

<link name="world" />

<xacro:include filename="$(find motoman_mh5_support)/urdf/mh5_macro.xacro" />
<xacro:motoman_mh5 prefix=""/>
<xacro:include filename="$(find eef)/urdf/eef.urdf" />
<xacro:include filename="/home/cyc/catkin_ws/src/motoman/motoman_mh5_support/urdf/env_macro.xacro" />
<xacro:env prefix=""/>

<joint name="world_table_joint" type="fixed">
	<parent link="world" />
	<child link = "table" />
	<origin xyz="0.0 0.0 0.4" rpy="0.0 0.0 0.0" />
</joint>
<joint name="table_robot_joint" type="fixed">
	<parent link="table" />
	<child link = "base_link" />
	<origin xyz="-0.6 0.0 0.4" rpy="0.0 0.0 0.0" />
</joint>
<joint name="robot_eef_joint" type="continuous">
	<parent link="link_t" />
	<child link = "eef_base_link" />
	<origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0" />
</joint>

</robot>

Please check it for me,
thank for your help.

kctoayo88, 2019-02-19:
I solve it after modifying the xacro file content and adjusting the path of it.
thanks

closed 2019-02-19 by kctoayo88

felixvd, 2019-02-19:
1) Please post what you changed in your file so that readers can try to solve their problem too. 

2) Questions like this are generally better suited to [answers.ros.org](http://answers.ros.org), just for the future.

SunYuansong, 2019-06-21:
Hello,
I meet the same problem when I want to add the environment model to ur5 robot. The demo.launch works well, but it goes wrong when connecting to the real robot.
The error part says:
`[ERROR] [1561129695.153342082]: Semantic description is not specified for the same robot as the URDF
[ WARN] [1561129695.153489929]: Link 'table1' is not known to URDF. Cannot disable collisons.
[ WARN] [1561129695.153503756]: Link 'table2' is not known to URDF. Cannot disable collisons.
[ WARN] [1561129695.153512933]: Link 'table1' is not known to URDF. Cannot disable collisons.
[ WARN] [1561129695.153518336]: Link 'table2' is not known to URDF. Cannot disable collisons.
[ WARN] [1561129695.153549509]: Link 'table1' is not known to URDF. Cannot disable collisons.`
@kctoayo88 Could you please post your solution?

brenhertel, 2019-12-24:
Hello, 
I am running into the same issue as mentioned here. If anybody has found a solution, could they please post it?
Thanks.

felixvd, 2020-02-29:
Sanity check since it came up in [another ROS Answers post](https://answers.ros.org/question/342488/link-is-not-known-to-urdf-cannot-disable-collisons/): If you are running into this problem, is your robot description published? `demo.launch` publishes it by default, but `move_group.launch` does not. Try setting `load_robot_description` to `true` in `planning_context.launch`.

Spandre, 2020-05-06:
I have a similar problem. Basically i have added the camera description together with the robot macros to obtain the model of the robot and the camera, that can be simulated on Gazebo (the camera is the intel realsense D435i and the only solution I have found was to add its description to the robot urdf file). When starting the simulation the robot is loaded with its namespace and then, when I try to see the configuration in RViz, i notice the presence of this error togher with a not existing frame (I think this problem is due to the presence of the namespace that is added to the robot links: i.e. edo_base_link becomes /edo/edo_base_link, meanwhile the cam has as parent link only the edo_base_link RF that does not exist. I'm not really shure about this). If anyone has some hints I would appreciate that. If is necessary to have some code answer to me and I will post it as soon as possible. Thanks