moveit/moveit #1356: URGENT: Messed up melodic-devel branch.

rhaschke, 2019-02-19:
Dear @davetcoleman and @v4hn,
cc: @vmayoral

looks like yesterday evening I accidentally pushed my local working HEAD to the ros-planning remote instead of my own fork, thus unintentionally merging several open PRs. I apologize for this mistake.

I see the following options to fix the problem:
* Reset melodic-devel to its last valid commit: 4782c5a5bf237006ffc23a198bea042e0fbb759f (cleanest solution).
* Accept the changes and apply necessary fixes on top of them. [Currently catkin_lint check is broken](https://travis-ci.org/ros-planning/moveit/builds/495002198) due to a pending release (https://github.com/ros-planning/moveit/pull/1343#issue-252455694).

This also requires a similar handling of the MoveIt2 fork **before** any other PRs are merged.

vmayoral, 2019-02-19:
Thanks for reporting @rhaschke. 

Second option will be easier since we've gone a long way at https://github.com/AcutronicRobotics/moveit2/commits/master. I guess I can interactively rebase it and remove the commits as well. Quickly checked and doesn't seem to conflict with anything we're porting right now (still very early stage).

jonbinney, 2019-02-19:
+1 for accepting the changes and applying fixes on top. It doesn't look like there are any huge changes since `4782c5a`.

davetcoleman, 2019-02-19:
Here's the diff of what was accidentally pushed in:

https://github.com/ros-planning/moveit/compare/4782c5a5bf237006ffc23a198bea042e0fbb759f...melodic-devel

+1 to accepting changes and applying fixes on top. Could you explain what the xmlrpc_casts changes are for?

davetcoleman, 2019-02-19:
This brings up a concern I have about accidentally pushing to the wrong remote also. I think it would be a good idea to disable the ability to push to the branches, instead requiring PRs for everything (including cherry-picks). From my experience, this is what mature software projects do.

rhaschke, 2019-02-19:
Looks like everyone votes for keeping the merge. I will then revert #1343 as it causes Travis to fail.
Any objections?

jonbinney, 2019-02-19:
Sounds good to me

closed 2019-02-20 by rhaschke

rhaschke, 2019-02-20:
> Could you explain what the `xmlrpc_casts` changes are for?

These are XmlRpc parsing utilities introduced a while ago (#1253) in the context of `kinematic_constraints` parsing from param server. I just moved them into the more general `libmoveit_utils.so` for reuse in other contexts.

130s, 2019-02-20:
> This brings up a concern I've had about any of us accidentally pushing to the wrong remote. I think it would be a good idea to disable the ability to push to the branches, instead requiring PRs for everything from forks (including cherry-picks). From my experience, this is what mature software projects do.

+1

As documented in ["No Direct Commits" section on moveit.ros.org](https://moveit.ros.org/documentation/contributing/pullrequests/), some release commits have been directly committed so far. But I was always uncomfortable in doing so. I'm fine in making PRs for those release works, although I'm afraid no one would want to review dry changes like those. Can we setup a special merge rule for release PRs?

v4hn, 2019-02-20:
You can make it harder to push to the wrong remote by setting your own fork to `origin`, but even that can fail at times.
I'm not a fan of forcing developers to use the web front-end to merge changes, because I consider the web gui as an inferior replacement to arbitrary GUI and CLI tools that handle git repositories.
But if everyone else considers this necessary, I do not block it, as long as nobody disables specific merge-types again.

> ... `kinematic_constraints` parsing from param server. I just moved them into the more general `libmoveit_utils.so` for reuse in other contexts.

Thank you! When I started the implementation the utils package did not exist yet and it bugged me that this code was part of the `kinematics_constraint` package.

Apart from the premature catkin_lint commit that you reverted, the other changes look ok to stay upstream.

I am somewhat unhappy with 749846f3f46b3b5a2436e47a3c10d2e74d6df535.
It does indeed provide functionality that was not available before, but I would argue that static values only resolve half of the actual problem, if anything at all.
The patch, as is, allows to specify fixed values for the tolerances, so that motor controllers that read the fields actually receive fully-specified goals.
But this functionality already exists at least in [`ros_control` based controllers](http://wiki.ros.org/joint_trajectory_controller?distro=melodic#Parameters) too, so the specification in MoveIt adds a redundant mechanism for those.

From my perspective, specifying the values in the action goal only makes sense if they can vary per request. I believe this is actually the case in your use-case, closing the gripper to grasp versus opening the gripper to release. But then we currently do not have interfaces to specify the values at any higher level of abstraction.

rhaschke, 2019-02-20:
> I'm not a fan of forcing developers to use the web front-end to merge changes, because I consider the web gui as an inferior replacement to arbitrary GUI and CLI tools that handle git repositories.

I strongly support this. I frequently used fast-forwarding for several cherry-picks I made and tested locally. Requiring to file PRs for this slows down the process. 

> I am somewhat unhappy with 749846f.

You are right @v4hn, this feature is not useful yet. It's a first step along the road to configure execution parameters per query. One blocker I found, is that MoveControllerManagers don't yet generally have a concept of tolerances during execution...

davetcoleman, 2019-02-20:
> I'm not a fan of forcing developers to use the web front-end to merge changes

But that is exactly out policy, except in the corner case of cherry-picking between branches after a PR is merged or in the accidental pushing of code that's not ready. This to me is a clear flaw in our quality control approach