moveit/moveit #1357: Planning groups can only contain chains

christian-rauch, 2019-02-20:
### Description

MoveIt! now expects that a planning group only contains joints that are in a chain, i.e. it will print an error like `Group 'all' is not a chain` if the group `all` is a superset of a chain.
I generated a MoveIt! configuration using the setup assistant with ROS melodic and this used to work before with version 0.9.

### Your environment
* ROS Distro: melodic
* OS Version: e.g. Ubuntu 18.04
* Version: binary, 0.10.8

### Steps to reproduce
1. generate a MoveIt! configuration with a planning group that contains all joints
2. run MoveIt! version 0.10.8

### Expected/Actual behaviour
Previously, MoveIt! (version 0.9) was loading the planning group and created a chain from root-frame to the common parent frame of the leaf-frames. This changed and I am not able to use this group.

When did this behaviour change and how can I revert to the old behaviour?

rhaschke, 2019-02-20:
Could you please report the exact error message. I'm using a bimanual robot composed from several sub groups and this definitely still works.
Looks like your error message comes from a IK solver. Which one are you using?

christian-rauch, 2019-02-20:
The full message, e.g. when running `demo.launch`,  is:
```
[ERROR] [1550688524.963909221]: Group 'all' is not a chain
[ERROR] [1550688524.963939963]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'all'
[ERROR] [1550688524.963969462]: Kinematics solver could not be instantiated for joint group all.
```

rhaschke, 2019-02-20:
That's what I assumed: The _kinematics solver_ (KDL) expects your group to be a chain.
It was like this (implicitly) since ages, but I made the check more explicit now.
Please verity that the IK solver also moved "non-group" joints before.

christian-rauch, 2019-02-20:
Ok, good to know.
I will just add another planning group for this purpose and figure out how to load it as the default planning group.

closed 2019-02-20 by christian-rauch