moveit/moveit #1359: Planning Failed With Object Attached

ssnover95x, 2019-02-22:
### Description

Hello, I'm currently setting up movement for a robot I'm working on using MoveIt. I'm trying to enable some functions which use the move group interface in C++ to plan a trajectory to a joint space goal and then execute that goal asynchronously. I've managed to get the trajectory to plan and execute without problems in approaching an object that's in the planning scene. However once I've attached that object I can no longer successfully plan trajectories. I'm curious if I'm not attaching the object correctly.

The failure that I'm seeing is as follows: `Fail: ABORTED: No motion plan found. No execution attempted.` This suggests that the plan function is completing, but that the trajectory in the returned output parameter is empty. When it's passed to `asyncExecute` it returns this message.

I'm attaching the object as follows:
```
this->mMoveGroup.attachObject(<string to manipulated object>, <string to last link in chain>, <vector of a couple strings of links which are nearby and not a problem for the object to touch>);
```

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary: amd64/bionic 0.10.8-0bionic.20190124.144626

### Expected behaviour
The trajectory should be executed.

### Actual behaviour
Error printed: `Fail: ABORTED: No motion plan found. No execution attempted.`

I suspect this isn't a bug, just an improper understanding of how this is supposed to be done. The error message is not very helpful as it's not clear what the problem is. I don't believe there are collisions as I've seen warning messages about collisions before. The trajectory to be computed is relatively straightforward compared with the ones that are working. The only thing that seems to be complicating the situation is the attached object.

felixvd, 2019-02-25:
> I suspect this isn't a bug, just an improper understanding of how this is supposed to be done.
When you suspect this, using [ROS Answers](answers.ros.org) is usually better.

You might be misunderstanding this error message: `Fail: ABORTED: No motion plan found. No execution attempted.` implies that no suitable trajectory was found by the plan function.

Maybe these thoughts can help you fix your problem:
- Is the collision model of your object set up properly?
- Is the object touching other objects in the scene? (e.g. the table the object was standing on)
- Is the object that you attached touching more robot links than the one it is attached to? Maybe your gripper links protrude into the object in the scene? Allow the necessary collisions in the collision matrix (e.g. by adding them to the vector of links the object can touch)

> I don't believe there are collisions as I've seen warning messages about collisions before.

Have you checked if any collisions are displayed in Rviz? In the logs, too? It's certainly the most likely.

ssnover95x, 2019-02-28:
Mainly I was leaning towards posting it here since the error that shows in the console is not indicative of the problem. I believe the message is printed while the `asyncExecute` function is called because the passed trajectory is empty, though I may be wrong as it's difficult to trace where it came from. This would imply that the `plan` function did not generate a failure in its return (I compare the return value with the success value) despite not filling the trajectory output parameter with any meaningful value.

However, I'm not sure what happened but someone else on my team built the code and this issue didn't pop up for them. When I went rebuilt from a completely clean workspace I didn't see the issue again either. So I'll close this since it's not apparent what the root cause was or how I can recreate it.

closed 2019-02-28 by ssnover95x