moveit/moveit #1364: Contribute to moveit! through GSoC 2019

thebhatman, 2019-02-26:
Hello All,
I am Manjunath Bhat, a second-year undergraduate from the Indian Institute of Technology(IIT), Kharagpur. I am keenly interested in contributing to moveit! by working on the creation of Grasping library. I am familiar with ROS and OMPL, and now I am getting myself acquainted with the planning library moveit!. 
How should I start with the project? Please help @davetcoleman .

davetcoleman, 2019-02-27:
PickNik did not apply in time for GSoC this year, but perhaps there will be some slots again from Open Robotics @caguero ?

caguero, 2019-02-27:
It will depend on the number of slots that we get from Google but that's always a possibility :)

davetcoleman, 2019-03-06:
We are currently looking for moveit maintainers/developers who would be interested in mentoring a GSoC student this summer. Please reach out to me if you want to *mentor* a student. 

If you are a student looking for more information, I can't personally help you but see this recent response of mine: https://discourse.ros.org/t/how-to-get-involved-with-moveit/8121/2

MohmadAyman, 2019-03-20:
> We are currently looking for moveit maintainers/developers who would be interested in mentoring a GSoC student this summer. Please reach out to me if you want to mentor a student.

I would like to volunteer to mentor a student this summer.

Mahmoud-Selim, 2019-04-01:
Hello,
My name is Mahmoud. I'm a senior computer & systems engineering student. I'm very interested in participating to moveit in GSOC this year through OSRF. I'm interested in the "Perception Pipeline Improvements" project. I have a strong knowledge in opencv, deep learning and machine learning and a good knowledge with the PCL library. I was working on my proposal but i have a couple of questions. I was wondering if i could contact anyone for that matter (tried sending emails to OSRF but got no answer) ? 
Thank you so much for your time.

BryceStevenWilley, 2019-05-18:
MoveIt has one GSoC student for summer 2019, who's activity is being tracked in https://github.com/ros-planning/moveit/issues/1427. Closing this issue for now.

closed 2019-05-18 by BryceStevenWilley