moveit/moveit #1368: nvidia-docker file needs to be updated.

MohmadAyman, 2019-02-28:
### Description

[nvidia-docker version 1 is now deprecated](https://github.com/NVIDIA/nvidia-docker/wiki/About-version-2.0) , our own [gui-docker](https://github.com/ros-planning/moveit/blob/melodic-devel/.docker/gui-docker) is based on v1.

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?

felixvd, 2019-04-03:
I couldn't get the gui-docker script to work on Ubuntu 18.04. This needs fixing.

Installing nvidia-docker2 and following the [ROS.org wiki instructions](http://wiki.ros.org/docker/Tutorials/Hardware%20Acceleration#nvidia-docker2) worked for me(after [this fix](https://answers.ros.org/question/295552/cannot-reproduce-nvidia-docker2-installation-from-the-tutorial/)). Inserting the MoveIt image did not work for me immediately, but I can't reproduce the error and it works now.

I haven't seen a way to do it without building locally. The easiest way to do this might be to replace `gui-docker` with the instructions from the ROS wiki. I will submit a PR for it in a minute where we can discuss further.

Further reading: https://github.com/NVIDIA/nvidia-docker/issues/534

felixvd, 2019-08-27:
This was fixed in #1532 I believe.

closed 2019-10-17 by rhaschke