moveit/moveit #1371: [MSA][melodic] Build failure on buildfarm: Could not find a package configuration file provided by "Qt5"

130s, 2019-03-03:
### Description

Reported at https://github.com/ros-planning/moveit/issues/1326#issuecomment-467351712 after premature 1.0.0 release went into rosdistro.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source
* If source, which git commit or tag?: 1.0.0

### Steps to reproduce & Actual behaviour

See the [buildfarm log: Mbin_uB64__moveit_setup_assistant__ubuntu_bionic_amd64__binary/36](http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_setup_assistant__ubuntu_bionic_amd64__binary/36/console#console-section-0)

### Expected behaviour
Build passes, a job on buildfarm passes.

### Backtrace or Console output
```
:
1.1. build binarydeb
:
23:22:24 CMake Error at CMakeLists.txt:35 (find_package):
23:22:24   By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
23:22:24   asked CMake to find a package configuration file provided by "Qt5", but
23:22:24   CMake did not find one.
23:22:24 
23:22:24   Could not find a package configuration file provided by "Qt5" (requested
23:22:24   version 5.9.5) with any of the following names:
23:22:24 
23:22:24     Qt5Config.cmake
23:22:24     qt5-config.cmake
23:22:24 
```

rhaschke, 2019-03-03:
Fixed via #1373.

closed 2019-03-03 by rhaschke

130s, 2019-03-04:
I haven't confirmed occurrence of this issue other than on ROS buildfarm. Have you confirmed @rhaschke that #1373 fixes (if so can you share how you can do so)?

rhaschke, 2019-03-04:
I don't have a "proof" that #1373 fixes this issue. It's just the difference in cmake files and package.xml, which led me to this conclusion.