moveit/moveit #1385: cleanup alignment of Eigen objects

rhaschke, 2019-03-08:
As pointed out in https://github.com/ros-planning/moveit/pull/1382#issuecomment-471003025, the alignment of Eigen objects is not sufficient anymore when building with AVX support.

Looks like we need to [modify all structs/classes having Eigen objects](https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html#title2) as members and also consider the other cases discussed [here](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html).

@negril, do you have resources to tackle this?
@davetcoleman, do you have somebody to work on this?

davetcoleman, 2019-03-08:
What is AVX? PickNik is doesn't have the resources to work on this ourselves, but of course outside contributions are welcome!

BryceStevenWilley, 2019-04-28:
For anyone who doesn't know, AVX is Advanced Vector Extensions, a type of SIMD (single instruction multiple data) operation [supported by Eigen](http://eigen.tuxfamily.org/index.php?title=FAQ#Which_SIMD_instruction_sets_are_supported_by_Eigen.3F), and when used usually speeds up linear-algebra heavy code significantly.