moveit/moveit #1386: Hardcoded file locations in loadTestingRobotModel

mlautman, 2019-03-08:
### Description

`moveit::core::loadTestingRobotModel("panda_description");` fails silently creating a test log file (eg: `MISSING-gtest-test_cartesian_interpolator.xml`). This is because the method hardcodes the srdf file location to work for `moveit_resources/pr2_description` but not franka or panda.

Overview of your issue here.

### Your environment
* ROS Distro: [melodic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Change the `robot_name` in `test_aabb.cpp` to `panda_description`

rhaschke, 2019-03-09:
@BryceStevenWilley, could you please look into this issue as you initially wrote this code?
I think we need a kind of switch-case for the three robots (pr2, panda, fanuc) to correctly choose paths.

BryceStevenWilley, 2019-03-11:
Acknowledged, I'll try to get to this this week.  From what I can tell, the issue is the fact that the PR2's SRDF is in the `pr2_description` [here](https://github.com/ros-planning/moveit_resources/tree/master/pr2_description) but Panda's SRDF is in the `panda_moveit_config` [here](https://github.com/ros-planning/moveit_resources/tree/master/panda_moveit_config/config) and not `panda_description`. The best option seems to be keeping PR2 as a hard coded non-standard srdf location, but using Panda's location as the default, so this is future proof too.

BryceStevenWilley, 2019-03-13:
@mlautman With #1391 merged this should be fixed: changing `pr2` to `panda` no longer crashes when loading the URDF/SRDF. It still segfaults (fails silently) since the tests are hardcoded to use PR2 link names, but I'm guessing that's what you were working on.

rhaschke, 2019-03-13:
Closes via #1391

closed 2019-03-13 by rhaschke