moveit/moveit #1389: Remove unnecessary null pointer checks

elfring, 2019-03-11:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [BenchmarkExecutor](https://github.com/ros-planning/moveit/blob/822a991e0ea4f713fe672e3e337d7f0702bcc224/moveit_ros/benchmarks/src/BenchmarkExecutor.cpp#L87 "Destructor for BenchmarkExecutor")
- [CoupledDeleter::operator()](https://github.com/ros-planning/moveit/blob/822a991e0ea4f713fe672e3e337d7f0702bcc224/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp#L87)
- [mesh_filter::GLRenderer](https://github.com/ros-planning/moveit/blob/7f281a3836d37e3d68e468bfbf8e871fa02cf0f9/moveit_ros/perception/mesh_filter/src/gl_renderer.cpp#L75 "Destructor for mesh_filter::GLRenderer")
- [MotionPlanningParamWidget](https://github.com/ros-planning/moveit/blob/822a991e0ea4f713fe672e3e337d7f0702bcc224/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_param_widget.cpp#L52 "Destructor for MotionPlanningParamWidget")

rhaschke, 2019-03-11:
Thanks for this hint. You are welcome to file a PR for this :wink:

rhaschke, 2019-03-21:
Addressed by #1410.

closed 2019-03-21 by rhaschke