moveit/moveit #1393: CHOMP not using the signed version of distance field?

tsijs, 2019-03-12:
### Description

CHOMP not using the signed version of distance field??

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* binary: 0.9.15

### Steps to reproduce
I do not fully understand how the hybrid collision fields get initialized in the CHOMP plugin (with the allocator templated stuff), but I think the CHOMP plugin does not make use of the signed version of the distance field collision checker. 
I created a new plugin, with identical CHOMP plugin source code and when I print the distance field distances, they are always either positive or 0. Also the negative squared distances stored in the voxels is always 0. 
Is there something I am missing or could somebody maybe explain how the collisiondetectorallocator creates the distance field if I am wrong?

BryceStevenWilley, 2019-04-03:
I took a look at this, and surprisingly this seems to be correct: [the default] `use_signed_distance_field` value is `false`, and I can't find anywhere that it's changed. I haven't yet confirmed that there are no negative distances during execution though.

Ideally, this is an easy fix: `use_signed_distance_field` should be true, CHOMP and STOMP both require SDFs to work.

tsijs, 2019-04-03:
I tried the easy fix, this seems to work for me.

tsijs, 2019-04-09:
Ho, that was premature. I thought I already fixed it in my previous pull request. Will create a pull request some time soon though (just the false in robot_distance_field.h needs to be switched).