moveit/moveit #1395: MoveIt planner not avoiding Octomap

MarcoT11, 2019-03-12:
### Description

I implemented a kinect sensor in gazebo and visualize the point cloud in RViz. So far so good, when I now plan a movement with the interaction marker move_group doesn't recognize the Octomap and plans right through it. When executing the planned path the robot arm then crashes in the simulated obstacles. 
How can I connect the point cloud with the planning scene? so that it avoids the obstacles?
Cheers Marco

### Your environment
* ROS Kinect
* Ubuntu 16.04

### Expected behaviour
The robot arm avoids the point cloud / octomap when planning the path and executing the movement.

### Actual behaviour
The path is planned through the point cloud / octomap

### When launching:

ROS_MASTER_URI=http://localhost:11311

process[joint_state_publisher-1]: started with pid [4390]
process[move_group-2]: started with pid [4391]
process[rviz_frdc51035_3834_9019541648462265413-3]: started with pid [4393]
[ INFO] [1552404162.195927853]: Loading robot model 'widowx_arm'...
[ WARN] [1552404162.196027458]: Skipping virtual joint 'world_joint' because its child frame 'base_footprint' does not match the URDF frame 'world'
[ INFO] [1552404162.196063752]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1552404162.289673207]: rviz version 1.12.16
[ INFO] [1552404162.289716126]: compiled against Qt version 5.5.1
[ INFO] [1552404162.289730787]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1552404162.320880648]: joint_1 -2.617 2.617 1
[ INFO] [1552404162.320920875]: joint_2 -1.571 1.571 1
[ INFO] [1552404162.320939753]: joint_3 -1.571 1.571 1
[ INFO] [1552404162.320956898]: joint_4 -1.745 1.745 1
[ INFO] [1552404162.320971095]: joint_5 -2.617 2.617 1
[ INFO] [1552404162.347132216]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1552404162.350261031]: MoveGroup debug mode is OFF
Starting context monitors...
[ INFO] [1552404162.350303439]: Starting scene monitor
[ INFO] [1552404162.353461624]: Listening to '/planning_scene'
[ INFO] [1552404162.353492377]: Starting world geometry monitor
[ INFO] [1552404162.356477044]: Listening to '/collision_object' using message notifier with target frame '/world '
[ INFO] [1552404162.359109876]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1552404162.368381841]: Listening to '/attached_collision_object' for attached collision objects
Context monitors started.
[ INFO] [1552404162.396910461]: Initializing OMPL interface using ROS parameters
[ INFO] [1552404162.447350554, 0.954000000]: Using planning interface 'OMPL'
[ INFO] [1552404162.454662763, 0.962000000]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1552404162.455372009, 0.963000000]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1552404162.455908661, 0.963000000]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1552404162.456662107, 0.964000000]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1552404162.457166986, 0.964000000]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1552404162.457780543, 0.965000000]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1552404162.457855552, 0.965000000]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1552404162.457879185, 0.965000000]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1552404162.457897057, 0.965000000]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1552404162.457918739, 0.965000000]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1552404162.457935344, 0.965000000]: Using planning request adapter 'Fix Start State Path Constraints'
[ WARN] [1552404162.619826558, 1.128000000]: The complete state of the robot is not yet known.  Missing gripper_joint
[ INFO] [1552404162.714877966, 1.223000000]: Added FollowJointTrajectory controller for widowx_controller
[ INFO] [1552404162.714968669, 1.223000000]: Returned 1 controllers in list
[ INFO] [1552404162.730354535, 1.237000000]: Trajectory execution is managing controllers
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1552404162.813064194, 1.320000000]: 

MoveGroup using: 
     - ApplyPlanningSceneService
     - ClearOctomapService
     - CartesianPathService
     - ExecuteTrajectoryAction
     - GetPlanningSceneService
     - KinematicsService
     - MoveAction
     - PickPlaceAction
     - MotionPlanService
     - QueryPlannersService
     - StateValidationService

[ INFO] [1552404162.813134750, 1.320000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1552404162.813162877, 1.320000000]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1552404163.254872736, 1.762000000]: Stereo is NOT SUPPORTED
[ INFO] [1552404163.254943482, 1.762000000]: OpenGl version: 4.5 (GLSL 4.5).
0x1039a20 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
0x1016f50 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
0x1041cd0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
0x1794bb0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
0x1937580 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
0x17c2d50 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
0x1937750 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
0x19a8ff0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x85dd80) ): Attempt to set a screen on a child window.
[ INFO] [1552404166.806754545, 5.305000000]: Loading robot model 'widowx_arm'...
[ WARN] [1552404166.806822275, 5.305000000]: Skipping virtual joint 'world_joint' because its child frame 'base_footprint' does not match the URDF frame 'world'
[ INFO] [1552404166.806850654, 5.305000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1552404166.865684160, 5.365000000]: joint_1 -2.617 2.617 1
[ INFO] [1552404166.865717953, 5.365000000]: joint_2 -1.571 1.571 1
[ INFO] [1552404166.865732507, 5.365000000]: joint_3 -1.571 1.571 1
[ INFO] [1552404166.865746375, 5.365000000]: joint_4 -1.745 1.745 1
[ INFO] [1552404166.865760941, 5.365000000]: joint_5 -2.617 2.617 1
[ INFO] [1552404166.928448593, 5.428000000]: Starting scene monitor
[ INFO] [1552404166.931960497, 5.432000000]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1552404166.983094742, 5.483000000]: Constructing new MoveGroup connection for group 'widowx_arm' in namespace ''
[ INFO] [1552404168.130100800, 6.627000000]: Ready to take commands for planning group widowx_arm.
[ INFO] [1552404168.130177674, 6.627000000]: Looking around: yes
[ INFO] [1552404168.130212336, 6.627000000]: Replanning: yes
[ WARN] [1552404168.149924106, 6.647000000]: Interactive marker 'EE:goal_wrist_2_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.

rhaschke, 2019-03-12:
What do you mean by "visualize the point cloud in rviz"? Is it just rviz' point cloud display?
Or is the point cloud part of the planning scene?
Did you followed the [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html)?

MarcoT11, 2019-03-13:
I added a PointCloud2 in RViz which uses the Topic /kinect/depth/points.
This idea came from this https://www.youtube.com/watch?v=kjgEdRJ72Tw tutorial, which also refers to the original tutorial from the MoveIt website.

MarcoT11, 2019-03-13:
Same happens when I do the tutorial from the MoveIt website 

[image]
[image]
[image]

On [http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html](url) it says 

> If you set the initial and the final location of the robot in a way that there is no straight path between them, then the planner will automatically avoid the octomap and plan around it

felixvd, 2019-03-13:
The octomap in your picture does not look like the one from the tutorial. Are you sure everything is configured the same way? If you expand the PlanningScene module in Rviz, can you find the Octomap there? If not, and it disappears from view if you uncheck the PointCloud2 Rviz module, then the PlanningScene inside MoveIt does not know about the Octomap, and it needs to be registered there.

MarcoT11, 2019-03-13:
Hey Felix, when I do the Perception Pipeline Tutorial of the MoveIt website after the Getting Started guide and launch 

> roslaunch moveit_tutorials obstacle_avoidance_demo.launch

this is the output
[image]

the tutorial says when launching that I should get a image like this
[image]

So I added PointCloud2 by myself.
When I expand the PlanningScene module I can't find the octomap, neither in the MoveIt tutorial nor in my robot arm application. 
Yes when I uncheck PointCloud2 RViz module it dissapears.
How can I register the octomap in the PlanningScene?

MarcoT11, 2019-03-13:
For my example I first launch Gazebo.
With _rostopic list | grep kinect_ I get 
> /kinect/depth/camera_info
> /kinect/depth/image_raw
> /kinect/depth/points
> /kinect/parameter_descriptions
> /kinect/parameter_updates
> ...

then I created the _kinect_config.yaml_
```
 sensor:
    - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
      point_cloud_topic: /kinect/depth/points
      max_range: 5.0
      point_subsample: 1
      padding_offset: 0.1
      padding_scale: 1.0
      filtered_cloud_topic: filtered_cloud
```

then I updated the _sensor_manager.launch_ file:
```
<launch>

 <!-- Param for kinect config -->
 <rosparam command="load" file="$(find widowx_arm_moveit_new)/config/kinect_config.yaml" />

  <!-- Params for the octomap monitor -->
  <param name="octomap_frame" type="string" value="arm_base_joint" />
  <param name="octomap_resolution" type="double" value="0.05" />
  <param name="max_range" type="double" value="5.0" />
  
</launch>
```

in the end I launch MoveIt with the controllers. The _sensor_manager.launch_ file is included in move_group
```
<launch>

  <!-- By default, we do not start a database (it can be large) -->
  <arg name="db" default="false" />
  <!-- Allow user to specify database location -->
  <arg name="db_path" default="$(find widowx_arm_moveit_new)/default_warehouse_mongo_db" />

  <!-- By default, we are not in debug mode -->
  <arg name="debug" default="false" />

  <arg name="use_gui" default="false" />

 <rosparam file="$(find widowx_arm_moveit_new)/config/joint_names.yaml" command="load"/>

  <include file="$(find widowx_arm_moveit_new)/launch/planning_context_k.launch">
    <arg name="load_robot_description" value="true"/>
  </include>

  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <param name="/use_gui" value="false"/>
    <rosparam param="/source_list">[/joint_states]</rosparam>
  </node>

  <include file="$(find widowx_arm_moveit_new)/launch/move_group_k.launch">
    <arg name="publish_monitored_planning_scene" value="true"/>
  </include>

  <include file="$(find widowx_arm_moveit_new)/launch/moveit_rviz.launch">
    <arg name="config" value="true"/>
  </include>

  <!-- If database loading was enabled, start mongodb as well -->
  <include file="$(find widowx_arm_moveit_new)/launch/default_warehouse_db.launch" if="$(arg db)">
    <arg name="moveit_warehouse_database_path" value="$(arg db_path)"/>
  </include>

</launch>
```

Somehow there is no connection between _/kinect/depth/points_ and _move_group_
[image]

MarcoT11, 2019-03-13:
The MoveIt Panda exaple looks like this
[image]

moriarty, 2019-03-14:
@MarcoT11: It would be easier if you could link to the git repository and branch where you're keeping the launch files, but check if the include of sensor_manager is being guarded by an **if arg** :

> in the end I launch MoveIt with the controllers. The _sensor_manager.launch_ file is included in move_group
> 
> ```
> <launch>
> 
>   ....
> 
> </launch>
> ```

I don't see the _sensor_manager.launch_ file in that snippet you've included, I assume it's being included in:
> ```
>   <include file="$(find widowx_arm_moveit_new)/launch/move_group_k.launch">
>     <arg name="publish_monitored_planning_scene" value="true"/>
>   </include>
> ```

It's sometimes guarded by a argument, for example `if="$(arg allow_trajectory_execution)"`. 
And this argument may have a default value, the panda sets the default value to **true** on L23:
https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/launch/move_group.launch#L42-L45

```
<!-- Sensors Functionality -->
<include ns="move_group" file="$(find panda_moveit_config)/launch/sensor_manager.launch.xml" if="$(arg allow_trajectory_execution)">
  <arg name="moveit_sensor_manager" value="panda" />
</include>
```

When launching the panda `demo.launch` this argument is set to **true**
https://github.com/ros-planning/panda_moveit_config/blob/kinetic-devel/launch/demo.launch#L40-L46
```
  <!-- Run the main MoveIt executable without trajectory execution (we do not have controllers configured by default) -->
  <include file="$(find panda_moveit_config)/launch/move_group.launch">
    <arg name="allow_trajectory_execution" value="true"/>
    <arg name="fake_execution" value="true"/>
    <arg name="info" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
</include>
```

Also, just wanted to warn that this argument could be named differently. For example when using the Fetch it is `allow_active_sensing` which we default to **false**.
https://github.com/fetchrobotics/fetch_ros/blob/melodic-devel/fetch_moveit_config/launch/move_group.launch#L39

please post a link to `(find widowx_arm_moveit_new)/launch/move_group_k.launch`

ejichen, 2019-03-15:
@MarcoT11 I encountered a similar problem few days ago. The octomap wasn't showing in my rviz as well. I did two things to make the octomap visualized.

1.  In your ```obstacle_avoidance_demo.launch``` add ```  <node pkg="tf2_ros" type="static_transform_publisher" name="kinect_broadcaster" args="0 0 0 0 0 0 1 world panda_link0" />``` right before the ```  <node pkg="tf2_ros" type="static_transform_publisher" name="to_temp_link" args="0 0.4 -0.6 0 0 0  temp_link world" />```

2. Install octomap plugin by executing ```sudo apt-get install ros-kinetic-octomap``` in your terminal. You could follow this [page](http://wiki.ros.org/octomap) to make sure the octomap is installed or being referenced by the cmake file.

Not sure if these will help though.

MarcoT11, 2019-03-15:
Hello all and thanks for the help. After all I tried I finally found the mistake.

[image]

I simply forgot the s in sensors

[image]

Sometimes you are closer to the goal then you think.

closed 2019-03-15 by MarcoT11

felixvd, 2019-03-16:
> How can I register the octomap in the PlanningScene?

I haven't done it, but that sounds like a question for ROS Answers or the tutorial. Did you solve the problem? If yes, please post the solution here for posterity. If not, you will probably get more responses on ROS Answers.