moveit/moveit #1396: [Help] the difference between "Motion planning API"  "Motion planning pipeline" "Move group Interface"

ghx0x0, 2019-03-14:
hello,
   I'm a freshman in moveit.  I read and tried the moveit toturials, and seems I can plan trajectories in 3 ways, using motion planning API/ motion planning pipeline/ Move group:
   1. planning_interface::PlanningContextPtr context;
       context->solve(res);
   2. planning_pipeline->generatePlan(planning_scene, req, res);
   3. moveit::planning_interface::MoveGroupInterface::Plan my_plan;
       move_group.plan(my_plan) ;

  but, can anybody tell me the difference between them?

felixvd, 2019-03-14:
This is a question that's better suited to ROS Answers. I think it would be better if you posted it there. In the meantime, does [this article](https://moveit.ros.org/documentation/concepts/) help?

ghx0x0, 2019-03-14:
> This is a question that's better suited to ROS Answers. I think it would be better if you posted it there. In the meantime, does [this article](https://moveit.ros.org/documentation/concepts/) help?

thank you so much, but this material still can't give me a clear definition of the 3 planning APIs.
I'll ask for advises in ROS Answers.
And thanks for your suggestion :)

felixvd, 2019-04-04:
Did you solve this problem and/or have a link to your ROS answers post?

ghx0x0, 2019-04-12:
> Did you solve this problem and/or have a link to your ROS answers post?

Hello, I've solve the problem. But I can't register a ROS answers ID in China, the link is https://github.com/ros-planning/moveit_tutorials/issues/296.
Thanks for your help.

BryceStevenWilley, 2019-05-18:
Since this is solved, I'm going to go ahead and close.

closed 2019-05-18 by BryceStevenWilley