moveit/moveit #1398: Warnings when a child-frame of a parent which is not the planning-frame is removed

sdfryc-uts, 2019-03-15:
### Description
When I run a static transform publisher, which publishes a frame as a child of anything other than the robot root frame (e.g. a camera frame) and kill this publisher after some time, the following warning spams my command line:

[ WARN] [1413291847.379498804]: Unable to transform object from frame 'camera_frame' to planning frame '/map' (Lookup would require extrapolation into the past.  Requested time 1413291832.254778736 but the earliest data is at time 1413291837.374000000, when looking up transform from frame [camera_frame] to frame [map])

(...)

[ WARN] [1413291927.826811056]: Unable to transform object from frame 'camera_frame' to planning frame '/map' (Lookup would require extrapolation into the past.  Requested time 1413291832.254778736 but the earliest data is at time 1413291917.822000000, when looking up transform from frame [camera_frame] to frame [map])

(...and so on...)

As you can see, the requested time stays the same (which is the time I killed the static transform publisher).

The error comes from line 1311 in this file (branch kinetic-devel):
https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L1311
Overview of your issue here.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Binary build
* Release version - how can I find this?

### Steps to reproduce
Broadcast a static TF by running:
`$ rosrun  tf  static_transform_publisher  0  0  1  0  0  0  1   /[some robot frame]  /dummy_frame   100`

Then kill it via Ctrl + C, wait for the dummy_frame to fade out in rviz

### Expected behaviour
My goal is to use the ar_track_alvar package to detect objects using ar tags. I have configured the ar_track_alvar package to broadcast the TF of these tags relative to the robots camera frame. If the camera can no longer see a tag, it ceases to broadcast the TF of the ar marker, and that is when these warning messages start to spam the console.

sdfryc-uts, 2019-03-18:
Further information: I have discovered that this warning only appears if I have octomap mapping configured/enabled. Note: By 'Octomap mapping' I'm referring to the implementation within MoveIt!, and not the stand-alone package.

machinekoder, 2021-07-07:
I can reproduce this error, however, also with a normal tf publisher.

v4hn, 2021-07-20:
> My workaround is to keep publishing the unused tf frames.

That's also what I did in the past, but it would be appreciated if someone would spend the time to actually fix the underlying bug:
The error should only appear if a frame is actually *required* for a transform, but currently it is triggered **for every disappearing frame** (See [here](https://github.com/ros-planning/moveit/blob/d1b4cd3dcfdf81325f75acf06b6997e41d1580d6/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L1338))

takeshi-pumas, 2022-02-27:
I've been going crazy about this the whole weel, Nice to finally put a rest to it.

v4hn, 2022-02-28:
@takeshi-pumas where's your PR? :)