moveit/moveit #1399: Missing interactive markers

imcardle, 2019-03-15:
### Description

I'm running through the MoveIt! getting started tutorial but I'm not able to see the "interactive markers" in rViz. I've clicked on the "Interact" button, as suggested in the tutorial - but still the markers don't show up.

### Your environment
ROS Distro: Melodic
OS Version: Ubuntu 18.04 (running in VirtualBox)
Binary build: Where can I find this information?
RViz: 1.13.1
Qt: 5.9.5
OGRE: 1.9.0

### Steps to reproduce
Follow the tutorial steps here:
https://ros-planning.github.io/moveit_tutorials/

### Expected behaviour
Interactive markers should appear in the 3D view of rViz when the "Query Start State" and "Query Goal State" options are ticked, and when the "Interact" toolbar button is pressed.

### Actual behaviour
The robot is shown, and toggling the "Query Start/Goal State" options changes the colour of the robot, but there are no interactive markers so I am unable to change the start/goal positions by dragging the robot.

### Backtrace or Console output

The console output includes the following warnings - which may or may not be related to this issue:
[ WARN] [1552568009.202946440]: Unable to update multi-DOF joint 'virtual_joint': Failure to lookup transform between 'world' and 'panda_link0' with TF exception: "world" passed to lookupTransform argument target_frame does not exist.
[ WARN] [1552568009.499416675]: The complete state of the robot is not yet known. Missing virtual_joint

Raised a similar issue in the rViz repo, but seems the cause is more likely to be in the MoveIt! tutorial:
ros-visualization/rviz#1351

rhaschke, 2019-03-15:
@mlautman, could you take care of this, as this is related to the tutorials?

imcardle, 2019-03-15:
OK - I think I've found a big clue as to the cause of this issue.  I took a quick look at the recent changes to the panda_moveit_config made on 10th Feb and noticed that the virtual joint was changed from "fixed" to "floating".  I tried changing it back on my local environment and now I am able to see those interactive markers!
[https://github.com/ros-planning/panda_moveit_config/commit/d97b987a7bfc4ec0d10fa01d1ae518a67daf31d4](https://github.com/ros-planning/panda_moveit_config/commit/d97b987a7bfc4ec0d10fa01d1ae518a67daf31d4)
The issue still needs to be looked at by someone with more experience than I have, because I don't know the reason for the change or what it might affect if we change it back.

saifullah3396, 2019-08-06:
The solution provided by @imcardle does not work for me with the same environment configuration. So I believe that's not exactly the solution to this problem.

rhaschke, 2019-08-30:
Probably fixed via https://github.com/ros-planning/moveit_tutorials/pull/387.
Open a new issue, if there is still a problem.
One thing to check is: Is the correct group selected? Only the `arm` group provides interactive markers, while the hand/gripper doesn't.

closed 2019-08-30 by rhaschke