moveit/moveit #1401: planner not found

aliakbarhamzeh1378, 2019-03-18:
### Description

Hi, I have a robot very similar to Kuka youbot
I'm having trouble when I want to simulate it in moveit :
1-Don't move the arm with change goal position in rviz .
 2-The forward kinematic works fine, but when I try to test the inverse kinematics, that show me, planner not found
Can anyone help me solve this problem?

### Your environment
* ROS Distro:|Kinetic
* OS Version: e.g. Ubuntu 16.04

felixvd, 2019-03-20:
You should post more details about your setup, what you are doing, and the error message. Ideally, post it on ROS Answers. It looks like you will get better answers there.

aliakbarhamzeh1378, 2019-03-20:
> You should post more details about your setup, what you are doing, and the error message. Ideally, post it on ROS Answers. It looks like you will get better answers there.

### thanks for answer
this function (go_to_pose_goal) in move_group_python_interface_tutorial.py give me that error...
I tested a lot of position, but it still gives me the same.
this is my setup_assistant output:
[model_assistant.zip](https://github.com/ros-planning/moveit/files/2987943/model_assistant.zip)

felixvd, 2019-03-20:
Please paste the console output instead of uploading a zip file, and post this as a question on [this website](https://answers.ros.org/).

BryceStevenWilley, 2019-05-18:
As Felix mentioned, this type of question is best suited for ROS Answers. Closing.

closed 2019-05-18 by BryceStevenWilley