moveit/moveit #1413: Build from Source Failed

jingxixu, 2019-03-23:
### Description
Building from source fails.

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build? Source
* If source, which git commit or tag? I followed instructions here: https://moveit.ros.org/install/source/ 

### Steps to reproduce
Just follow instructions from https://moveit.ros.org/install/source/

### Expected behaviour
Should build successfully.

### Actual behaviour
Failed as follows:
```
____________________________________________________________________________________________________
Errors     << moveit_ros_planning_interface:cmake /home/jxu/ws_moveit/logs/moveit_ros_planning_interface/build.cmake.003.log
CMake Error at /usr/local/share/cmake-3.12/Modules/FindBoost.cmake:2048 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.58.0

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python

  Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  CMakeLists.txt:37 (find_package)

cd /home/jxu/ws_moveit/build/moveit_ros_planning_interface; catkin build --get-env moveit_ros_planning_interface | catkin env -si  /usr/local/bin/cmake /home/jxu/ws_moveit/src/moveit/moveit_ros/planning_interface --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/jxu/ws_moveit/devel/.private/moveit_ros_planning_interface -DCMAKE_INSTALL_PREFIX=/home/jxu/ws_moveit/install -DCMAKE_BUILD_TYPE=Release; cd -
....................................................................................................
```

jingxixu, 2019-03-24:
It seems that I solved it. There are three things that we need to pay attention:

### First
Under _Download and Build MoveIt!_, it says
> This branch builds for ROS Kinetic and newer, e.g. Ubuntu 16.04 and newer.

But this is **NOT** true. We should use commands from _Advanced: Building Older Releases Of MoveIt!_.

### Second
In line 29 of `moveit/moveit_ros/planning_interface/CMakeLists.txt`, I change  `if(Boost_VERSION LESS 106700)` to `if(Boost_VERSION LESS 100000)`.

In this way, given that boost version of Ubuntu 16.04 is 1.58.0. CMake will try to find boost_python2.7 instead of boost_python.

### Third
The latest branch of package panda_moveit_config does not show interactive markers. I am using the commit `3a1e58db644d3f61634bf9802c133f391b7be61b`.

jingxixu, 2019-03-24:
I am open to any comment or feedback; otherwise, this issue can be closed.

rhaschke, 2019-03-24:
Thanks for your feedback. As far as I can see, you tried to build MoveIt's master branch on Kinetic, right?
I'm confused about the boost-python issues: [Travis CI successfully builds on Kinetic](https://travis-ci.org/ros-planning/moveit). Do you use any special setup?

The third issue seems to be a duplicate of #1399.
@mlautman could you, please, have a look at this asap? You introduced the offending commit.