moveit/moveit #1414: Motion Planning API tutorial stuck in loop waiting on haveCompleteState.  Related to #1399

tylerjw, 2019-03-25:
### Description

The [Motion Planning API](https://ros-planning.github.io/moveit_tutorials/doc/motion_planning_api/motion_planning_api_tutorial.html)  gets stuck in an eternal loop because `psm->getStateMonitor()->haveCompleteState` never returns true.  The `virtual_joint` is always missing.  I changed the loop to show what joints are missing:

```c++
  std::vector<std::string> missing_joints;
  ros::WallDuration sleep_t(1.0);
  while (!psm->getStateMonitor()->haveCompleteState(missing_joints) && ros::ok())
  {
    std::stringstream ss;
    for (auto & joint : missing_joints) ss << joint << ", ";
    ROS_INFO_STREAM_NAMED(node_name, "missing_joints (" << missing_joints.size() << "): " << ss.str());
    missing_joints.clear();
    sleep_t.sleep();
  }
```

If I revert the change ros-planning/panda_moveit_config@d97b987 mentioned in #1399 from "floating" to "fixed" and am getting past the loop.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 16.04
* Source @fec7ef52b

### Steps to reproduce
Change the loop starting on line 95 to the above code in [motion_planning_api_tutorial.cpp](https://github.com/ros-planning/moveit_tutorials/blob/master/doc/motion_planning_api/src/motion_planning_api_tutorial.cpp).

Setup Tutorial Enviroment: [Getting Started](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html)

Start RViz and wait for it to finish loading:
`roslaunch panda_moveit_config demo.launch`

Run the launch file:
`roslaunch moveit_tutorials motion_planning_api_tutorial.launch`

Observe the output:
`missing_joints (1): virtual_joint, ...`

Change the panda_moveit_config/config/panda_arm.xacro line 24 type from `floating` to `fixed`.

Rerun the tutorial, observe it crashes later in the program.

rhaschke, 2019-03-25:
@mlautman, another user affected by your change to panda_moveit_config. Please fix this asap.

gavanderhoorn, 2019-03-25:
There are at least 4 Q&As on ROS Answers about this as well.

rhaschke, 2019-08-30:
Fixed via https://github.com/ros-planning/moveit_tutorials/pull/387.

closed 2019-08-30 by rhaschke