moveit/moveit #1415: Invoking "cmake" failed

anubhav1772, 2019-03-26:
My environment - 
* ROS Distro: melodic
* OS Version: Ubuntu 18.04
* Source build - https://github.com/ros-planning/moveit.git

Hi, I am trying to build a ROS package 'moveit' but getting the following error:

> -- +++ processing catkin package: 'moveit_kinematics'
> -- ==> add_subdirectory(hrwros/moveit/moveit_kinematics)
> -- Boost version: 1.65.1
> -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
> CMake Error at /home/anubhav1772/catkin_ws/devel/share/ur_kinematics/cmake/ur_kinematicsConfig.cmake:148 (message):
>   Project 'moveit_kinematics' tried to find library 'ur3_moveit_plugin'.  The
>   library is neither a target nor built/installed properly.  Did you compile
>   project 'ur_kinematics'? Did you find_package() it before the subdirectory
>   containing its code is included?
> Call Stack (most recent call first):
>   hrwros/moveit/moveit_kinematics/cached_ik_kinematics_plugin/CMakeLists.txt:2 (find_package)

I ran find_package(wr3_moveit_plugin REQUIRED) but getting syntax error. Could someone help me locating the package or pointing out the error? Your help/advice would be highly appreciated.

Thank you.

rhaschke, 2019-03-26:
Looks like you have the package ur_kinematics in your workspace (which is an optional dependency for moveit_kinematics), but you are missing related UR packages. Either you pull in the remaining UR packages or you remove ur_kinematics.

anubhav1772, 2019-03-26:
I already have the 'ur_kinematics' inside **universal_robot** package and I think it's a full package. I think the problem is with **ur3_moveit_plugin** . I doubt if 'ur_kinematics' or 'ur3_moveit_plugin' is compatible with ros-melodic. Correct me if I am wrong.

rhaschke, 2019-03-26:
> CMake Error at catkin_ws/devel/share/ur_kinematics/cmake/ur_kinematicsConfig.cmake:148 (message):
Project 'moveit_kinematics' tried to find library 'ur3_moveit_plugin'. 
The library is neither a target nor built/installed properly.

The error message states that the library `ur3_moveit_plugin` is not provided as expected by ur_kinematics package. Please file an issue there. This is not related to MoveIt.

closed 2019-03-26 by anubhav1772