moveit/moveit #1421: loading collision scene from text does not load into propagation_distance_field

tsijs, 2019-03-30:
### Description

When loading a scene from text, it does not get loaded into the propagation_distance_field instance, it seems. Could possibly be related to issue #305.

### Your environment
* ROS Distro: Kinetix
* OS Version: Ubuntu 16.04
* Binary and source
* devel - latest

### Steps to reproduce
1. Save a collision scene with the moveit gui. 
2. Close rviz and ros. 
3. Restart rviz with the chomp demo from a moveit config. 
4. Now when planning around a loaded obstacle, chomp does not detect it. 

### Expected behaviour
The collision objects should be loaded into the distance field as well as in rviz.

### Actual behaviour
Chomp does not detect it.

tsijs, 2019-04-01:
Ok, for other people who encounter the same issue: I found a work-around. It's not a bug, it's a "feature". After changing the environment, you have to press the "publish" button before it gets recognized. But if someone wants to use this functionality, please consider the MongoDB plugin as well. The interface for that one is way nicer and it works better I feel:)

BryceStevenWilley, 2019-06-12:
I think it's pretty clear that you should press the publish button in order for the changes you make to the scene to be published. I'm going to close this, but feel free to reopen if you disagree/what changes you'd make to the panel/workflow.

closed 2019-06-12 by BryceStevenWilley