moveit/moveit #1425: Missing gazebo_ros, panda_moveit_config dependency found in .launch files

andersfischernielsen, 2019-04-02:
I'm currently doing some research on static analysis of ROS packages at the IT University of Copenhagen, more specifically dependency checking of `.launch` files in ROS packages.

There seems to be a `$(find gazebo_ros)` in
- `moveit/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/gazebo.launch`
that's not declared in `moveit/moveit_setup_assistant/package.xml`

Installing `moveit_setup_assistant` without having `gazebo_ros` installed would make `roslaunch` fail on this `.launch` file, right? 

I'm unsure whether this is a dependency bug, since gazebo_ros might be installed during the installation of the rest of the MoveIt framework. 

---

There's also a dependency on `panda_moveit_config` in 
- the `panda` launch files in `moveit/moveit_ros/benchmarks/examples/` that's undeclared in `moveit/moveit_ros/benchmarks/package.xml`.

Again, I'm unsure if this is a bug or not. 

---

### Your environment
* ROS Distro: [Melodic (docker)]
* OS Version: e.g. Ubuntu 16.04
* Source
* `master` branch (current)

gavanderhoorn, 2019-04-03:
This dependency was added in #1051, but the package manifest wasn't updated.

> Installing `moveit_setup_assistant` without having `gazebo_ros` installed would make `roslaunch` fail on this `.launch` file, right?

yes it would.

> I'm unsure whether this is a dependency bug, since gazebo_ros might be installed during the installation of the rest of the MoveIt framework.

a missing dependency declaration is definitely a problem, and depending on dependencies to bring in first-class dependencies of your own package is never acceptable (ie: if `gazebo_ros` is brought in transitively, that is not a rationale for not stating that dependency yourself).

The proper fix would appear to be to add `gazebo_ros` (and any other missing dependencies) to the [package.xml template](https://github.com/ros-planning/moveit/blob/c76c9d690cda1dd41af6f6d6ec069fc52ff0bc39/moveit_setup_assistant/templates/moveit_config_pkg_template/package.xml.template).

rhaschke, 2019-04-04:
I need to intervene here: gazebo is not a dependency of moveit_setup_assistant (MSA), but a dependency of the moveit_config package created via MSA (and this only if the user chooses to configure the simulation). It's the responsibility of the config file creation in MSA to declare this dependency. @MohmadAyman, can you please verify that this is done?

gavanderhoorn, 2019-04-04:
@rhaschke: it's missing from the [package.xml template](https://github.com/ros-planning/moveit/blob/c76c9d690cda1dd41af6f6d6ec069fc52ff0bc39/moveit_setup_assistant/templates/moveit_config_pkg_template/package.xml.template). That's also what @andersfischernielsen reports.

rhaschke, 2019-04-04:
@gavanderhoorn: gazebo_ros should be added to the generated package.xml if and only if the simulation stuff is configured for the created moveit_config package. There is a place holder [`OTHER_DEPENDENCIES`](https://github.com/ros-planning/moveit/blob/c76c9d690cda1dd41af6f6d6ec069fc52ff0bc39/moveit_setup_assistant/templates/moveit_config_pkg_template/package.xml.template#L31) in package.xml template, which should be used to fill in such optional dependencies. That's what I asked @MohmadAyman to confirm.

gavanderhoorn, 2019-04-04:
Isn't that what I wrote?

Maybe I misunderstood you, or you misunderstood me, not sure which.

rhaschke, 2019-04-04:
@gavanderhoorn, you proposed to _unconditionally_ add gazebo_ros to the package.xml template:
> The proper fix would appear to be to add gazebo_ros (and any other missing dependencies) to the package.xml template.

My key point is to add this dependency `conditionally`, if it is actually needed.
If we agree on this, it was indeed a misunderstanding :wink:

MohmadAyman, 2019-04-05:
Okay, [here](https://github.com/ros-planning/moveit/pull/1051/files#diff-f5f35f87956cc08cfc51d5eeaff894cbR33) I added to the `package.xml`
```xml
  <!-- Optional: Gazebo integration. This is a heavy dependency so is disabled by default -->
  <!-- run_depend>gazebo_ros</run_depend -->
```
While adding the gazebo_simulation screen, I realized now that it is no longer in there.

Thats aside, I don't think we should (or have to) add the dependency automatically because we don't know if the user will use the gazebo simulation or not. So I propose we get those deleted two lines back into the `packge.xml` and maybe add a warning or a sth to the `simulation screen` gui that tells the user they should uncomment those two lines from the package.xml.

What do you think?

rhaschke, 2019-04-05:
@MohmadAyman, you are exactly right, the gazebo_ros dependency should be added `optionally`. I tried to emphasize this point several times. **There is already a mechanism prepared for this**: the placeholder `OTHER_DEPENDENCIES`, which is filled [here](https://github.com/ros-planning/moveit/blob/d3f6b9e758c1b0248551676049f36923c110f782/moveit_setup_assistant/src/widgets/configuration_files_widget.cpp#L1093-L1104) with a urdf dependency package. You should add gazebo_ros to the list, if it's needed.

MohmadAyman, 2019-04-05:
> if it's needed.

@rhaschke  I see what you mean now, 
`// Pair 8 - Add dependencies to package.xml if the robot.urdf file is relative to a ROS package`
I thought `OTHER_DEPENDENCIES` had a different usage, we can add the `gazebo_ros` dep to it.

but there is still if the user used the simulation screen to generate the `URDF` and used it, `moveit` wouldn't know that, a warning in the `gui` of the `simulation screen` or maybe a console warning, what do you think?

rhaschke, 2019-04-05:
@MohmadAyman, I don't get your question. If the user chooses to generate the simulation files, you can also add the gazebo_ros dependency, cant' you?

MohmadAyman, 2019-04-05:
@rhaschke so whenever the user chooses to generate the new `urdf`, we add the dep, I get that. I just meant the the user could click `generate urdf` in the sim screen and not really use it. I'm not sure how likely this could happen, but okay lets add it when they generate the urdf.

simonbogh, 2021-02-09:
How does the `gazebo_ros` dependency play out if you actually use a different Gazebo version than the standard e.g. Gazebo 11 under Melodic, which would mean the dependency is `gazebo11_ros`, hence installing according to the dependency would remove Gazebo 11 and install Gazebo 9 instead?

closed 2022-06-29 by andersfischernielsen