moveit/moveit #1426: Hybrid collision checker plugin not found

tsijs, 2019-04-04:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build
* If source, which git commit: master branch currently

### Steps to reproduce
When I build moveit from source and run:

roslaunch panda_moveit_config demo_chomp.launch 

I get the error that the hybrid collision checker plugin does not exist.
Also, Chomp produces very non-smooth paths all the sudden. When I ran chomp with the binary build, this was not a problem. Could this be related?
Also can somebody confirm that this is not a local issue on my pc? I rebuild the package a couple of times, but the behavior is persistent. 

### Expected behaviour
It should exist I think?

### Actual behaviour
It does not exist or isn't found. 

### Backtrace or Console output

[ INFO] [1554394454.594068885]: Stereo is NOT SUPPORTED
[ INFO] [1554394454.594195698]: OpenGl version: 3 (GLSL 1.3).
[ERROR] [1554394455.314312931]: Exception while loading Hybrid: According to the loaded plugin descriptions the class Hybrid with base class type collision_detection::CollisionPlugin does not exist. Declared types are  FCL
[ INFO] [1554394455.314395279]: Using collision detector:FCL
[ INFO] [1554394455.389529864]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1554394455.398258091]: MoveGroup debug mode is ON
Starting planning scene monitors...
[ INFO] [1554394455.398452545]: Starting planning scene monitor
[ INFO] [1554394455.404542804]: Listening to '/planning_scene'
[ INFO] [1554394455.404726011]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1554394455.410877755]: Listening to '/collision_object' using message notifier with target frame 'yumi_base_link '
[ INFO] [1554394455.417568453]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1554394455.811779844]: Listening to '/attached_collision_object' for attached collision objects
Planning scene monitors started.
...

rhaschke, 2019-04-05:
Since #1251 (for Melodic) and #1282 (for Kinetic), the Hybrid collision checker is not available anymore as a plugin, but it is explicitly loaded as the [chomp code assumes exactly this collision checker](https://github.com/ros-planning/moveit/blob/d3f6b9e758c1b0248551676049f36923c110f782/moveit_planners/chomp/chomp_motion_planner/src/chomp_optimizer.cpp#L78).
So, the [configuration of the Hybrid collision checker in the launch file](https://github.com/ros-planning/panda_moveit_config/blob/142fb79a2ad7efcba34de71b8015ee05ece47092/launch/chomp_planning_pipeline.launch.xml#L11) is obsolete and was removed for Melodic already. 
We missed though to release this change to Kinetic. See https://github.com/ros-planning/panda_moveit_config/issues/25.
As a quick solution, I suggest to use panda_moveit_config from source.

tsijs, 2019-04-05:
That is clear, thanks for explaining. So the performance loss in CHOMP is not explained by this in that case. I will look into why that happened and report back here if I find something

tsijs, 2019-04-10:
I haven't found a reason why it seemed to perform worse. I also don't have concrete evidence that it did and I am not planning on installing moveit from binaries again just to figure this out. I suspect it was just one of the tuning parameters I unconsciously changed. This issue can be closed I think.

closed 2019-04-10 by tsijs