moveit/moveit #1427: GSoC: Improved Collision Checking and TrajOpt

j-petit, 2019-04-05:
This is a follow up discussion to https://discourse.ros.org/t/gsoc-creating-a-default-grasping-library/8521/8.

---
Project descriptions by @felixvd 
## 1) Implement continuous collision checking and integrate Bullet library
* List of prerequisities: ROS
* Description of programming skills: C++
* Difficulty level: Medium-high
* List of potential mentors: Felix von Drigalski, Bryce Willey
* Detailed Description: This project should add many useful and necessary features to MoveIt’s collision detection, including:
  * Continuous collision checking, to avoid collisions between discrete states
  * Optimizations to the collision checking process itself to improve performance (cache/scene reconstruction improvements, convex-convex collisions)
  * The ability to dynamically change large parts of the kinematics of the world (i.e. picking up a tray with many items should attach all of those items to the robot, not just the tray).
  * Integration with different collision checking libraries, such as Bullet
  * Next to API changes, this includes writing tests and tutorials.

## 2) TrajOpt motion planner and optimization libraries
* List of prerequisites: ROS, some motion planning background
* Description of programming skills: C++
* Difficulty level: Medium
* List of potential mentors: Bryce Willey
* Detailed Description: MoveIt was designed to be highly plugin-based and is able to support many different planing libraries, as seen with last year’s GSOC addition of STOMP and CHOMP. A next good candidate to be added to MoveIt is the TrajOpt planner. The changes made in the collision checking project (see above) will allow the TrajOpt planner to be added to MoveIt.
  * Assist with implementing features critical for TrajOpt, such as collision checking, including convex-convex collisions and continuous collision detection (see above)
  * Add TrajOpt as a library to be used in MoveIt
  * Add adapters that allow TrajOpt to post-process paths produced by other planners
  * Write Moveit tutorials and tests for TrajOpt motion planning

j-petit, 2019-04-05:
The cache/scene reconstruction improvements mentioned in 1) are not yet clear to me - @felixvd, could you give some more details?

simonschmeisser, 2019-04-05:
trying to clarify your question: currently MoveIt (actually the fcl collision checking plugin for MoveIt) does separate collision requests for each NxN combinations of objects in the PlanningScene. Fcl (and other collision checkers) also offer broad phase collision checking that quickly determines which parts of the scene are actually somewhat close (based on tree structures etc). This would reduce the need for some overlap checking. The first part of the narrow phase collision checking that follows afterwards is AABB based overlap checking which is still quite cheap. So this will probably bring some speedup but not incredible amounts ;)

simonschmeisser, 2019-04-05:
about convex/convex: this seems to have been integrated in fcl recently (@Levi-Armstrong knows more details) but is not yet used in any way by MoveIt. There are open (and stalled) discussions to add a "convex" tag to urdf https://github.com/ros/urdfdom/issues/108 and https://discourse.ros.org/t/add-attribute-to-indicate-if-a-mesh-is-a-convex-hull/4479/12

There are unfortunately no PRs for this so far

simonschmeisser, 2019-04-05:
For Proposal 2) @BryceStevenWilley also has a branch on integrating trajopt in OMPL, what is your status/conclusion there? https://bitbucket.org/ompl/ompl/branch/TrajOpt

felixvd, 2019-04-05:
I thought you're asking how the project affects the scene representation in MoveIt. As a quick overview, the scene is represented in MoveIt's PlanningScene as a RobotState and a `world`. The RobotState contains the robot links (including all fixed shapes defined in the URDF) and all attached items, while the world contains items spawned dynamically, like Octomaps from a depth camera or `CollisionObject`s that the robot may pick up/attach.

My understanding is that Tesseract combined the PlanningScene subunits into a [BasicEnv](https://github.com/ros-industrial-consortium/tesseract/blob/kinetic-devel/tesseract_core/include/tesseract_core/basic_env.h) class and also changed the CollisionObject msg into an [AttachableObject](https://github.com/ros-industrial-consortium/tesseract/blob/kinetic-devel/tesseract_msgs/msg/AttachableObject.msg) msg that contains VisualGeometry and CollisionGeometry.

I imagined porting at least the CollisionObject/AttachableObject changes first, to smooth the next transitions, but I might be thinking that because of [this open PR](https://github.com/ros-planning/moveit/pull/1060) that used those messages a lot. Either way, I think changing the PlanningScene to the BasicEnv class would be too broad a change, but the extensions to the collision checking would fit there and in `collision::world`.

Would be keen to hear @Levi-Armstrong and @BryceStevenWilley opinions.

j-petit, 2019-04-05:
If someone has time to spare, I wrote my [application](https://docs.google.com/document/d/1_mEUY--sBq2NtTcJ_v5dGlXu9uQRZcJ_w7v_FyOf7s4/edit?usp=sharing) following the OSRF template. Feel free to comment in the document.

simonschmeisser, 2019-04-05:
I think you need to focus on (a subset of) one of the two proposals in order to be able to finish within the given time frame. Giving some more details of what you want to achieve how also wouldn't hurt.

j-petit, 2019-04-06:
I would be happy to especially focus on the "Continuous Collision Checking" part of the proposed project. For me it is difficult the judge the scope of the project, as my experience with MoveIt is only limited. I added the following text to my application and will extend it with more specific goals until the application deadline.

> 
> MoveIt currently only implements discrete collision checking for movements of the controlled robot arm. A major drawback of discrete collision checking methods is that they may miss collisions between the sampled time steps. While there exists techniques to alleviate this problem, resulting algorithms can be relatively slow. To provide stronger guarantees, continuous collision detection (CCD) techniques have been proposed by the research community. They compute the first time of contact between two moving objects along a path.
> The new planning framework Tesseract of ROS-Industrial has implemented CCD utilizing the Bullet library. The aim of this project is porting the feature from Tesseract to MoveIt. As Tesseract draws its heritage back to MoveIt, both motion planning frameworks share similarities which makes porting feasible.
> 

- - -
FCL also implements CCD (as far as I understood from this [paper](http://gamma.cs.unc.edu/FCL/fcl.pdf)). Why not utilize then FCL instead of Bullet?

BryceStevenWilley, 2019-04-09:
@simonschmeisser Yeah, I did a lot of work for that a while back. In short, most of TrajOpt is easily portable (could do with some quality updates, but nothing impossible), but the actual optimization software used is a bit confusing. The only openly available library is [BPMPD](http://old.sztaki.hu/~meszaros/bpmpd/), of which the source code is not available. The other alternative is Gurobi, which is only free for academics. BPMPD is definitely usable (it's what's tesseract is using now), but as a stretch goal, it should probably change to [IPOPT](https://projects.coin-or.org/Ipopt) or [Ceres](http://ceres-solver.org/). The rest of TrajOpt isn't too different though; it should be similar to how CHOMP and STOMP are interfacing with MoveIt now. 

@j-petit There's been some discussion on this at https://github.com/ros-planning/moveit/issues/29#issuecomment-239026362. However, there's a key difference between what FCL is doing for CCD and what tesseract/TrajOpt uses Bullet for. IIRC, FCL is calculating the smallest distance it can travel to guarantee it won't miss any collisions. However, with Bullet, we would be taking a link at two different poses and ["casting"](https://github.com/ros-industrial-consortium/tesseract/blob/master/tesseract_collision/src/bullet/bullet_cast_simple_manager.cpp) a convex hull around those poses, resulting in a single new mesh to collision check. This is described best in the [TrajOpt paper](http://joschu.net/docs/trajopt-paper.pdf). There's some approximation error there as well, but the key idea is mostly to speed up collision checking.

tsijs, 2019-04-10:
If I could blend in here: 
I know that the package @Levi-Armstrong maintains with their own "tesseract" environment representation also has qpoases and gurobi as optional solvers, of which qpoases is completely open source. 
They were discussing to use Casadi here (https://github.com/ros-industrial-consortium/trajopt_ros/issues/67) as an interface instead of interfacing all solvers manually. Casadi supports a variety of solvers and the interfacing by using its callback API isn't too hard. 
It might be something to consider if somebody is going to port this code to a moveit plugin anyway.

felixvd, 2019-05-03:
@tsijs Thanks for the heads up. Sounds like it would also be interesting to know how your efforts to use TrajOpt with MoveIt went. What other packages did you find need to be modified and why? It sounds like there is a risk of duplicating some work. If you have time to share, please feel free.

j-petit, 2019-05-06:
I have been selected for the GSoC 2019 :)! Thanks to the community for the valuable feedback during writing my project application. Looking forward to work together with my mentor @felixvd. Did MoveIt get a second student? In the next days, I will deepen my understanding of the project and hopefully come back with many questions / details to discuss.

felixvd, 2019-05-07:
Congratulations! @BryceStevenWilley and I will mentor you together, just for the record :)

MoveIt sadly only got one GSoC student this year, but there are a few more interns at PickNik working on MoveIt as well. We are checking if it makes sense to share a Slack. I note that you are in time zone GMT+1, I am in GMT+8 (JST) and Bryce is somewhere in the US, so live calls might be tricky.

For preparation, I would suggest reading through and trying out [Tesseract](https://rosindustrial.org/news/2018/7/5/optimization-motion-planning-with-tesseract-and-trajopt-for-industrial-applications) yourself and implementing a test scene there and in MoveIt. [This PR](https://github.com/ros-planning/moveit/pull/1439) might end up being related if you work on the PlanningScene, too.

Personally I think it would be perfectly appropriate and helpful to keep an open running log of what you understood of both MoveIt's and Tesseract's architecture here. It doesn't have to be perfect or even correct, but it should help you organize, share and confirm your ideas, and will be a good point of reference for later. So, feel free to drop your thoughts often.

BryceStevenWilley, 2019-05-07:
Congrats j-petit! I second everything Felix just said, especially about dropping your thoughts here; hopefully a lot of people who have relevant knowledge will start watching this thread and can give advice when necessary. 

 I'd also suggest getting familiar with [FCL](https://github.com/flexible-collision-library/fcl) and [Bullet](https://pybullet.org/wordpress/), mostly just going through the tutorials, maybe looking at some advanced user documentation, and noting any API differences.

j-petit, 2019-05-07:
Thanks for the advice, then I'll use this issue to log my activities. Just for the record: I am in GMT**+2** because of Daylight Saving Time and mostly based in Munich.

simonschmeisser, 2019-05-07:
Nice! I'm interested in helping out with mentoring where needed and based in Freiburg, Germany (+2 as well)

Levi-Armstrong, 2019-05-09:
@j-petit I have been working with TrajOpt and Tesseract for over a year and wanted to provide some insight into some things you will will need to address for TrajOpt to be full integrated into MoveIt. Currently TrajOpt collision cost/constraint requires the minimum translation vector between two objects in order to know how to move these objects out of collision. This is currently provided GJK which is used for checking ConvexHull to ConvexHull collision shapes. What this means is that when using TrajOpt you can not check a detailed mesh to detailed mesh because it will not produce sufficient information for TrajOpt to function correctly. 

Currently convex hulls collision shapes are not support by urdfdom and urdfdom_header. I currently have an developer working on a PR for urdfdom and urdfdom_header to hopefully address this issue assuming the maintainers are supportive of adding new geometry types. 

Another thing I found is that when loading mesh files, it is using assimp load the data and then it flattens all shapes in the file into a single triangle list. This is ok for visual objects and detailed mesh collision shapes but is not ok for convex shapes. In the case of the convex shapes you would want it to represent each shape in the file as its own collision object. 

I believe these are the main items I have found outside of MoveIt that would need to be address for it to be fully supported. I hope this helps.

Also wanted to give you a head up that we will be refactoring TrajOpt in the very near future to improve the flexibility of the planner and integrate new optimization techniques provided by CASADI, PAGMO, etc..

felixvd, 2019-05-09:
Thanks for the input!

I agree that continuous collision detection without TrajOpt makes sense to focus on first. If changes to the CollisionObject and PlanningScene interface make that easier, those may be good to think about, because they are probably worth doing anyway.

j-petit, 2019-05-09:
To make this clear one more time for me: When we talk about integrating CCD into MoveIt, we assume that we want to do it by using the underlying algorithms of TrajOpt (e.g. similarly to casting the convex hull over two discrete poses and check this for collisions)?

Levi-Armstrong, 2019-05-09:
In the case of motion planning I think casting the collision objects provides better information about how to move two moving objects out of collision.

BryceStevenWilley, 2019-05-09:
@j-petit yes, in this context, we're talking about integrating CCD by casting convex hulls. It's a stepping stone towards integrating TrajOpt, but it has its own independent uses as well.

Thanks for the input Levi! I do agree that it's useful for moving objects out of collision. However, there is a rotation error that happens when using CCD via CH, so we do need to be clear that it's slightly different than normal CCD, such as that discussed in https://github.com/ros-planning/moveit/issues/29#issuecomment-238347670

j-petit, 2019-05-09:
What is the difference between [collision_detection_fcl](https://github.com/ros-planning/moveit/blob/82fec7ba549494e281fb175cb0e9968037dffc86/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_world_fcl.h) and [collision_detection](https://github.com/ros-planning/moveit/blob/82fec7ba549494e281fb175cb0e9968037dffc86/moveit_core/collision_detection/include/moveit/collision_detection/collision_world.h#L154)? There seems to be some overlap in the header files I linked. I thought that all collision detection is processed through the structure given by FCL? Could someone clarify that for me?

rhaschke, 2019-05-09:
The collision checking is designed as a plugin: collision_detection defines the API, while collision_detection_fcl is (one) concrete implementation of this API using FCL. 
In principle, it should be possible to replace FCL with another collision checker, e.g. Bullet, but we are not aware of an alternative implementation of the API.

simonschmeisser, 2019-05-09:
There is a dummy one that always returns no collision ...

j-petit, 2019-05-13:
TrajOpt: For bringing CCD to MoveIt we don't need to solve the optimization problem described in the [paper](http://joschu.net/docs/trajopt-paper.pdf)? Am I correct in this conclusion?

BryceStevenWilley, 2019-05-13:
You are correct there. There is a lot of information in the paper about adapting CCD to be used by the optimization, that's only if you are using an optimization-based planner.

j-petit, 2019-05-14:
I have some questions concerning the support mapping as defined in the [TrajOpt paper](http://joschu.net/docs/trajopt-paper.pdf):
- Does the coordinate system in which the object is defined matter?
- When it is stated that the support mapping implicitly represents the object, does this mean that if we know the mappings result for all directions, we could reconstruct the object?
- For the support mapping of the convex hull, the formula (see picture below) is presented in the paper without any further explanation. I have some difficulties getting my head around the intuition behind it.
[image]

General issues:
- Do we want to support the CCD only for convex polytopes?
- For CCD, we need to calculate the signed distance and therefore compute the distance between two shapes via GJK and penetration depth via EPA. In Tesseract, Bullet is used for this. FCL uses implementations via libccd. Why is it not possible to keep using libccd? We always talked about (partially) integrating Bullet into MoveIt.

felixvd, 2019-05-14:
Caveat: I'm just sharing some intuition and corrections and additions are always welcome.

>     * When it is stated that the support mapping implicitly represents the object, does this mean that if we know the mappings result for all directions, we could reconstruct the object?

If it is convex, yes. There would be no unique mapping to flat surfaces, but those would be implicitly defined by the edges. The support mapping of a cube would return only points on the edges (or for 6 some directions be ill defined?), for example.

>     * For the support mapping of the convex hull, the formula (see picture below) is presented in the paper without any further explanation. I have some difficulties getting my head around the intuition behind it.
>       [image]

You can read [image] as a measure of distance, but it is still conveniently signed so that the equation works out. The term is positive when points are "behind" the origin and negative when the origin is "behind" the points when looking in the direction **v**.

I scribbled a really simple example down [here](https://user-images.githubusercontent.com/4535737/57699355-84fd3400-7692-11e9-8889-ee4323abc6df.png).

Another way to look at it: [image] is a point in space, s_C(\mathbf{v})\cdot \mathbf{v} projects the point along the direction 
[image], and 
[image] is the euclidean distance of that point to the origin. The term inside that norm can be used to compare the distance along the projected direction.

I feel like this is not perfectly intuitive either, but considering what the dot product means geometrically probably helps. Maybe someone has an easier way to put things.

> General issues:
> 
>     * Do we want to support the CCD only for convex polytopes?

It is my understanding that this method only works for convex shapes.

BryceStevenWilley, 2019-05-14:
> Does the coordinate system in which the object is defined matter?

Not sure how exactly to answer the question: for the support vectors, I don't think the coordinate system matters as long as both objects (both link placements for the convex hull, or the link and the object) are in the same system, i.e. the vector v points in the same direction for each object.

Felix gave some good answers for the support vector questions, don't think I could add much there.

> Do we want to support the CCD only for convex polytopes?

For the moment, yes. We could also consider being able to specify multiple convex polytopes per link as to approximate a nonconvex link.

> Why is it not possible to keep using libccd? We always talked about (partially) integrating Bullet into MoveIt

We should try to use both! Being able to use different collision checking plugins has never been well developed, as we really only have FCL working properly (spheres and distance fields don't work quite as well), and it's definitely interesting to get both work. From what I understand, Bullet might also have some performance benefits when compared to FCL. 
In addition, I've personally run into some [issues with FCL](https://github.com/flexible-collision-library/fcl/issues/221) and signed distance in the past. These issues might be fixed, but if they're not, it might be more straight-forward to just use the Bullet library instead of finding and fixing those issues in FCL itself. (Also small clarification, I believe Bullet uses `libccd` internally as well, at least according to [libccd](https://github.com/danfis/libccd)).

Levi-Armstrong, 2019-05-15:
> Does the coordinate system in which the object is defined matter?

As @BryceStevenWilley stated it does not matter as long as both are in the same coordinate system. That said, most of this is dictated by the library, like FCL and Bullet. In both these cases they are represented in a global coordinate system.

> Do we want to support the CCD only for convex polytopes?

I agree, this provides useful information and is significantly faster than triangle mesh to triangle mesh in most cases. The way both Bullet and FCL represent detailed meshes, are as a bounding volume hierarchy (BVH) of triangles. Then all other shapes supported are convex shapes and since a triangle is a convex shape it leveraged CCD.

>  Why is it not possible to keep using libccd? We always talked about (partially) integrating Bullet into MoveIt

When discussing collision checking you have two parts, Broad Phase Collision checking and Narrow Phase Collision checking. The Narrow Phase Collision checking is what libccd performs which is checking the two objects for collision. The Broad Phase Collision checking is the BVH check. In the both Bullet and FCL when you add a collision shape it gets added to global BVH. Also some collision shapes themselves are represented by BVH of convex shape. For example, triangle meshes are represented by a BVH of triangle shapes. A octomap is represented by a BVH of box shapes. Once all shapes have been added and locations set. The libraries perform two operation. First they check each collision shape against the global BVH and if the bounding box of the shape intersects the bounding box of the another shape it then and only then enters into the narrow phase collision check (This is the most computation expense operation).  By performing the Broad Phase check first you avoid doing  unnecessary Narrow Phase collision checks.

Bullet has rewritten libccd or written its own within its API and does not depend on libccd but may have in the past. In the case of FCL it depend only certain components of libccd but most of it has been rewritten to handle degenerate case in CCD.

I personally, would not recommend using libccd directly as it provides little value without the Broad Phase component. I would recommend, working directly with FCL and Bullet to resolve any issues found because to write your own you would run into the same degenerate case to be resolved. 

> We always talked about (partially) integrating Bullet into MoveIt.

I not sure what is meant by (partially) integrated, but bullet can be fully integrated into MoveIt assuming the issues are address described in my previous comment. I have fully integrated both within tesseract_collision which could be directly used within MoveIt with minor contact data translation. We will be releasing a new version of the core packages which includes tesseract_collision as ROS agnostic libraries to support integration like this. If MoveIt does not want to depend on tesseract_collision then I would propose copy the code directly into MoveIt and go from there.

felixvd, 2019-05-15:
Regarding @j-petit 's question about the support mapping of the convex hull:

> [image]

I took the time to draw this out because another friend of mine was also confused. The fundamental idea to grasp is just that the dot product projects vectors onto each other. In this case, just read "a dot product projects a point onto a vector". Take any two points in cartesian space:

[image]

If you multiply each point with the unit vector (1, 0), you project each point onto the x-axis. Trivially, that is the x-coordinate of each point. 

[image]

However, you can do this with any direction, and imagine a new axis, like this:

[image]

The points are projected onto the new axis in the same way. The dot product's result is the position on the axis. The opposite direction just turns the axis around:

[image]

Knowing this, it should become clear that the formula you quoted simply says "take whichever point is furthest in the direction v" or "choose the point that has the highest value on the axis v":

> [image]

PS: This also works for non-unit vectors, but they introduce distortion, so it [does not look as pretty](https://user-images.githubusercontent.com/4535737/57748017-f7155d80-7712-11e9-9ac9-510cb1cb8613.jpg).

j-petit, 2019-05-15:
Thanks for the explanations @BryceStevenWilley, @felixvd and @Levi-Armstrong!

j-petit, 2019-05-16:
#### Integrating Bullet into MoveIt
I am getting an overview what needs to be done in order to support Bullet for collision checking. Feel free to add / comment. My findings so far:

Interface implementations
- collision_detection::CollisionDetectorAllocator
- collision_detection::CollisionPlugin
- collision_detection::CollisionRobot
- collision_detection::CollisionWorld

General classes
- collision object wrapper which wraps Bullet collision objects (analog to collision_detection::FCLObject; in Tesseract CollisionObjectWrapper)
- collision manager analog to collision_detection::FCLManager (in Tesseract BulletDiscreteBVHManager and BulletCastBVHManager
- collision object geometry representation analog to collision_detection::FCLGeometry

BryceStevenWilley, 2019-05-16:
@j-petit, @felixvd, @davetcoleman, and I just had a call discussing this effort, and we came up with some milestones we want to hit. These will be reevaluated along the way, and, depending on how feasible these targets are, the dates might change.
* June 10th (3-4 weeks): a draft PR proof-of-concept making a Bullet collision checking plugin. This PR might be a bit messy, and will most likely change some CollisionPlugin interfaces. The idea is to quickly prove that it's possible to have a Bullet collision checker in MoveIt, without having make everything perfect. In this PR, we'd include CCD via convex hulls.
* July 8th (another 4 weeks): a ready to review PRs that (if necessary) change the CollisionPlugin/CollisionRobot/CollisionWorld interfaces, and also add Bullet as a CollisionPlugin. Majority of this work will be properly changing the Collision Plugin interface for the existing collision plugins, and including the robot and world in the same collision scene, as discussed [here](https://github.com/ros-planning/moveit/issues/1427#issuecomment-480203815) and [here](https://github.com/ros-planning/moveit/issues/1427#issuecomment-480211140). 
* July 29th (3 more weeks): Start efforts trying to add CCD via Convex hulls to FCL, so it as well could support TrajOpt and the additional convex-convex collisions.
* August 14th (Jens' last week with GSOC): Document collision plugins in depth, detailing how to write your own, and how to select which plugin is used at runtime.

j-petit, 2019-05-17:
#### High level understanding of collision detection in MoveIt:
Some passages are copied from the docstrings. See [comparison document](https://docs.google.com/spreadsheets/d/1eHsJTh1b1I0IClNrv1znwqhd5Xy9oJrZiahSMHwKz1U/edit?usp=sharing) for more details.

Each `PlanningScene` has a map member `PlanningScene::collision_` which contains possible multiple `CollisionDetectors`. `CollisionDetectors` bundle `CollisionWorld` and `CollisionRobot` of a `PlanningScene`. 

A collision detector type is specified with an allocator which is a subclass of `CollisionDetectorAllocator`. This identifies a combination of `CollisionWorld`/`CollisionRobot` which can be used together. A new `PlanningScene` contains an FCL collision detector. The member `PlanningScene::active_collision_` points to the active collision detector.

Two collision cases are distinguished in MoveIt. Collisions between robot-world and self-collisions/robot-robot. This is the motivation between having  `CollisionWorld` and `CollisionRobot`.

When collision checks are expected from the planning scene (planning_scene->checkCollision()), then the active collision detector and with its `CollisionWorld` and `CollisionRobot` is returned and then the checks are performed.

#### Open questions:
I don't fully understand what the purpose of the `CollisionDetectorAllocator` is. Is this simply a wrapper which provides functionality to handle the `CollisionWorld` / `CollisionRobot` pair?

BryceStevenWilley, 2019-05-19:
Solid summary.

> A new PlanningScene contains an FCL collision detector. 

This is something that I came across a while back: the call is [right here](https://github.com/ros-planning/moveit/blob/2d0fe65e8d98f5b846cbe59270977b168bd8710e/moveit_core/planning_scene/src/planning_scene.cpp#L173). A minor nit, but IMO, this should be a part of the PlanningScene constructor, and that input should default to the FCL detector, i.e. I think dependency injection is a better idea here. 

> I don't fully understand what the purpose of the CollisionDetectorAllocator is. Is this simply a wrapper which provides functionality to handle the CollisionWorld / CollisionRobot pair?

To the best of my knowledge, yes. In some cases, you could have a collision world that is represented by a distance field, and a collision robot that is a standard mesh based robot. Trying to collision check those two would be difficult, and general, programming every possible `CollisionRobot` to work with every other `CollisionWorld` would be intractable. So by always allocating these from the `CollisionDetectorAllocator`, you avoid those problems.

BryceStevenWilley, 2019-05-19:
During the call on Thursday, I mentioned that I'd give @j-petit some clear direction on where to start his first week of GSoC. Putting it here, so anyone can feel free to add comments.

A good deliverable that you could have for this would be to have (or at least have significant progress on) a simple bullet collision checker, something equivalent to Tesseract's [Bullet Discrete Simple Manager](https://github.com/ros-industrial-consortium/tesseract/blob/master/tesseract_collision/src/bullet/bullet_discrete_simple_manager.cpp). This will help you start simple, and you can also bootstrap the tests by reusing the [FCL tests](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection_fcl/test/test_fcl_collision_detection.cpp). Since this isn't adding new functionality to MoveIt, the majority of the development work will be porting tesseract's code here, and the biggest difference is that most of the object addition/removal/updates should go through the [notifyObjectChange](https://github.com/ros-planning/moveit/blob/2d0fe65e8d98f5b846cbe59270977b168bd8710e/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp#L258) function, as compared to Tesseract's individual add collision object, set collision object transform, etc. (question for other maintainers: Should that belong to the `CollisionWorld` class? I can imagine a lot of MoveIt infra relying on the fact that planning scene updates are passed to the collision world like this). I think the best place for you to start is to take another close look at how the collision manager is handled in [CollisionWorldFCL](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp), I imagine that the collision manager in Bullet would be updated in a very similar way.

Some other thoughts for the upcoming weeks:
* After adding a discrete collision detector, you can focus on reimplementing `checkRobotCollision` between two different states to use the convex hull casting, dealing with the convex tags and assumptions then, so you don't have to worry about that right off the bat.
* Once we have a working CCD collision checker, we can get some performance metrics between all of the current checkers on some varying scenes (empty, normal collision objects, and very dense). There's a pretty good script for this [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp) (which was very well hidden).
* Then, we can start experimenting with how to combine the Collision Robot and the Collision World, and compare and quantify any performance gains that we get out of that. Once we have the numbers, I think we'd have a good idea about how to proceed with major changes to the MoveIt API, if necessary.

j-petit, 2019-05-20:
> This is something that I came across a while back: the call is [right [...]
Should I open a separate issue / PR for discussing this?

----
For integrating Bullet, should I use the version modified for tesseract or the standard one provided by e.g.`apt-get install libbullet2.87` (Ubuntu 18.04)?

simonschmeisser, 2019-05-20:
Without having any further insight in changes between bullet2 and bullet3 I'd suggest using bullet3 as it has  a CUDA backend which might be fun. :wink: 

@BryceStevenWilley there was talk about modifying CollisionRobotFCL such that it exposes the whole collision hierarchy to FCL instead of doing n² individual queries. This would then give us broadphase collision checking. Do you think that would fit within the scope?

j-petit, 2019-05-20:
> Without having any further insight in changes between bullet2 and bullet3 I'd suggest using bullet3 as it has a CUDA backend which might be fun.

When I use `libbullet2.87`, this is [bullet3](https://github.com/bulletphysics/bullet3/releases/tag/2.87), right?

j-petit, 2019-05-20:
I looked into the environment representation / handling of MoveIt today. My understanding so far:

`WorldPtr collision_detection::CollisionWorld::world_` contains the environment representation. When instantiating `CollisionWorldFCL`, an observer is created which tracks any handling of objects in the environment representation. The observer has as a member the function pointer to a callback fcn. Any action which manipulates an object, notifies the observer. The observer can be notified through different functions like `collision_detection::World::notifyObserverAllObjects`, `collision_detection::World::notify(...)`, `collision_detection::World::notifyAll(...)` which all in the end invoke the callback which updates the specific to FCL CollisionWorld.fcl_objects_ representation of the environment as well as the FCLCollisionManager. This ensures that these representation are always up-to-date.

For the actual collision check, it suffices then to execute collide() of the FCL collision manager.

j-petit, 2019-05-21:
> During the call on Thursday, I mentioned that I'd give @j-petit some clear direction on where to start his first week of GSoC. Putting it here, so anyone can feel free to add comments.
>
> A good [...]

I am a bit unsure what is the best way to tackle this problem. My approach would be to more or less keep the `tesseract_bullet::BulletDiscreteSimpleManager` and replace the manager member of `CollisionWorldFCL` with it to create the new `CollisionWorldBullet`.

Another idea would be to not use the  `tesseract_bullet::BulletDiscreteSimpleManager` but to transfer its functionality directly into the new `CollisionWorldBullet`. I think this would be the cleaner solution / more efficient solution, but probably take longer. Any thoughts?

j-petit, 2019-05-21:
In tesseract, there is the `enum tesseract::tesseract_fcl::CollisionFilterGroups` with values `DefaultFilter |  StaticFilter |  KinematicFilter |  AllFilter` which for example stops the collision checking process. I don't understand its purpose and can't find any further documentation. The enum is used as member `tesseract::tesseract_fcl::CollisionObjectWrapper::m_collisionFilterGroup`. Maybe @Levi-Armstrong can give me a quick hint?

felixvd, 2019-05-22:
> For the actual collision check, it suffices then to execute collide() of the FCL collision manager. The robot itself is passed as an argument (fcl_object) which is constructed from `CollisionRobot` and the `RobotState`.

Does this collide() function check if the object/robot in the argument collides with the world? How does this work in Bullet? Is there a difference between CollisionRobot and an fcl_object? Open question to anyone.

BryceStevenWilley, 2019-05-23:
> I am a bit unsure what is the best way to tackle this problem.

Discussed offline, but I think the first example (keeping `tesseract_bullet::BulletDiscreteSimpleManager`) is the best way to go about this at the moment. We can refactor to not be dependent on tesseract later. 

> In tesseract, there is the enum tesseract::tesseract_fcl::CollisionFilterGroups with values DefaultFilter |  StaticFilter |  KinematicFilter |  AllFilter

My best guess here is that you can avoid collision-checking the static scene against the static scene, as mentioned in [this comment](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/include/tesseract_collision/fcl/fcl_utils.h#L183). 

> Does this collide() function check if the object/robot in the argument collides with the world?

This loops over each robot link/geometry and checks it against the already constructed collision world, as seen [here](Does this collide() function check if the object/robot in the argument collides with the world?).  As for bullet, the major difference seems to be just the difference we've discussed previously: that it does the robot self collision checks and world checks at the same time instead of separately.

j-petit, 2019-05-23:
> Does this collide() function check if the object/robot in the argument collides with the world? How does this work in Bullet? 

Do you mean in Tesseract? If yes: when a new KDLenv is created with a robot, a collision manager is added and all the robot links are added to the manager (see [here](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_ros/src/kdl/kdl_env.cpp#L637)). When now a collision check should be performed through [contactTest()](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/src/bullet/bullet_discrete_simple_manager.cpp#L201) of the manager 

> Is there a difference between CollisionRobot and an fcl_object? Open question to anyone.

j-petit, 2019-05-23:
> Does this collide() function check if the object/robot in the argument collides with the world? How does this work in Bullet? 

Do you mean in Tesseract? If yes: when a new KDLenv is created with a robot, a collision manager is added and all the robot links are added to the manager (see [here](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_ros/src/kdl/kdl_env.cpp#L637)). The coordinate transformations of the links within the manager are updated when KDLEnv::setState() is used to change the robot state in the environment. When now a collision check should be performed through [contactTest()](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/src/bullet/bullet_discrete_simple_manager.cpp#L201) of the manager, there are two nested for loops for each pair of collision objects in the `cows_` vector of the manager. Bullet is used then to perform the actual check of each pair.

> Is there a difference between CollisionRobot and an fcl_object? Open question to anyone.

Yes, [fcl_object](http://gamma.cs.unc.edu/FCL/fcl_docs/webpage/generated/classfcl_1_1CollisionObject.html) is a format used by FCL to directly interact with the FCL manager. It represents an atomic instance of an object which can be collision checked via FCL. `CollisionRobot` of MoveIt on the other hand contains all the information necessary for collision checks of a single robot. This includes checking self-collisions and against other robots.

BryceStevenWilley, 2019-05-24:
@isucan generously joined us on our weekly sync and provided some reasoning for the separation of the `CollisionRobot` and the `CollisionWorld` classes.
* It's to your advantage if you can collision check several poses of the robot at the same time
* With a separate world and robot, to check multiple poses in parallel, you simply have to copy and change the robot, not the whole world
  * This (and much of the design in MoveIt) was made with the assumption that the collision world would be large and expensive to copy.
    * True in some cases (point clouds, maybe some complex octomaps), but not for others (simple geometry worlds would be easy)
  * Also made it easy to do some-to-all collision checks of the world (don't have to bother with checking the static world against itself)
    * Bullet [should handle this](https://github.com/ros-planning/moveit/issues/1427#issuecomment-495096436) too
* OMPL and parts of MoveIt use this parallel checking of the robot state against the world, naively combining the world and the robot classes might break that
* Certain functions weren't implemented in Bullet when Ioan started

* Overall sentiment: it should be possible to simply copy the `CollisionWorld` along with the `CollisionRobot`, but it might be more expensive to do so, especially for large point cloud based worlds. We should get some metrics on how big the world needs to be before the optimizations from combining the world and the robot are outweighed by the expense of copying the world for parallel state checks.

j-petit, 2019-05-24:
Following roadmap for the next days:
- [x] add tesseract manger as member of `CollisionWorld` and `CollisionRobot`
- [x] when instantiating new `CollisionRobot` add robot links to the manager
- [ ] can update the robot link state in the tesseract manager
- [ ] successful self-collision checks
- [ ] adding objects to `CollisionWorld` adds them to the manager
- [ ] transfer of collision objects from one manager to the other
- [ ] checking robot-world collisions
- [ ] passing test scenario for FCL but now using bullet

Levi-Armstrong, 2019-05-25:
>> With a separate world and robot, to check multiple poses in parallel, you simply have to copy and change the robot, not the whole world

This may have been the case at the time of implementation, but now in the case of FCL and Bullet the core collision shape should never need to be copied. In these libraries the collision shapes are stored as a shared pointer and they leverage wrappers around these to store the location and other data that needs to change. This allows you to use a single collision shape multiple times within an environment were the pointer is only copied. 

>>    This (and much of the design in MoveIt) was made with the assumption that the collision world would be large and expensive to copy.
        True in some cases (point clouds, maybe some complex octomaps), but not for others (simple geometry worlds would be easy)

The coping of the collision environment should be negligible if done properly because it is mostly copying shared pointers. Also because of the way things get loaded currently the majority of static objects are within the urdf and considered part of the robot and the world is only used for PointCloud data in my experience. As a result there are a lot of unnecessary collision checks between static objects. 

In the case of Bullet it leverages masks to identify static and kinematic objects and allows user defined to prevent unnecessary collision checks which I believe is a better way to approach this and provided more flexibility to the developer by having the ability to add custom masks. I have implemented this same concept native to Bullet for FCL within the tesseract fcl contact manager. Also as the scene changes it is much easier to manage updating masks than moving collision objects between say the collision robot and world to prevent unnecessary static object collision checks.

>>    Also made it easy to do some-to-all collision checks of the world (don't have to bother with checking the static world against itself)

I believe a more elegant way of handling this would be to support masking instead of two separate classes/environment. This separation could then be handled directly within a single environment which is how Bullet handle this.

>> We should get some metrics on how big the world needs to be before the optimizations from combining the world and the robot are outweighed by the expense of copying the world for parallel state checks.

What would you consider a large world environment. I could easily test it and provide metrics on the what it takes to clone the environment.

j-petit, 2019-05-29:
I have problems understanding the `ContactTestType enum` in Tesseract. As far as I know, the value of the enum is only concerned with the number of contacts between a _single_ pair of objects. However, what I then do not understand is why `cdata.done` is set [here](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/include/tesseract_collision/contact_checker_common.h#L107) which [breaks](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/src/bullet/bullet_discrete_simple_manager.cpp#L258) the N^2 collision checking nested loop as soon as the first contact is found (for all pairs of objects). This is somewhat contradictory.

Another issue is how a single call to `algorithm->processCollision(...)` is even able to produce multiple contact points? I cannot find repeated calls to the callback which adds to the results vector in the bullet function (for example [here](https://pybullet.org/Bullet/BulletFull/classbtConvexConvexAlgorithm.html#ab566631355a17f7a4bcbddf9454eb9e5)). Or is this for the case if we use compound collision shapes which uses this [algorithm](https://pybullet.org/Bullet/BulletFull/classbtCompoundCompoundCollisionAlgorithm.html)?

Other related question I am still struggling with is what kind of objects we actually want to check for collision (only convex polytopes?) and what this implies for the number of contact points. What exactly constitutes a contact point? I see that if the objects only touch we have a real single contact point, but we cannot generalize this case. For example in the picture below, if we assume to triangles as our shapes, are then the blue or the green points the multiple contact points? Or something completely different?
[image]

Bullet returns [two points](https://pybullet.org/Bullet/BulletFull/classbtManifoldPoint.html#a07ab418bffd8dba065a550912be00013) for each contact (this makes sense to me as it also returns the distance between those two points, they represent either the closest points are the ones when calculating the minimum translation distance). FCL on the other hand only reports a [single](https://flexible-collision-library.github.io/d8/d61/structfcl_1_1ContactPoint.html) contact point back. Is this the midpoint on the line connecting the two points for the minimum translation distance?

BryceStevenWilley, 2019-05-30:
>  As far as I know, the value of the enum is only concerned with the number of contacts between a single pair of objects.

That might be the good assumption, but [the enum comments](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_core/include/tesseract_core/basic_types.h#L200) say that the `FIRST` enum is for any pair of objects. This is simply to determine a boolean if any collisions are present in the scene. 

>  what kind of objects we actually want to check for collision (only convex polytopes?)

Yep. We could use non-convex polytopes, but they are much harder/slower to find the collision points, and it should be faster to decompose links into convex polytopes (discussion [here](https://github.com/ros-planning/moveit/issues/742)) and check those. You'll have to design a way to handle this.

> what this implies for the number of contact points

I'd take a look at [this article](http://www.dyn4j.org/2010/05/epa-expanding-polytope-algorithm/) about EPA, which is the algorithm Bullet and FCL use to find collision points.  (Not 100% sure about this, but) I don't believe that these algorithms interpolate points along edges or faces (i.e. the green points in the image above, as they aren't points on the original triangle), so in the above example the only 2 collision points to be found are the blue points. Specifically for TrajOpt, we'll only care about the deepest collision point.

> FCL on the other hand only reports a single contact point back.

This should be the deepest point of contact (one of the blue points in the image above). From that point, the normal vector, and the penetration distance, you can recreate the point on the surface of the object being penetrated.

j-petit, 2019-05-30:
> That might be the good assumption, but [the enum [...]
That is also what I thought but then on the other hand the breaking of the loop only happens in the [inner loop](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/src/bullet/bullet_discrete_simple_manager.cpp#L258) which leads to additional collision checks being performed (one per collision object which hasn't been selected in the outer loop). Maybe this was not intended and is a bug in tesseract?

> 
> > FCL on the other hand only reports a single contact point back.
> 
> This should be the deepest point of contact (one of the blue points in the image above). From that point, the normal vector, and the penetration distance, you can recreate the point on the surface of the object being penetrated.

With single contact point I mean that for each contact (FCL can return multiple per object pair as Bullet) we only get one coordinate back (Bullet returns two).

Levi-Armstrong, 2019-05-30:
> That is also what I thought but then on the other hand the breaking of the loop only happens in the inner loop which leads to additional collision checks being performed (one per collision object which hasn't been selected in the outer loop). Maybe this was not intended and is a bug in tesseract?

If using enum FRIST, After the first contact is found it sets the `done` flag which gets checked at the outer loop. It should exit after the first contact is found. 

>With single contact point I mean that for each contact (FCL can return multiple per object pair as Bullet) we only get one coordinate back (Bullet returns two).

With Bullet, if you are checking two convex shapes you will get a single contact results between the objects which contains two contact points (One for each object). If you are checking a convex shape to triangle mesh you will get multiple contact results. Each one will be for a single triangle that is within your contact threshold. This is the same for triangle mesh to triangle mesh. This is more or less the same as FLC. In FCL you get a single contact point which if I remember correctly it is the point projected along the minimum translation vector to the surface of one of the shapes. Then provided the normal(translation vector) and distance you can determine the second contact point from FCL and is what is happening inside the FCL implementation.

j-petit, 2019-05-30:
Thanks @Levi-Armstrong, then I think this is indeed a bug in the [bullet_discrete_simple_manager.cpp](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/src/bullet/bullet_discrete_simple_manager.cpp#L258) where the check is only in the inner loop. I took a look at the bullet_cast_simple_manager, there we have two checks, one in the inner, another in the outer loop.

j-petit, 2019-06-03:
I did a first quick benchmark of the simple bullet manager I have implemented into MoveIt (see last two commits in PR #1484). I haven't adapted running the bullet collision detection in multiple threads, but benchmarking a single thread resulted in:
[image]
Run this [launchfile](moveit_ros/planning/launch/collision_checker_compare.launch) to reproduce.

rhaschke, 2019-06-03:
How does this compare to FCL collision checking?

BryceStevenWilley, 2019-06-04:
@rhaschke The FCL results are at the bottom of the image above: Bullet does 338 checks per ms and FCL does 999 checks per ms.  The results can be explained by the fact that the state used is always collision free (results can be cached pretty well) and that FCL is using a bounding-volume-hierarchy, which is faster on collision free queries. Bullet/Tesseract handles BVH's pretty well, I created a [new card](https://github.com/orgs/ros-planning/projects/2#card-22337469) to track the fact that Bullet will eventually have that to be competitive to FCL.

It's a good start, but we definitely need more variation. To make these benchmarks more useful, we need a variant that does random states (probably on a more densely clutter scene than the default, which would just have self-collisions) that reports back time and percent of states in collision, and one that does a state always in collision.  Shouldn't take long to add those variants.

simonschmeisser, 2019-06-04:
the number for fcl has one digit less, so it's 99 not 999

BryceStevenWilley, 2019-06-04:
Thanks for the correction @simonschmeisser, completely misread that.

[43 more replies]