moveit/moveit #1429: Planner is unable to find simple UR10e movements

natharos, 2019-04-10:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build. According to "apt-cache policy ros-kinetic-moveit-planners" i have version 0.9.15-0xenial-20190320-235453-0800
I also use the universal_robot, industrial_core and ur_modern_driver packages.

### Steps to reproduce
I have an UR10e robot and am trying to plan and execute a Cartesian path of 10 cm. I use the standard OMPL RRTConnectkConfigDefault planner and have set the tolerances to 0.01 (joint), 0.01 (position) and 1.0 (orientation). The robot is not near any singularities. There are no collision objects in the scene. The robot is not in collision with itself. I try to move the end-effector approximately 10 cm downwards, then 10cm forwards and then back to the starting position. By using the interface provided by Universal Robots I have verified that the movement is possible.

### Expected behaviour
I would expect the planner to find a path in a few iterations.

### Actual behaviour
When a trajectory is found, it is executed properly. But the planner takes very long (over 100 iterations) to find a trajectory, and sometimes does not find a trajectory at all. 

Should I use another planner? Or do I have to change the tolerances even more (they originally were 0.0001, 0.0001, 0.001)? Or is there anything else that I should take into account?

### Backtrace or Console output
Outputs like:
[ INFO] [1554906650.685948567]: Computed Cartesian path with 24 points (followed 27.272727% of requested trajectory)

felixvd, 2019-04-10:
Could you post the code to reproduce the issue? Does it occur at different poses and orientations?

natharos, 2019-04-11:
The code can be found below. Another strange thing that occurred this morning is that when I run this script in the folder linked to my git repository, the path planner often fails while it tries to use 30-300 interpolation points, while when I run this script in a local folder with exact the same source code, it uses approx 900 interpolation points and immediately finds the path... I did a clean build (emptied the devel and build map) for both and did run the 'source devel/setup.bash'. The source code (both my own code and the packages I use) is exactly the same and compiled and run on the same laptop, but still somehow the behaviour is different. Where in the MoveIt library does ROS decide how many interpolation points are used?

```
#!/usr/bin/env python

import sys
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
from trajectory_msgs.msg import *

from std_msgs.msg import String

from cobot.srv import *
from ur_msgs.msg import RobotModeDataMsg
from geometry_msgs.msg import Pose
from copy import deepcopy

# PICK SERVICE SERVER----------------------------
def handle_pick(req):

	print "pick"
	pick_pose = copy.deepcopy(req.pose)

	# Always point hand backward
	pick_pose.orientation.x = 0
	pick_pose.orientation.y = 0
	pick_pose.orientation.z = 0
	pick_pose.orientation.w = 1

	waypoints = []

	if (pick_pose.position.y > 0):
		y_mid = y_right
	else:
		y_mid = y_left

	# Start with current pose
	cur_pose = group.get_current_pose().pose
	waypoints.append(cur_pose)
	print "Added waypoint 1"

	# Move to belt
	belt_pose = copy.deepcopy(pick_pose)
	belt_pose.position.x -= x_offset
	belt_pose.position.y = y_mid
	belt_pose.position.z = z_high 
	waypoints.append(copy.deepcopy(belt_pose))
	print "Added waypoint 2"

	# Move down, in front of the pot
	front_pot_pose = copy.deepcopy(pick_pose)
	front_pot_pose.position.x -= x_offset
	front_pot_pose.position.z = z_low
	waypoints.append(copy.deepcopy(front_pot_pose))
	print "Added waypoint 3"

	# Move in
	goal_pose = copy.deepcopy(pick_pose)
	goal_pose.position.z = z_low
	waypoints.append(copy.deepcopy(goal_pose))
	print "Added waypoint 4"

	# Lift it up
	up_pot_pose = copy.deepcopy(pick_pose)
	up_pot_pose.position.z = z_high
	waypoints.append(copy.deepcopy(up_pot_pose))
	print "Added waypoint 5"
[... 141 lines cut ...]
	print "============ Printing robot state"
	print robot.get_current_state()
	print ""

	print "Old tolerances (joint, pos., or.)"
	print(group.get_goal_joint_tolerance())
	print(group.get_goal_position_tolerance())
	print(group.get_goal_orientation_tolerance())

	group.set_goal_joint_tolerance(0.001)
	group.set_goal_position_tolerance(0.001)
	group.set_goal_orientation_tolerance(1.0)

	print "New tolerances (joint, pos., or.)"
	print(group.get_goal_joint_tolerance())
	print(group.get_goal_position_tolerance())
	print(group.get_goal_orientation_tolerance())

	while not rospy.is_shutdown():
		rate.sleep()
    # end while

	rospy.spin()
# end main
```

natharos, 2019-04-15:
I found the fix, apparently a certain combination of the max_eef_step and the jump factor for the Cartesian made the generation of the trajectory possible.

closed 2019-04-15 by natharos