moveit/moveit #1430: How to set inactive move group position

tsijs, 2019-04-10:
### Description

I made some custom planning plugins, which I want to use in the moveit benchmarking suite. It all works quite nicely, but the plans always fail due to collisions with the right (inactive) planning group. See [1]
I could set all values in the collision matrix between the inactive and active group to 0, but that seems a hacky way to solve this. I would prefer a method to be able to set the inactive planning group to a static position which is what is done somewhere in the ompl planners, but I cannot find where.
I looked at the code of the plugins, but I could not find what code sets the other planning groups to their static positions. 

[1]
[image]

felixvd, 2019-04-19:
1) What do you mean by "inactive"? I don't think you want to deactive collisions either. The right arm's planning group might not be involved in planning for the left arm, but surely the left arm can still collide with it.

2) I don't know at what level of planning you are working on. If you are using the `move_group` interface, the left arm's planning group should be updated by the `PlanningSceneMonitor` automatically. When you start the plan, the current position is taken from there.

3) If you only want to change the state of the left arm in your local copy of the planner (although you should use its actual pose if you don't want to collide with it), the `setJointGroupPositions` function in `robot_state.cpp` looks the most appropriate to me.

Caveat: I hope someone corrects me if I am wrong.

PS: Consider a ROS answers post about this if it does not end up being an interface problem.

tsijs, 2019-04-20:
Thank you for taking the time to answer this. 
1. With inactive I mean "not being planned for" and in my case, not relevant (see the picture, the current state is stretched out to the back, so it is not in the way of the right arm. Therefore turning off collisions on all but the first joint is a ok work-around for me now. 

2. I am indeed using the move_group interface I think (the gui in this case which uses it I think?), and the automatic updating happens correctly when using the ompl or chomp plugin, but not when using my own plugins. 

3. I tried setJointGroupPositions on the start state of the request, but that does not seem to influence the eventual position shown in rviz, which is just zeros. Maybe I don't understand this point good enough to try it properly. Could you maybe explain a little further?

felixvd, 2019-04-21:
1. If you are trying to reduce calculation cost by setting the left arm as "inactive", you should not do that. 

2. If you are trying to get it out of the way because it is in an incorrect (not the actual) position in your planning plugin, then there is probably something wrong with your custom plugin. 

3. Setting the joint group positions directly like this seems like a dirty workaround. Try to make sure that your planner correctly retrieves the current state of the environment and robot instead.

I can't help with the details, but you should check how the other plugins update the current state of the rest of the robot model and the environment. I would look for a point where you may need to update the PlanningScene, which contains the RobotModel (states of the links in your SRDF) and the collision_world (collision objects, octomaps etc. added during execution).

tsijs, 2019-04-22:
1. The left arm is not in the scope of my current project, hence it does not hurt to set it "inactive", since it physically actually is inactive. I cannot see why that would hurt. 
2. This is the case here. I know there probably should be something wrong with my plugin, I just cannot find what... 
3. I fully agree, I tried that before I made this issue, but I didn't find the combination of lines of code that did the trick. There are a few things that, if I removed them, resulted in the state not getting updated properly anymore, but that didn't help that much. 
Anyway, I don't have time to investigate this further for now, but when I do I will try again I guess. Thank you for your help anyway. If I don't succeed I will make a ros answers post as you suggested. 

I will close this topic now.

closed 2019-04-22 by tsijs

felixvd, 2019-04-23:
Just to answer "why that would hurt": If you disable collision checking between move_groups temporarily, you run the risk of forgetting or not being able to turn them back on. Your planner might crash before collisions are turned back on, the move_group would remain altered, and the next action could result in a catastrophic collision. It's just a high-risk thing to do that I wouldn't recommend practicing even once.

Good luck!

gavanderhoorn, 2019-04-23:
@tsijs wrote:
> The left arm is not in the scope of my current project, hence it does not hurt to set it "inactive", since it physically actually is inactive. I cannot see why that would hurt.

can't you just remove the planning group for the left arm?

tsijs, 2019-04-23:
Euh well, that's the thing. It was never in the move group to begin with, but the planner still sets the position of the other movegroup somehow.

gavanderhoorn, 2019-04-23:
If there is no group defined in the srdf, it can't be a group.

There could still be joints and links, but not a group.

Which moveit config are you using?

tsijs, 2019-04-23:
> If there is no group defined in the srdf, it can't be a group.
> 
> There could still be joints and links, but not a group.
> 
> Which moveit config are you using?

My own, but it is heavily based on Yoshua's old one. 

My choice of words was perhaps a bit too vague in the last comment:
- there are 5 move groups, one for every arm, one for every gripper and one for both arms. 
- I plan for just the right arm with the gui (or benchmarking tool). 
- the left arm position in rviz, as well as internally in moveit (since the collision checker finds collisions) changes after planning. 

> Just to answer "why that would hurt": If you disable collision checking between move_groups temporarily, you run the risk of forgetting or not being able to turn them back on. Your planner might crash before collisions are turned back on, the move_group would remain altered, and the next action could result in a catastrophic collision. It's just a high-risk thing to do that I wouldn't recommend practicing, before it becomes a bad habit.
> 
> Good luck!

- I agree on a conceptual level, but disagree on a pragmatic level;)