moveit/moveit #1432: Propose to create a place to hold advanced applications of MoveIt

RoboticsYY, 2019-04-11:
Following up the discussion in [MoveIt meeting july-25th-2018](https://discourse.ros.org/t/moveit-maintainer-meeting-recap-july-25th-2018/5504), we released the [ROS2 Grasp Library](https://discourse.ros.org/t/ros2-grasp-library-moveit-openvino-initial-release-0-4-0/8285) these days. This package enables Intel OpenVINO toolkit to accelerate GPD and provides MoveIt interface. To comply with our commitment, we also created a demo with a UR5 robot working with the ROS2 Grasp Library and MoveIt. The demo can be launched in both simulation and real execution, which intends to show how to use them together with the correct configuration. However, we didn’t find a proper place in MoveIt to add this demo and its tutorial.

First to my thought, MoveIt tutorial is an option for this. But to my understanding, MoveIt tutorial aims to provide tutorials for MoveIt API and tools with a consistent MoveIt Configure Package, which seems not proper to hold advanced applications, that may use different robot arm configurations or contain some usage of external code such as perception functions.

Therefore, we propose to add a repo under ros_planning to hold the code of or links to advanced industrial robot applications developed on MoveIt. This could bring some great advantages.

One good reason is to provide another way to cover MoveIt features. MoveIt is a fast evolving project, as @davetcoleman indicated (https://github.com/ros-planning/moveit/issues/1284), the MoveIt tutorial may be difficult to maintain and easily “become out of date”. And the use cases of MoveIt tutorial just cover small subset of MoveIt API. From my experience, I have witnessed several groups dropped MoveIt to use robot factory control interface, just because they don’t know how to deal with the situation, when a bad goal IK result causes the robot arm motions flipping whole around. I believe many features in MoveIt could resolve this, but it is not obvious for beginners. Sometimes even though users are familiar with the MoveIt tutorial, it is still difficult for them to use MoveIt. Here is a recent [example](https://discourse.ros.org/t/workspace-for-testing-moveit/8617). However, advanced applications that aim to solve real world problems can have good coverage of MoveIt features. And it is natural for people to see something really works before exploring deep into the details. 

Another advantage is that this can enhance widespread acceptance of MoveIt by demonstrating the extensibility of the MoveIt API. MoveIt is a flexible motion planning framework. There are many techniques that can be well integrated with MoveIt, such as the grasp planning and object detection for visual manipulation, or software architecture and hardware acceleration for motion planning and perception AI. All of these are not necessary to go into MoveIt, because MoveIt is already a large project. However, there is a lack of information to connect them well with MoveIt. And I believe there are many groups or individuals who are developing applications around these topics. If there is a place where users can easily share, explore and try some advanced applications, the advantage of using MoveIt would be more convincing. 

Along with the advantages, there could be some practical issues that make the project difficult to maintain. For example, each application may have a different MoveIt Config package due to the different hardware used. It may be difficult to find a common standard for testing the application execution. Some hardware used in the applications may not be replaceable or accessible to the users. In practice of an internal project called “MoveIt App Zoo” of our group, we tried to reduce the maintenance burden by designing some rules on how to add new applications to the repo:

1.	Each application should be able to run in simulation without hardware. This requires the inclusion of proper sensors data used in the application, which can a recorded rosbag or in some specific format. 
2.	Each application should provide a tutorial that follows the tutorial template, which consists of an overview description, hardware and software dependencies, environment setup and build (application build option can be switch on/off), application launch procedure and options, and the expected outputs.
3.	It should be possible to customize the application to new scenarios. The clear instructions should be included in a customization notes. 

Any comments or thoughts?

sharronliu, 2019-04-14:
>     1. Each application should be able to run in simulation without hardware. This requires the inclusion of proper sensors data used in the application, which can be a recorded rosbag or data in some specific format.

I like this. Apps running in simulation tools and fake controllers is good recommendation for users to replicate and view the outputs of the apps for evaluation purpose. Also it enables the automatic testing of apps in simulation for continuous integration purpose.

davetcoleman, 2019-04-15:
Thanks for this write-up @RoboticsYY. Some potential actions we could take:

- Make the ``ros-planning`` repo you suggested, perhaps named moveit_example_apps
- Add a new page to moveit.ros.org that indexes these "Example Applications"
- Make mention of this other tutorial resource from within the moveit_tutorials

Still, some of the work you all have done I think could be integrated back into the main tutorials, also. I encourage you to still think about that, as well.

sharronliu, 2019-04-16:
> Still, some of the work you all have done I think could be integrated back into the main tutorials, also. I encourage you to still think about that, as well.

Dave, sure will update the main tutorials, along with the above three actions.

davetcoleman, 2019-04-25:
@RoboticsYY are comfortable updating the moveit website? I'd like a new menu item for "Applications" like this:

[image]

And add that to the footer as well.

From there we'd have a page similar to the "Related Projects", except it would be Applications:

[image]

We could also add a final tutorial page in our current moveit_tutorials that redirects users to this advanced applications page.

RoboticsYY, 2019-04-26:
Yes, I can draft this new page on MoveIt website, and later sync with your team to add the first application example and integrate the application tutorial back to moveit_tutorials.

Best regards,
Yu

[image]<https://user-images.githubusercontent.com/561060/56751700-ed1bc280-6743-11e9-8fae-3e0da28318e5.png>

[image]<https://user-images.githubusercontent.com/561060/56751779-150b2600-6744-11e9-93a3-4d0343474d20.png>