moveit/moveit #1435: jog_arm integration test does not pass Travis CI

AndyZe, 2019-04-12:
The integration test for jog_arm within moveit_experimental does not pass Travis CI

From the comment here: https://github.com/ros-planning/moveit/pull/1360#discussion_r275058292

AndyZe, 2019-04-24:
My guess is, ROS_MASTER_URI and ROS_HOSTNAME aren't set in travis.yml. I found this [example](https://github.com/robotology/yarp/blob/master/.travis.yml) 

```
# ROS-related env variables. Used only by integration tests
    - ROS_HOSTNAME=localhost
    - ROS_MASTER_URI=http://localhost:11311
```

jschleicher, 2019-05-14:
I could trace the test failure down to an assertion in `decelerateForSingularity`.
After refactoring the test case into a rostest using python unittest as suggested in https://github.com/ros-planning/moveit/pull/1447#issuecomment-486428996
I'm building the branch https://github.com/jschleicher/moveit/commit/9e80b1de89982a95e6ac3febda991747373ef946

Starting from the docker container and the steps described in [moveit_ci](https://github.com/ros-planning/moveit_ci) I can build the package using catkin in Release mode and experience the following error / backtrace locally as well:

```
catkin config --extend /opt/ros/melodic/ --install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS_RELEASE="-O3" $CMAKE_ARGS
catkin build
cd src/moveit/moveit_experimental/jog_arm/
catkin run_tests --this --no-deps

Output:

...
You can start planning now!

[ INFO] [1557820770.342187963]: Ready to take commands for planning group panda_arm.
[ INFO] [1557820770.346702473]: Waiting for first joint msg.
[ INFO] [1557820770.660960942]: Received first joint msg.
[ INFO] [1557820770.661031260]: Waiting for first command msg.
[ INFO] [1557820770.661122783]: Received first joint msg.
[ INFO] [1557820770.661187218]: Waiting for first command msg.
[ INFO] [1557820773.593100741]: Received first command msg.
[ INFO] [1557820773.593206300]: Received first command msg.
[ INFO] [1557820773.593310148]: Waiting for low-pass filters to stabilize.
[ERROR] [1557820773.711824582]: Command joint name unknown.
[ WARN] [1557820773.713062169]: nan in velocity filter
jog_arm_server: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:425: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
[Testcase: testjog_arm_integration_test] ... ok

[ROSTEST]-----------------------------------------------------------------------

[moveit_experimental.rosunit-jog_arm_integration_test/test_jog_arm_generates_joint_trajectory_when_joint_jog_command_is_received][FAILURE]
0 not greater than 97.0
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/home/jschleicher/moveit_ws/src/moveit/moveit_experimental/jog_arm/test/integration/test_jog_arm_integration.py", line 73, in test_jog_arm_generates_joint_trajectory_when_joint_jog_command_is_received
    self.assertGreater( len(received), test_duration/publish_period - 3)
  File "/usr/lib/python2.7/unittest/case.py", line 943, in assertGreater
    self.fail(self._formatMessage(msg, standardMsg))
  File "/usr/lib/python2.7/unittest/case.py", line 410, in fail
    raise self.failureException(msg)
--------------------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1
...
```

Running inside gdb the backtrace is:
```
[ WARN] [1557743816.475590725]: nan in velocity filter
#2  0x00007ffff49df39a in __assert_fail_base (fmt=0x7ffff4b667d8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x5555555bbc6d "index >= 0 && index < size()", 
    file=file@entry=0x5555555bb7e8 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=line@entry=425, 
    function=function@entry=0x5555555bdb40 <Eigen::DenseCoeffsBase<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 1>::operator()(long)::__PRETTY_FUNCTION__> "Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = doub"...) at assert.c:92
#3  0x00007ffff49df412 in __GI___assert_fail (assertion=0x5555555bbc6d "index >= 0 && index < size()", 
    file=0x5555555bb7e8 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=425, 
    function=0x5555555bdb40 <Eigen::DenseCoeffsBase<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 1>::operator()(long)::__PRETTY_FUNCTION__> "Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = doub"...) at assert.c:101
#4  0x000055555558a087 in jog_arm::JogCalcs::decelerateForSingularity(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, Eigen::JacobiSVD<Eigen::Matrix<double, -1, -1, 0, -1, -1>, 2> const&) ()
#5  0x000055555558dae3 in jog_arm::JogCalcs::cartesianJogCalcs(geometry_msgs::TwistStamped_<std::allocator<void> >&, jog_arm::JogArmShared&, pthread_mutex_t&) ()
#6  0x000055555559042f in jog_arm::JogCalcs::JogCalcs(jog_arm::JogArmParameters, jog_arm::JogArmShared&, pthread_mutex_t&, std::shared_ptr<robot_model_loader::RobotModelLoader> const&) ()
#7  0x00005555555b0879 in jog_arm::JogROSInterface::startJogCalcThread() ()
```

AndyZe, 2019-05-20:
Fixed with #1466

closed 2019-05-20 by AndyZe