moveit/moveit #1436: How to creat FollowJointTrajectory

pawelwisniewski9603, 2019-04-14:
Hi Everyone!
I`m new in ros and i`m looking for help. I have made a URDF file for my robot - kuka kr150 r3100 prime - it works fine, next i wanted to try MoveIt. I instaled - works fine, made a movelt package for my robot. When i launch demo.launch - everything is ok.  Next, with this tutorial:
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/controller_configuration/controller_configuration_tutorial.html
i made controllers.yaml and rest of files. But they wrote on the top of tutorial:
"In this section, we will walk through configuring MoveIt! with the controllers on your robot. We will assume that your robot offers a **FollowJointTrajectory** action service for the arms on your robot" and i don`t have it(FollowJointTrajectory) and i dont have any idea how to create it.
(and sorry for my english)
I would be grateful for Your help!
Paweł

* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04

RoboticsYY, 2019-04-15:
This usually means that your robot's ROS driver has to function as a ROS [action](http://wiki.ros.org/actionlib)  server. The message format of the action should be [control_msgs/FollowJointTrajectory Action](http://docs.ros.org/api/control_msgs/html/action/FollowJointTrajectory.html). 

Maybe you can try [kuka_experimental](https://github.com/ros-industrial/kuka_experimental).

pawelwisniewski9603, 2019-04-15:
Hi!
Thanks for Your answer!
I read about Ros Action Service so I tried to do my owm.
I created new package - ros_action_service and copied action file:
http://docs.ros.org/electric/api/control_msgs/html/action/FollowJointTrajectory.html
Next, i tried to do full_service_action.cpp :
#include <ros/ros.h>
#include <actionlib/server/simple_action_server.h>
#include <control_msgs/FollowJointTrajectoryAction.h>
#include <trajectory_msgs/JointTrajectory.h>

class RobotTrajectoryFollower
{
protected:

  ros::NodeHandle nh_;
  // NodeHandle instance must be created before this line. Otherwise
strange error may occur.

actionlib::SimpleActionServer<ros_action_server::FollowJointTrajectoryAction>
as_;
  std::string action_name_;

public:

  RobotTrajectoryFollower(std::string name) :
    as_(nh_, name, false),
    action_name_(name)
  {
    //Register callback functions:
    as_.registerGoalCallback(boost::bind(&RobotTrajectoryFollower::goalCB,
this));

as_.registerPreemptCallback(boost::bind(&RobotTrajectoryFollower::preemptCB,
this));

    as_.start();
  }

  ~RobotTrajectoryFollower(void)//Destructor
  {
  }

  void goalCB()
  {
    // accept the new goal
    //goal_ = as_.acceptNewGoal()->samples;
  }

  void preemptCB()
  {
    ROS_INFO("%s: Preempted", action_name_.c_str());
    // set the action state to preempted
    as_.setPreempted();
  }
};

int main(int argc, char** argv)
{
  ros::init(argc, argv, "action_server");

  RobotTrajectoryFollower
RobotTrajectoryFollower("/full_ctrl/joint_trajectory_action");
  ros::spin();

  return 0;
}

Next, I modified my Cmakelist.txt.
 while compilind  i got an error:
[... 209 lines cut ...]
# install(DIRECTORY include/${PROJECT_NAME}/
#   DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION}
#   FILES_MATCHING PATTERN "*.h"
#   PATTERN ".svn" EXCLUDE
# )

## Mark other files for installation (e.g. launch and bag files, etc.)
# install(FILES
#   # myfile1
#   # myfile2
#   DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}
# )

#############
## Testing ##
#############

## Add gtest based cpp test target and link libraries
# catkin_add_gtest(${PROJECT_NAME}-test test/test_ros_action_server.cpp)
# if(TARGET ${PROJECT_NAME}-test)
#   target_link_libraries(${PROJECT_NAME}-test ${PROJECT_NAME})
# endif()

## Add folders to be run by python nosetests
# catkin_add_nosetests(test)

felixvd, 2019-04-16:
This is not a MoveIt bug or enhancement, so I would suggest to post this over at [ROS Answers](https://answers.ros.org/).

felixvd, 2019-11-18:
This is off-topic and not active anymore, so I vote for this to be closed. For people coming here from Google, the first answer is correct: Your robot driver needs to advertise the FollowJointTrajectory action which MoveIt then calls.

closed 2019-11-18 by rhaschke