moveit/moveit #1442: moveit/source docker images should have clang-format, clang-tidy installed

felixvd, 2019-04-18:
Since the source images are meant for development and those two tools are required to submit a PR, they should be included. Who can add `RUN apt-get update && apt-get install  -y --no-install-recommends clang-tidy clang-tools` to the Dockerfile(s)?

felixvd, 2019-04-18:
Actually I am getting this error after installing and running inside the melodic-source container, as suggested [here](https://moveit.ros.org//documentation/contributing/code/). I can't find a helpful Google result. Anyone have the same issue?

```
root@e85708800658:/home/catkin_ws# for file in $(find /home/catkin_ws/build -name compile_commands.json) ; do run-clang-tidy -fix -p $(dirname $file); done
[...]
Unable to run clang-apply-replacements. Is clang-apply-replacements binary correctly specified?
Traceback (most recent call last):
  File "/usr/bin/run-clang-tidy", line 134, in check_clang_apply_replacements_binary
    subprocess.check_call([args.clang_apply_replacements_binary, '--version'])
  File "/usr/lib/python2.7/subprocess.py", line 185, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

BryceStevenWilley, 2019-04-18:
Usually I run that as `run-clang-tidy-6.0.py`, which (on xenial) is on my path as `/usr/bin/run-clang-tidy-6.0.py`.  It seems that [clang-tools](https://packages.ubuntu.com/bionic/clang-tools) on melodic installs `clang-tools-6.0`, which should only install everything with the `-6.0` suffix (had to dig through the [source package](https://packages.ubuntu.com/bionic/clang-tools-6.0) to find that). 

If running with the `-6.0` suffix doesn't work, you can run 
```
grep -nrw `which run-clang-tidy` -e "default='clang-apply-replacements-*"
```
That should spit out the name of the binary that it's failing to find, i.e. `default='clang-apply-replacements-6.0'`.  Whatever it's failing to find, it should likely be present, but with a different version number.

felixvd, 2019-04-19:
I'm just dumb and didn't install clang-tools. It works fine now. My bad.

felixvd, 2019-05-03:
> Who can add RUN apt-get update && apt-get install -y --no-install-recommends clang-tidy clang-tools to the Dockerfile(s)?

I'm guessing @davetcoleman would know, so here's a friendly ping.

rhaschke, 2019-05-03:
[ clang-tidy and clang-tools are already installed in the `ci` docker image](https://github.com/ros-planning/moveit/blob/eb5292d4007ff8b89e3489592041deaf2a5cfa56/.docker/ci/Dockerfile#L34), which serves as a basis for the `source` image. Are you really missing these tools?
Are you running the melodic-source or master-source container?

felixvd, 2019-05-03:
Whoops, so that's where they are. Yes, from what I can see I was running an image built from the melodic-source container, but maybe I had an old image built from release before. I just rebuilt from master-source and everything is installed, so I'll close this. My bad.

closed 2019-05-03 by felixvd