moveit/moveit #1445: Can I set OMPL `optimization_objective` programmingly?

umhan35, 2019-04-23:
### Description

For now, I cannot tell from the log if moveit uses the `optimization_objective` programmingly set in code using `setPlannerParams` like this:

```cpp
setPlannerId("geometric::RRTConnect");

auto planner_params = std::map<std::string, std::string>{ {"optimization_objective", "MaximizeMinClearanceObjective"} };

setPlannerParams(getPlannerId(), group, planner_params);
```

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Binary build, 1.0.1-0bionic.20190320.192509

### Steps to reproduce
See the code in description

I also have a file `~/rosconsole.config` containing `log4j.logger.ros.moveit_ros_perception=DEBUG`

### Expected behaviour
If it is allowed to set OMPL `optimization_objective` programmingly, there should be a debug or info message in the terminal.

### Actual behaviour
The `optimization_objective` is not shown in the terminal. Only parameters are shown if specified in `ompl_planning.yaml`.

### Backtrace or Console output

Not very relevant.

https://gist.githubusercontent.com/umhan35/882b482cf2fa8e169280554be0a354d9/raw/8890cd75ab4eb6d2cc2248ae84ff55329867effd/gistfile1.txt

BryceStevenWilley, 2019-04-23:
This is a good point, there should be some confirmation at the DEBUG level. Currently, there is a [DEBUG call](https://github.com/ros-planning/moveit/blob/ff50476c4070eb86d0a70aa39281d5805db13fa5/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L281) when none of the optimizations are found. As a workaround until those logs are added, you could pass in a nonsense string instead of "MaximizeMinClearance", and you should see a debug log from the model based planner context defaulting to "PathLength". Then you can be sure that your call to `setPlannerParams` is getting passed through/happening at the right time.

mamoll, 2019-11-20:
Not quite addressing the problem, but related: https://github.com/ros-planning/moveit/pull/1695