moveit/moveit #1446: Scaling the distance metric for manipulators

PhilFreeman, 2019-04-23:
Feature request

MoveIt lacks the ability to set scaling parameters for the configuration states in the distance metric. It looks like there is a function ModelBasedStateSpace::setDistanceFunction(), but there is no obvious interface to invoke this from an application layer. Being able to scale the individual states in a manipulator's configuration space can improve planning performance and is very useful when mixing units (e.g. meters and radians) into a meaningful distance metric.

* ROS Distro: Kinetic
* Ubuntu 18.04.02
* MoveIt Binary build

felixvd, 2019-11-17:
This should be a small change, so it could be a WMD issue.

mamoll, 2019-11-17:
A reasonable default weight for a revolute joint would be to be proportional to the sum of link lengths between the joint and the tcp. This could be computed automatically. 

Note that this scaling would affect the collision checking resolution, so this needs to be adjusted as well, based on the weights.

v4hn, 2019-11-18:
This issue might require a small change only, but is quite involved.
Anyone attempting this should be reasonably familiar with OMPL (to observe changes in planning behavior) and be prepared for a somewhat convoluted codebase in the relevant classes. :smile:

mamoll, 2019-11-18:
This is potentially a small change, but the impact could be huge. Any planning pipeline probably uses the MoveIt-defined distance metric. For example, the notion what an "optimal" motion is, is going to be affected by this. By redefining it along the lines I described above, short paths will get closer to minimizing swept volume, which is often desirable. I am not sure if detailed knowledge of OMPL is necessary. The planners in OMPL just operate over some space with a metric; they don't need to know if it's a ModelBasedStateSpace with some specific metric.

I'd happy to assist someone who wants to work on this.

PhilFreeman, 2019-11-19:
I started to look at what it would take, and quickly got distracted by other things. :) I'm just not familiar enough with the build pipeline to have a productive dev environment. I know the math, and I've explored the code enough that I think this is quite feasible - but not all that straight forward. Of course, I'd leave the default metric as is. The needed feature would be to allow a user to define an alternate metric in whatever way they want. That metric might even just be simple weights that aren't a function of state. The problem that triggered this for me was a 6-DOF manipulator on a track. The track was, obviously in m, but the joints of the robot were metricized in rad. The mixture of the two resulted in some clearly lousy planned paths.

mamoll, 2019-11-19:
The default distance function called by ModelBasedStateSpace is `moveit::core::JointModelGroup::JointModelGroup::distance()`, which computes a weighted L1-norm over the joints (more or less). The weights are set for each joint with `moveit::core::JointModel::setDistanceFactor()`.  This function is currently not called anywhere. A first step might be to add a planning group parameter, say, `joint_weights` to a yaml config file (maybe kinematics.yaml?) that needs to be set to a list of non-negative float values. The length of the list needs to be exactly equal to the number of joints in that group. The `setDistanceFactor` is then called for each joint with the corresponding weight.

If someone can demonstrate that a particular weighting scheme (like the one above) is almost always better than the default weighting scheme (all weights equal to 0), then perhaps the default weights should be changed.

gonfreces, 2019-11-20:
I'll work on this! I am using Kinetic and Ubuntu 16.04. Would that be an issue as the environment described is Melodic and Ubuntu 18.04.02

mamoll, 2019-11-20:
Should not be a major issue, I'd think. The changes will be very localized, so likely easy to port to different versions if needed.

gonfreces, 2019-11-20:
Great to hear that. Also, I had a chat with Dave. So, we thought of creating a joint weight property in the panda_arm.xacro file and then send these values to setDistanceFactor() function in joint_model.h to set up the distance_factor for each joint. The user can edit the weight values in the panda_arm.xacro file to set their own desired values. Secondly, this feature can be added to the moveit setup assistant as a slider function similar to setting up the robot poses. What do you think of this solution?

gavanderhoorn, 2019-11-20:
> So, we thought of creating a joint weight property in the panda_arm.xacro file

Don't you mean `panda_arm.srdf`?

The MSA changing `.xacro`s would seem strange (and undesirable) to me.

mamoll, 2019-11-20:
Adding this to the setup assistant and adding UI for setting weights should come at a later stage. For now, perhaps just reading in a YAML file that this kind of information and setting up the weights for each group would be good:
```
arm:
     shoulder: 4
     elbow: 2
     wrist_1: 1
     wrist_2: 1
     wrist_3: 1
arm_with_torso:
     torso: 5
     shoulder: 4
     elbow: 2
     wrist_1: 1
     wrist_2: 1
     wrist_3: 1
arm_with_long_tool:
     shoulder: 4
     elbow: 3
     wrist_1: 2
     wrist_2: 2
     wrist_3: 2
```
(In other words, the YAML file list a number of joint groups and associate a weight with each joint in a joint group.)

davetcoleman, 2019-11-20:
> In other words, the YAML file lists a number of joint groups and associates a weight with each joint in a joint group.

The way RobotModel / JointModel data structures are currently designed, joint weights cannot be specified per group, [only per robot](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_model/include/moveit/robot_model/joint_model.h#L510). We could change that, but it would be a much deeper refactor.

> Another question/comment then: how would this generalise to other robots and MoveIt configuration packages that don't use xacro for the .srdfs?

The xacro detail should be ignored, my suggestion to @gonfreces is that we add this to the SRDF spec. That would be convenient because the generate joint model function has the SRDF passed in. However, I also like modeling this config off of how ``joint_limits.yaml`` works.

mamoll, 2019-11-20:
I see. In that case, perhaps just a YAML syntax like this:
```
joint_weights: # at most one such list is defined per robot
     shoulder: 4
     elbow: 2
```
Any joint that isn't listed will have a default weight of 1.

mamoll, 2019-11-21:
BTW, I like the idea of having SRDF support for this, but getting this recognized as "standard" and well-supported for many robots will take a while. Having a simple mechanism like the YAML file makes it possible to experiment a bit and see what tends to work well. We also need to understand how this affects collision checking resolution and if there are other side-effects before making this part of a standard.

davetcoleman, 2019-11-21:
I don't think it would take too much effort to add support for this in SRDF, since we have full control of that spec within ros-planning (unlike, cough, URDF)

However I struggled yesterday deciding if joint weights for distance metrics fall under the "semantic" info or not. If we add this to SRDF, we should def. merge in joint limits, also... and the grasping geometry info that we've been developing in moveit_grasps (@mlautman )

v4hn, 2019-11-22:
The weights are currently only considered for OMPL planning, so I don't think it qualifies for the srdf yet.

I strictly prefer specifying the weights for each joint model group in `ompl_planning.yaml`.
We already allow to specify JMGs with the same joints but different parameters (e.g. IK solver).
I don't see why it would make sense to support these weighting factors in any other way.

Yes, you can't rely on the `JointModel`/`JointModelGroup` structures that way.
But then, I guess it's time to move OMPL's distance computation altogether to the `ModelBasedStateSpace`.

@davetcoleman From my perspective, joint limits and other things could clearly be moved to the srdf, but maybe it would also rework MoveIt's config structure for no real profit, confusing users?

For new features, e.g. "grasping geometry", it might very much make sense to add them to the SRDF if they are independent of the algorithm.
But nobody proposed anything to add there in the last 4 years...

mamoll, 2019-11-22:
@v4hn I had assumed that other planning libraries would (perhaps indirectly) also call JointModelGroup::distance. CHOMP and TrajOpt are both optimizing path length (plus smoothness, clearance, etc.). Presumably they use some notion of distance to compute path length. It seems desirable that they use the same one. Maybe not?

rhaschke, 2019-11-22:
I think, it would be nice to be able to specify different joint weights for different JMGs as suggested in 
https://github.com/ros-planning/moveit/issues/1446#issuecomment-557542002. However, I agree to @mamoll that joint weights are not limited to a specific planning algorithm.
On the other hand, as @davetcoleman pointed out in https://github.com/ros-planning/moveit/issues/1446#issuecomment-556417347, currently there is only a single unique joint weight (`distance_factor_`) in the JointModel. It would be great to consider those as defaults for all JMGs, but to allow overwriting them in the JMG. To this end, the getDistanceFactor() method should be moved to JMG?

v4hn, 2019-11-22:
> I had assumed that other planning libraries would (perhaps indirectly) also call JointModelGroup::distance.

A quick `grep` in moveit_planners seems to indicate the other planners in the repo don't call it.
It's easy to imagine planners that require distance computation different from simple weighted L1 norm.
Imho, specifying a distance function is outside the scope of the `JointModelGroup` and I could imagine deprecating it altogether.

> To this end, the getDistanceFactor() method should be moved to JMG?

This seems to be a reasonable middle ground if an outside implementation would be more costly and we want to keep the L1 norm around as a default that might be used by planners.

v4hn, 2019-11-22:
BTW, if we keep it around the `*::distance` function should really be documented!

GabrielKoenig, 2019-11-30:
Hey guys
I would like to join his discussion and contribute my thoughs. From my point of view, one important thing is missing so far in this discussion. It is the fact that there is already the possibility to set the `joint_weights` within moveit. Within the kinematics.yaml file you can add the two paramters for example as follows:

```
joint_weights:
   weights: [0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 1, 1]
   joint_names: [panda_joint1, panda_joint2, panda_joint3, panda_joint4, panda_joint5, panda_joint6, panda_joint7, panda_finger_joint1, panda_finger_joint1]
```

They are loaded in https://github.com/ros-planning/moveit/blob/master/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp#L82) . 

For example for my robot I want to set the joint weights in relation to the level arm distance for a kinematic chain robot, with the assumption that one unit movement of the joint closer to the end-effector has a smaller probability for collision than base joint unit movement. I figured out if I set the `joint_weights` as described in the kinematics.xaml file, they are only used within the kinematic_plugin (moveit_core) but have no influence on the `distance_factor_` and thus no influence on the distance calculation (distance metric) within the JointModel class and therfore also not for the OMPL planner. Even though the expected behaviour is as desired (base joint move less relative to the endeffector) in moveit, I don't know why? (e.g. if you set the weights for one joint to 0, this joint is "fixed" when trying to move the panda arm).  Could it be that joint_weights have a direct influence how the the state space is sampled? I would also have expected that the joint_weights would have an influence on the distance metric and set equall to the `distance_factor`. But this is not the case. 

Therfore @v4hn I think it can be missleading creating a separate yaml file to set joint weights (the user would not now which joint_weights influences what). And therefore @davetcoleman and @mamoll I think one need to consider that adding support to srdf for the joint weight can lead to conflicts as they might get overwritten (or vice versa) by the paramters within the kinematics.yaml. Or am I wrong?

My questions summarized are therfore: 
- Have the `joint_weights` an influence on the sampling behaviour?
- What is exactly the differenc between the `joint_weights` paramter which can be specified within the kinematics.yaml and the `distance_factor_ `paramter used for the JointModel class resp. the OMPL planners? 
- Would it make sense to set the `joint_weights` equally to the `distance_factor_` of each joint? (considering if somebody doesn't use the `kdl_kinematics_plugin` the `joint_weight` parameter is only used for the distance function of the JointModel class).

gavanderhoorn, 2019-11-30:
Last comment a continuation of [What is the difference between joint_weights (KDLKinematics parameter) and distance_factor_ (JointModel class of moveit) parameter?](https://answers.ros.org/question/338786) on ROS Answers.

v4hn, 2019-12-01:
--- On topic
Thank you for pointing out these plugin-specific weight parameters. I believe if we introduce a new way to specify them throughout the system, it would make sense to use them for the plugin too.
Maybe we even want to deprecate the ik parameter altogether then.

--- Off topic

> Even though the expected behaviour is as desired (base joint move less relative to the endeffector) in moveit, I don't know why? (e.g. if you set the weights for one joint to 0, this joint is [...]

As you found out yourself, the weights you mention *only* influence the KDL plugin (other plugins might even use other names for a parameter like that - if they even support one.
The reason why the parameters still influence your planning is quite simple:
If you specify a `Pose` target, the IK plugin will be called - which considers these weights.
The joint space planning afterwards will find a trajectory to the computed joint goal.
"Just" restricting the goal state usually changes the behavior of the planning system a lot.

pathakarvindk, 2022-11-04:
> Hey guys I would like to join his discussion and contribute my thoughs. From my point of view, one important thing is missing so far in this discussion. It is the fact that there is already the possibility to set the `joint_weights` within moveit. Within the kinematics.yaml file you can add the two paramters for example as follows:
> 
> ```
> joint_weights:
>    weights: [0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 1, 1]
>    joint_names: [panda_joint1, panda_joint2, panda_joint3, panda_joint4, panda_joint5, panda_joint6, panda_joint7, panda_finger_joint1, panda_finger_joint1]
> ```
> 
> They are loaded in https://github.com/ros-planning/moveit/blob/master/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp#L82) .
> 
> For example for my robot I want to set the joint weights in relation to the level arm distance for a kinematic chain robot, with the assumption that one unit movement of the joint closer to the end-effector has a smaller probability for collision than base joint unit movement. I figured out if I set the `joint_weights` as described in the kinematics.xaml file, they are only used within the kinematic_plugin (moveit_core) but have no influence on the `distance_factor_` and thus no influence on the distance calculation (distance metric) within the JointModel class and therfore also not for the OMPL planner. Even though the expected behaviour is as desired (base joint move less relative to the endeffector) in moveit, I don't know why? (e.g. if you set the weights for one joint to 0, this joint is "fixed" when trying to move the panda arm). Could it be that joint_weights have a direct influence how the the state space is sampled? I would also have expected that the joint_weights would have an influence on the distance metric and set equall to the `distance_factor`. But this is not the case.
> 
> Therfore @v4hn I think it can be missleading creating a separate yaml file to set joint weights (the user would not now which joint_weights influences what). And therefore @davetcoleman and @mamoll I think one need to consider that adding support to srdf for the joint weight can lead to conflicts as they might get overwritten (or vice versa) by the paramters within the kinematics.yaml. Or am I wrong?
> 
> My questions summarized are therfore:
> 
> * Have the `joint_weights` an influence on the sampling behaviour?
> * What is exactly the difference between the `joint_weights` paramter which can be specified within the kinematics.yaml and the `distance_factor_ `paramter used for the JointModel class resp. the OMPL planners?
> * Would it make sense to set the `joint_weights` equally to the `distance_factor_` of each joint? (considering if somebody doesn't use the `kdl_kinematics_plugin` the `joint_weight` parameter is only used for the distance function of the JointModel class).

I have tried the same by defining the joint_weights in the kinematics.yaml file. I have defined "0" for all the joints, still manipulator is moving. 

Do we need to do something else to implement the same?

GabrielKoenig, 2022-11-04:
Hi @pathakarvindk 
It hat been quite some time since I used moveit (for my masterthesis), hence I don't know anymore what you need to set...
Probably try also to set the distance_factor the same as your joint weights. Probably you are not using the kdl_kinematics plugin. But as I said I don't know the current code base ...
Cheers
Gabriel

pathakarvindk, 2022-11-04:
> Hi @pathakarvindk It hat been quite some time since I used moveit (for my masterthesis), hence I don't know anymore what you need to set... Probably try also to set the distance_factor the same as [...]

You are right @GabrielKoenig 
I am using trak_ik.
Do we have something similar for trak_ik.

Thank you