moveit/moveit #1452: chomp/OptimizerAdapter: Start state is empty

aaronchongth, 2019-04-30:
### Description
Attempted to use `chomp/OptimizerAdapter` after `OMPL`. Followed tutorials and faced the problem of `Start state is empty`. The example print-out can be seen at the bottom,

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build, with `moveit_planners` built from source
* `ros-melodic-moveit` binaries `1.0.1-0bionic.20190320.192509`
* `moveit_planners` from `melodic-devel`

### Steps to reproduce

Start the tutorial through the **Getting Started** portion of [MoveIt! Tutorials](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/getting_started/getting_started.html).

After setting up the workspace, start following the steps of [Planning Adapter Tutorials](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/planning_adapters/planning_adapters_tutorial.html), under **Running OMPL as a pre-processor for CHOMP**.

I noticed that there might be some errors in this section, in order to use this setup with `chomp/OptimizerAdapter`, we would need the `ompl` planner to be called instead of `chomp`, so I believe an error we should be launching
```
roslaunch panda_moveit_config dmeo.launch
```
instead of 
```
roslaunch panda_moveit_config demo_chomp.launch
```

The next error I got was 
```
[ERROR] [1556590982.275248450]: Exception while loading planning adapter plugin 'chomp/OptimizerAdapter': According to the loaded plugin descriptions the class chomp/OptimizerAdapter with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds industrial_trajectory_filters/AddSmoothingFilter industrial_trajectory_filters/NPointFilter industrial_trajectory_filters/UniformSampleFilter
```

After digging around, I found that the `chomp/OptimizerAdapter` was not built in the binaries, so I did the following
```
cd /tmp
git clone -b melodic-devel https://github.com/ros-planning/moveit
cp -r /tmp/moveit/moveit_planners ~/ws_moveit/src/.
```
Rebuilding them again now,
```
cd ~/ws_moveit
catkin clean -y
catkin build
```
I then got a dependency issue, which was solved with
```
sudo apt-get install ros-melodic-moveit-resources
catkin build
```
The workspace built without any issues, however when I launched the demo via
```
roslaunch panda_moveit_config demo.launch
```
adding the **Motion Planning** panel in `rviz`, under **Planning** tab -> **Planning Group**: panda_arm ->**Goal State**: random_valid, **Plan and Execute**. I got the following logs

### Console output and the files I edited from tutorials
https://gist.github.com/aaronchongth/116e85a65d26e3daef7c47fbed98e209

Master-Matt, 2019-05-17:
I have the same issue in both Melodic and Kinetic using either the python or C++ tutorials

BryceStevenWilley, 2019-05-19:
I'm looking into this now, but also having some trouble getting CHOMP to work.

> After digging around, I found that the chomp/OptimizerAdapter was not built in the binaries.

As mentioned in the [tutorial](https://ros-planning.github.io/moveit_tutorials/doc/planning_adapters/planning_adapters_tutorial.html#installing-moveit-from-source), you do need to have MoveIt installed from source.

@aaronchongth, unfortunately, the gist you linked no longer exists. If you still have that console output, would you mind pasting it into this issue? @Master-Matt, could you also give some more information about your setup?

aaronchongth, 2019-05-19:
@BryceStevenWilley thanks for looking into this, I accidentally cleanup all my gists a couple of days ago, I'll try to recreate the build, and this time with moveit fully built from source

Master-Matt, 2019-05-19:
I have been working around this issue for a while now. I have tried it on kinetic with moveit both built from source and using binaries. Now. Getting the same results with Melodic in either configuration. I'm working with a fanuc 200id from the fanuc experimental repo, but also tested with panda to confirm the results. The odd part is that when moving the interactive marker in rviz, it works fine, but when using either the python or C++ tutorial code, right after the info for configuring the CHOMP planning scene I get the start state empty error. I will grab some outputs and post them here to show the difference. One thing I'm wondering is if I should be building the industrial robot simulator from source, as right now I'm only using the binary. I've tested multiple configurations as well and the only planner that works every time is straight OMPL. Using CHOMP or STOMP adaptors or planners by themselves causes the planning to fail. I also tried using the python command line moveit commander with the same results. In all cases, I can get it to work in rviz by manually planning then executing goals.

aaronchongth, 2019-05-19:
I have just done a clean rebuild with `moveit` on master branch, the issue still persists, here's a gist of the log from launching the tutorial
```
roslaunch panda_moveit_config demo.launch
```
```
roslaunch moveit_tutorials move_group_interface_tutorial.launch
```

https://gist.github.com/aaronchongth/54ee50ba8115912db66b6027c57841f3

Master-Matt, 2019-05-19:
Here is a working example (Using RViz to manually plan and execute)
[](https://gist.github.com/Master-Matt/3417b22ce99ad205c8d59bcdfc3b9219)
Here is using the python move_group interface.
[](https://gist.github.com/Master-Matt/fe7f7de90e07fb9232c378a071f418db)

aaronchongth, 2019-05-19:
@Master-Matt hello matt! If its not too much trouble, could you post the steps and repo links to recreate the successful planning? I am fascinated by this difference in behaviour

aaronchongth, 2019-05-19:
8

Master-Matt, 2019-05-20:
I really didn't do anything special. I installed moveit from source. I also installed ros_perception from source, followed by the panda config package and the moveit tutorials stuff. After "rosdep"ing the workspace, and apt-get for the rest, I built it all with catkin build then ran the move_group python and c++ tutorials. I first used OMPL, then the other planners. I am using a pointcloud input for the octomap so I am doing collision checking as well.

closed 2019-05-25 by BryceStevenWilley

SunYuansong, 2019-06-22:
Hello,
@aaronchongth Have you worked it out? I want to use OMPL as a pre-processor for CHOMP, following the guidance of the moveit official website. The single ompl and single chomp both run successfully, but the OMPL+CHOMP just cannot run. 
Here is the log:
https://gist.github.com/SunYuansong/f89665fe592274b2b1b44ef42717468b
I'm using the latest moveit built from source, so the #1473 should have been merged into it.

rhaschke, 2019-06-25:
@SunYuansong, it would be helpful if you could provide a minimal example that triggers the failure.

rhaschke, 2019-06-28:
@SunYuansong, I was able to reproduce the issue on the master branch. A corresponding fix was filed.

SunYuansong, 2019-07-02:
@rhaschke Sorry for late and thanks a lot! The CHOMP adapter works now.

luckychay, 2020-02-12:
@rhaschke, I pulled the latest version of master and built it. Just the same as @SunYuansong ,the single ompl and single chomp both run successfully, but the OMPL+CHOMP just can't work and  I still got an error as following:
`[ERROR] [1581525146.485400698]: Exception while loading planning adapter plugin 'default_planner_request_adapters/CHOMPOptimizerAdapter': According to the loaded plugin descriptions the class default_planner_request_adapters/CHOMPOptimizerAdapter with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  chomp/OptimizerAdapter default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds default_planner_request_adapters/ResolveConstraintFrames`

I'm using ROS kinetic on ubuntu 16.04 LTS . And  I'm wondering whether I missed anyting or not.

PR #1473 (2019-05-25): Fix several Chomp issues
This PR comprises several independent commits to improve the Chomp code, namely
- some code cleanup, reducing redundancy and explicit loops
- fix start state handling: 
  - An empty start state should be accepted. In this case, the current state should be used. Fixes #1452
  - Checking start state constraints was plain wrong, resulting in #1470.
- Avoid extra conversion from/to MotionPlanResponse _message_