moveit/moveit #1454: Adding new planner to moveit build fail

sumedh11n, 2019-05-01:
### Description
I am trying to add a new planner to the ompl planner list through moveit. 
I have followed the following steps:
1. Installed moveit from source.
2. Installed ompl from source. (ompl src dir cloned to 'ws_moveit/src' path, where ws_moveit is my moveit installation dir)
3. Added my planner cpp file and header file to /ws_moveit/src/ompl/src/ompl/geometric/planners
and headers_geometric.txt and modified src/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp and planning_context_manager.h

Now, when I try catkin_make console shows error as:
_CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_package_xml.cmake:54 (message):
  catkin_package_xml() package name 'ompl' in
  '/home/sumedh/ws_moveit/src/moveit/moveit_planners/ompl/package.xml' does
  not match current PROJECT_NAME 'moveit_planners_ompl'.  You must call
  project() with the same package name before.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:99 (catkin_package_xml)
  CMakeLists.txt:26 (catkin_package)_

[image]

I also tried manually adding package.xml as mentioned [here](https://moveit.ros.org/install/source/dependencies/)

But, when I remove ros-melodic-ompl with sudo apt-get remove ros-melodic-ompl
the console shows the following error:
_CMake Error at CMakeLists.txt:22 (find_package):
 "By not providing "Findompl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ompl", but
  CMake did not find one.

  Could not find a package configuration file provided by "ompl" with any of
  the following names:

    omplConfig.cmake
    ompl-config.cmake

  Add the installation prefix of "ompl" to CMAKE_PREFIX_PATH or set
  "ompl_DIR" to a directory containing one of the above files.  If "ompl"
  provides a separate development package or SDK, be sure it has been
  installed."_

[image]

Also, I am not sure what should be the CMAKE_PREFIX_PATH and whether I should run the cmake -DCMAKE_INSTALL_PREFIX=/opt/ros/hydro command.

I am new to this and I have literally tried everything that i found on the internet.
Please guide me through this. @mamoll @davetcoleman  Thanks!

### My environment
* ROS Distro: [melodic]
* OS Version: Ubuntu 18.04
* Source build
* If binary, which release version?
* If source, which git commit or tag? https://bitbucket.org/ompl/ompl/

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

mamoll, 2019-05-01:
It'd be best to stick to a released version of OMPL that is known to work with a particular version of ROS. In your case, you should checkout version 1.4.2.

The first error can probably be fixed by changing the project name as suggested by the error message.

The second error can probably be fixed by setting the environment variable `CMAKE_PREFIX_PATH `:
```
export CMAKE_PREFIX_PATH=/path/to/your/ompl/installation/prefix
```

sumedh11n, 2019-05-01:
> It'd be best to stick to a released version of OMPL that is known to work with a particular version of ROS. In your case, you should checkout version 1.4.2.
>
> The first error can probably be fixed by [...]

sumedh11n, 2019-05-01:
> It'd be best to stick to a released version of OMPL that is known to work with a particular version of ROS. In your case, you should checkout version 1.4.2.
>
> The first error can probably be fixed by [...]

Thanks for the response Mark.
I am already using ompl version 1.4.2 along with ROS melodic.
Regarding CMAKE_PREFIX_PATH I am not sure about what the installation prefix mean. Should it be the path to ompl (in my case /home/sumedh/ws_moveit/src/ompl)? If yes, then I have already set the path.
Also do you suggest clean installing ubuntu to 16.04 and trying with ROS kinetic?

BryceStevenWilley, 2019-05-18:
@sumedh11n `CMAKE_PREFIX_PATH` should be set you where OMPL is built/installed. If you are building in a catkin workspace, then the correct path should be `/home/sumedh/ws_moveit/build/ompl`.

closed 2019-05-18 by sumedh11n