moveit/moveit #1456: jog_arm: low-pass filter stop commands as well

jschleicher, 2019-05-07:
### Description

For acceleration the (joystick|spacemouse|...) command is low-pass filtered to avoid jumps in velocity. For stopping (desired velocity=0) this filtering is skipped (probably to allow hard-stop before collisions?).

### Steps to reproduce
1. Move the robot with an xbox-controller.
2. Release the button. There is a huge jerk observed.

### Expected behaviour
There should be a stopping ramp by low-pass filtering the velocity (even better: respecting the joint_limits max_deceleration).

### Actual behaviour
Hard stop.

@AndyZe What do you think about having two distinct functions `hard_stop()` and `soft_stop()` without and with filtering enabled?

AndyZe, 2019-05-07:
I think that's a fine idea. I can implement it in the next couple weeks, or happy to review a PR from you.

AndyZe, 2019-05-16:
See #1468

davetcoleman, 2019-08-13:
Closed by https://github.com/ros-planning/moveit/pull/1468, thanks @AndyZe !

closed 2019-08-13 by davetcoleman