moveit/moveit #1458: MoveGroup sensitive to length of orientation values

rmerriam, 2019-05-08:
### Description
The number of decimal digits of the orientation values affects whether a move occurs or not. 

### Your environment
Binary Melodic on Ubuntu 16.04

### Steps to reproduce
I'm working with the Ariac 2019 environment using MoveIt for controlling the arms. A custom ROS message is sent to a node that controls one of the arms in the environment.

`                   manipulator_group.setJointValueTarget(msg.part_source);`
`                    if ( !manipulator_group.move() == MV::MoveItErrorCode::SUCCESS) {`
`                        cerr << "ABORTED: " << pose_name << '\n';`
`                    }`

Below are two yaml files which demonstrate the issue. I'm in the middle of the competition and will try next week to provide additional code and data. The section of the message _part_source_ is passed to _setJointValueTarget_ in the code snippet above. 

The first yaml results in the arm moving. The second yaml does not move the arm. The only difference, and it took me 2 days to hunt this down, is the length of the values in the _orientation_ section of the message. With two additional digits a move occurs. Specifically, 
0.707107 works but 0.7071 does not. 

`named_position: [a1_neutral, a1_tray, src, a1_neutral, a1_rail]`
`use_pose: true`
`part_source: `
`  position: `
`    x: 0.356237`
`    y: 3.00498`
`    z: 0.784236`
`  orientation: `
`    x: 0`
`    y: 0.707107`
`    z: 0`
`    w: 0.707107`
`part_destination:` 
`  position: `
`    x: 0.0000`
`    y: 0.0000`
`    z: 0.0000`
`  orientation:` 
`    x: 0.0000`
`    y: 0.0000`
`    z: 0.0000`
`    w: 0.0000`
`

`named_position: [a1_neutral, a1_tray, src, a1_neutral, a1_rail]`
`use_pose: true`
`part_source: `
`  position: `
`    x: 0.356237`
`    y: 3.00498`
`    z: 0.784236`
`  orientation: `
`    x: 0`
`    y: 0.7071`
`    z: 0`
`    w: 0.7071`
`part_destination: `
`  position: `
`    x: 0.0000`
`    y: 0.0000`
`    z: 0.0000`
`  orientation:` 
`    x: 0.0000`
`    y: 0.0000`
`    z: 0.0000`
`    w: 0.0000`

felixvd, 2019-05-08:
Just spitballing here, but maybe the shorter version is rejected for being too far from normalized? Have you tried normalizing the quaternion before using it?

```
Eigen::Quaterniond quaternion(msg.orientation.w, msg.orientation.x, msg.orientation.y, msg.orientation.z);
  quaternion.normalize();
```

gavanderhoorn, 2019-05-08:
This could indeed be a normalisation issue.

If that is the case, it'd be nice if whoever is loading that data could warn the user though.

@rmerriam: would you perhaps have a link to the code that loads this file? It's not a MoveIt file I recognise.

rmerriam, 2019-05-08:
It started working when I normalized the numbers. It's actually a constant that I use to force the 'gripper' to be horizontal when reaching for parts. 

A warning would be nice. 

The yaml is for demonstrating the problem using rostopic to publish the message. The code snippet in the OP is MoveIt code. The pose is simply passed as the target and the move group called.

felixvd, 2019-05-10:
Is there no warning? Even in debug mode? Could you show the output with the logger set to debug? I am unsure where the error happens.

felixvd, 2019-11-26:
Ping @rmerriam