moveit/moveit #1460: How do I interact the actual robot(franka emika arm robot) with visualization by rviz in Moveit?

amranov, 2019-05-10:
I have visualized my robot arm in Rviz. And the actual robot side also connects to franka_arm (ROS installed already). I have read many documents and articles but they made me dizzy.

As what I have researched that I have to create a publisher(in Moveit?) then send data to ActionServer , then using Ros in Raspberry to control servo? Is this plan correct?

I am sorry if I make confusing, please help me to fix the plan for interacting with the actual robot!

felixvd, 2019-05-10:
The only instructions you should need to follow to use MoveIt are these (or the one for Kinetic, if that's the version you have installed): http://docs.ros.org/api/moveit_tutorials/html/index.html

MoveIt is used for motion planning. You don't create a publisher in MoveIt. I would suggest following a few more basic ROS tutorials to understand what goes on in the system.

If you still have questions, you should post it in more detail on [ROS Answers](https://answers.ros.org/) instead. This question is not an issue with MoveIt, so it does not belong here.

amranov, 2019-05-10:
@felixvd  thank you for reply, 
i will post the issue in ROS Answers

closed 2019-05-10 by amranov