moveit/moveit #1463: allow installation without ros_perception

jschleicher, 2019-05-14:
### Description

Debian packages depend on moveit-ros-perception, even if just using the moveit-kinematics package. That pulls in several hundred megabytes for packages (most noticably libopencv-dev), that some users may not need.

Is it possible to loosen the dependency on ros_perception? https://github.com/ros-planning/moveit/blob/0eae58c58d7dcef712d5e0f0534466c87955f1c2/moveit_ros/planning/package.xml#L22
Currently the dependency is only needed, because planning_scene_monitor includes the occupancy_map_monitor. Would it be feasible, to convert that piece of code into a plugin structure (could you point me into the right direction?) Or can you think of other ways reducing the installed size of moveit for applications, where semi-static plans or just accessing the kinematics and robot state is sufficient?

Another use-case could be processing the sensor data on a high-end pc with octomap and opencv installed, while realtime control and sophisticated planning using moveit-ros-planning are executed on an embedded (but maybe realtime) device.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Binary build

### Steps to reproduce
```
docker run -it ros:melodic-ros-base bash
apt install ros-melodic-moveit-kinematics
```

### Expected behaviour
100-150MB disc usage would be fine.

### Actual behaviour
`After this operation, 622 MB of additional disk space will be used.`

rhaschke, 2019-06-28:
Sorry for the late reply. I think `moveit_ros_planning` requires `perception` and the `occupancy_map_monitor`.
However, you were asking about `moveit_kinematics` originally, which unnecessarily depends on `moveit_ros_planning` for a [benchmark program](https://github.com/ros-planning/moveit/blob/cfb46fe3ba6b0d708f22c8692c337a47d6cde6da/moveit_kinematics/cached_ik_kinematics_plugin/src/measure_ik_call_cost.cpp). If we can remove this dependency, you are fine, right?

jschleicher, 2019-07-01:
My use-case is e.g. <br>[pilz_trajectory_generation](https://github.com/PilzDE/pilz_industrial_motion/tree/melodic-devel/pilz_trajectory_generation) -> moveit_ros_planning (for RobotModelLoader)
and <br> &nbsp;&nbsp; -> moveit_ros_planning_interface (to inherit from PlanningContext)

So #1529 is a good start in general, but my use case would need to relax the dependency `moveit_ros_planning->moveit_ros_perception` in quasi-static environments. In our case no sensor does update the octomap / planning scene but there is just the robot model with environment data loaded beforehand (which could still include dynamic obstacles in a multi-robot-case).

rhaschke, 2019-07-01:
This would actually require to split `moveit_ros_planning` into smaller units.

jschleicher, 2019-07-09:
Fixed in #1533.

closed 2019-07-09 by jschleicher