moveit/moveit #1470: CHOMP doesn't work with Floating joint

BryceStevenWilley, 2019-05-19:
### Description

While attempting to recreate https://github.com/ros-planning/moveit/issues/1452 (tutorial/panda_moveit_config also from source), I ran into another bug that causes CHOMP to fail with `Start state violated joint limits`. 

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source: moveit/2dfe65e8

### Steps to reproduce
`roslaunch panda_moveit_config demo_chomp.launch`, move arm slightly.

### Expected behaviour
Should be able to plan. 

### Actual behaviour

```
[ INFO] [1558234232.380938293]: Configuring Planning Scene for CHOMP ....
[ERROR] [1558234232.511067939]: Start state violates joint limits
[ WARN] [1558234232.534404999]: Fail: ABORTED: Unknown event
```

### Backtrace or Console output

I printed out the actual joint angles and bounds being checked (code on [this branch](https://github.com/BryceStevenWilley/moveit/tree/floating_joint_broken_debug)), which is being reported as: 
```
[ERROR] [1558245278.740119596]: Joint 0 failed
[ERROR] [1558245278.740181154]: jj: 0: 0.000000: min: -inf, max: inf
[ERROR] [1558245278.740200916]: jj: 1: -0.785000: min: -inf, max: inf
[ERROR] [1558245278.740223415]: jj: 2: 0.000000: min: -inf, max: inf
[ERROR] [1558245278.740237277]: jj: 3: -2.356000: min: -1.000000, max: 1.000000
[ERROR] [1558245278.740251680]: jj: 4: 0.000000: min: -1.000000, max: 1.000000
[ERROR] [1558245278.740263902]: jj: 5: 1.571000: min: -1.000000, max: 1.000000
[ERROR] [1558245278.740279026]: jj: 6: 0.785000: min: -1.000000, max: 1.000000
```
which suggests that for some reason the planning request joint state does not include the floating joint present between the panda base link and the world. This error is present throughout the entire planning pipeline (I've copy pasted the checking code in OMPL's manager and the highest level planning pipeline call), and OMPL somehow handles this fine, but CHOMP will check the start state and fail. Still need to find out either:
* why the planning scene expects the robot state to include the floating joint but doesn't
* how OMPL still plans fine in this case
* what's the correct way to resolve this.

rhaschke, 2019-05-20:
There is an issue in the way Chomp handles the start state. I'm looking at it.

BryceStevenWilley, 2019-05-25:
Fixed by #1473 for me.

closed 2019-05-25 by BryceStevenWilley