moveit/moveit #1471: jog_arm first order low-pass filter

AndyZe, 2019-05-21:
Per conversation in #1468, the 2nd-order filter that is currently used is known to overshoot. A first-order filter may be a better fit (as well as more efficient).

https://dankelley.github.io/r/2014/01/15/butterworth-filters.html

SansoneG, 2019-05-27:
Changing
```cpp
double new_filtered_msrmt =
     (1. / (1. + filter_coeff_ * filter_coeff_ + 1.414 * filter_coeff_)) *
     (previous_measurements_[2] + 2. * previous_measurements_[1] + previous_measurements_[0] -
      (filter_coeff_ * filter_coeff_ - 1.414 * filter_coeff_ + 1.) * previous_filtered_measurements_[1] -
      (-2. * filter_coeff_ * filter_coeff_ + 2.) * previous_filtered_measurements_[0]);
```
to
```cpp
double new_filtered_msrmt =
    (1. / (1. + filter_coeff_)) *
    (previous_measurements_[1] + previous_measurements_[0] -
     (-1. * filter_coeff_ + 1.) * previous_filtered_measurements_[0]);
```
in `low_pass_filter.cpp:66-70` seems to solve the issue.

Here is a plot of the new filter:
[image]

jschleicher, 2019-05-27:
@SansoneG Please provide a pull-request.

SansoneG, 2019-05-27:
> @SansoneG Please provide a pull-request.

#1483

rhaschke, 2019-08-07:
Fixed in #1483.

closed 2019-08-07 by rhaschke