moveit/moveit #1480: Missing/broken link to tutorial for jog_arm

jstrieter, 2019-05-23:
The tutorial link referenced under moveit_experimental/jog_arm is broken. I've searched for the tutorial elsewhere, such as on archive.org but can't find it anywhere. I have found some guidance from other issues submitted but it hasn't got me to where I understand exactly how to use jog_arm. Does anyone have a copy of the referenced tutorial? I'm trying to get it to work with a KUKA kr6. I have RSI communication working as well as planning/control of the real robot using moveit and rviz with gavanderhoorn's moveit package however I lack the understanding to get the action server communicating with the jog_arm_server. 

* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* ros-desktop-full from binaries, moveit from source
* moveit master branch

Thanks in advance

Jake

gavanderhoorn, 2019-05-24:
The tutorial for `jog_arm` is in a bit of a flux at the moment (see ros-planning/moveit_tutorials#268 fi).

> I'm trying to get it to work with a KUKA kr6. 

Let's hope I'm wrong, but last time I tried `jog_arm` with a position-only controlled robot that requires jerk-limited inputs, it didn't work.

> I have RSI communication working as well as planning/control of the real robot using moveit and rviz with gavanderhoorn's moveit package however I lack the understanding to get the action server communicating with the jog_arm_server.

`jog_arm` doesn't use a `FollowJointTrajectory` action, but publishes position setpoints directly to a `ros_control` `JointGroupPositionController` or a `JointGroupVelocityController`. 

If you're using `kuka_rsi_hw_interface`, you'll have to update the configuration to load and start the `JointGroupPositionController` (as there is no support for velocity control).

jstrieter, 2019-05-28:
Much appreciated. I'm working on adding the JointGroupPositionController to the controller spawner but I'm having trouble checking which controllers are loaded and running because "controller_manager list" doesn't return anything, just hangs. I found a comment you made on another issue reguarding AsyncSpinner(1). There was one instance of this in the RSI kuka_hardware_interface_node which could be the issue there. Should I change that to 0 or 2?

I can see that kuka_hardware_interface.h exposes PositionJointInterface so I shouldn't have to worry about that. I've added JointGroupPositionController to robot_interface_streaming.launch.

`
<arg name="js_ctrlr" value="joint_state_controller" />
<arg name="pos_ctrlr_name" value="position_trajectory_controller" />
<arg name="joint_pos_ctrlr_name" value="joint_group_position_controller" />
`
as well as an entry for:

`    $(arg joint_pos_ctrlr_name):
      type: "position_controllers/JointGroupPositionController"`

with joints and pid gains. That's also in the hardware_controllers.yaml file.
joint_group_position_controller is in my controller_spawner.launch. Am I missing anything regarding loading the JointGroupPositionController? 

I take it jog_arm will publish to the PositionJointInterface so long as they're both started?

m-talha, 2019-08-08:
Is there an update on this broken tutorial link? Any documentation is better than no documentation when trying to get started

AndyZe, 2019-08-08:
It's gonna be awhile (1-2 weeks, I guess) until that tutorial is ready because the location of jog_arm is being shifted right now. Approving these PR's takes some time. In the meantime, you can look at the README here (https://github.com/ros-planning/moveit/blob/93803955fbbc34b328fb0045f35c4ae4a78c8e34/moveit_experimental/jog_arm/README.md)

or look at this start on the jogging tutorial (https://github.com/ros-planning/moveit_tutorials/pull/286)

rhaschke, 2019-09-11:
The tutorial is merged now (https://github.com/ros-planning/moveit_tutorials/pull/286).

closed 2019-09-11 by rhaschke