moveit/moveit #1481: Race condition causes trajectory_execution_manager

henryponti, 2019-05-24:
### Description
When a trajectory is preempted, trajectory_execution_manager sometimes crashes with a boost mutex error. 

### Environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Moveit binary build
* ros-industrial/universal_robot source build

### Steps to reproduce
The script below should cause the crash within a few minutes.

<script src="https://gist.github.com/henryponti/21d14e59804d294fc3617e21ccdbd91e.js"></script>

### Expected behaviour
move_group should allow the trajectory to be preempted without crashing.

### Actual behaviour
move_group crashes.

### Backtrace

<script src="https://gist.github.com/henryponti/5ed3bafd6f66fb4aedbadea2df82f336.js"></script>

BryceStevenWilley, 2019-05-25:
Ran this with debug [symbols](https://gist.github.com/BryceStevenWilley/447ddf79a92433fd434c9bea23dd6072), seems that the `execution_thread_` that the Trajectory manager is holding gets cleared at some point, cause the segfault.  Crashing line [here](https://github.com/ros-planning/moveit/blob/d2439417db01448e6ab9aada45f295f37608327c/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1173).

henryponti, 2019-05-27:
Thanks for the additional info @BryceStevenWilley. I tried to check if the thread is joinable, but that didn't help:

https://gist.github.com/henryponti/cbef093c1c00f635bdb0f5a2c9204c56

BryceStevenWilley, 2019-05-28:
From what I could tell (at least with how it was manifesting on my machine), the issue isn't whether or not the thread is joinable or not, it's the fact that the thread is null. Calling any function on a null will cause the segfault. I think that `execution_thread_.reset();` is being called in another thread somehow, likely from some buggy way of  handling the mutex/boolean `execution_complete_` variable.

henryponti, 2019-05-28:
I agree. I tried to check if the pointer is null as well but it didn't help
either. It looks like some deeper digging will be necessary. I am not
terribly familiar with the architecture though, so I'm not sure I'll be
able to get to the bottom of the problem without some extra help.

wumin199, 2019-06-04:
is there any updates

rhaschke, 2019-06-04:
@BryceStevenWilley, you are exactly right: Essentially we have two non-synchronized threads competing for access to `execution_thread_`.
1. The `ActionServer` thread, calling `stopExecution()` before executing a new path.
```
#11 0x00007ffff4cb5adc in trajectory_execution_manager::TrajectoryExecutionManager::stopExecution(bool) () from /home/henry/catkin_ws/devel/lib/libmoveit_trajectory_execution_manager.so.1.0.1
#12 0x00007ffff4cb63b7 in trajectory_execution_manager::TrajectoryExecutionManager::waitForExecution() () from /home/henry/catkin_ws/devel/lib/libmoveit_trajectory_execution_manager.so.1.0.1
#13 0x00007fffc81c6760 in move_group::MoveGroupExecuteTrajectoryAction::executePath(boost::shared_ptr<moveit_msgs::ExecuteTrajectoryGoal_<std::allocator<void> > const> const&, moveit_msgs::ExecuteTrajectoryResult_<std::allocator<void> >&) () from /home/henry/catkin_ws/devel/lib/libmoveit_move_group_default_capabilities.so
#14 0x00007fffc81c6d27 in move_group::MoveGroupExecuteTrajectoryAction::executePathCallback(boost::shared_ptr<moveit_msgs::ExecuteTrajectoryGoal_<std::allocator<void> > const> const&) ()
   from /home/henry/catkin_ws/devel/lib/libmoveit_move_group_default_capabilities.so
#15 0x00007fffc81d7aec in actionlib::SimpleActionServer<moveit_msgs::ExecuteTrajectoryAction_<std::allocator<void> > >::executeLoop() ()
   from /home/henry/catkin_ws/devel/lib/libmoveit_move_group_default_capabilities.so
```
2. The ROS event loop, calling `stopExecution()` when receiving `stop` on the `trajectory_execution_event` topic:
```
#0  0x00007ffff4ce2b20 in trajectory_execution_manager::TrajectoryExecutionManager::stopExecution(bool)@plt ()
   from /vol/sandbox/ros/devel/lib/libmoveit_trajectory_execution_manager.so.1.0.1
#1  0x00007ffff4cee553 in trajectory_execution_manager::TrajectoryExecutionManager::processEvent (this=0x555556092340, event="stop")
    at /vol/sandbox/ros/src/moveit/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp:225
#2  0x00007ffff4ceea29 in trajectory_execution_manager::TrajectoryExecutionManager::receiveEvent (this=0x555556092340, event=...)
    at /vol/sandbox/ros/src/moveit/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp:233
```
3. Finally there is the `ActionServer` thread, calling `stopExecution()` when preempting:
```
#4  0x00007ffff4cf8cb1 in trajectory_execution_manager::TrajectoryExecutionManager::stopExecution (this=0x5555560921a0, auto_clear=true)
    at /vol/sandbox/ros/src/moveit/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp:1173
#5  0x00007fffb8502179 in move_group::MoveGroupExecuteTrajectoryAction::preemptExecuteTrajectoryCallback (this=0x5555560e4510)
    at /vol/sandbox/ros/src/moveit/moveit_ros/move_group/src/default_capabilities/execute_trajectory_action_capability.cpp:128

```

This raises several questions:
1. Why do we need two different mechanisms to interrupt a trajectory?
Probably, preempting through the ActionServer is the cleaner way. But, having a direct topic allows for an emergency stop from other nodes as well...
2. Why they are both used from the MoveGroupInterface?
3. How can we `mutex`-protect the access, particularly which variables, apart from `execution_thread_`, need to be protected?

henryponti, 2019-06-04:
I haven't had a chance to look into that over the past few days. I am
hoping to take another look early next week.