moveit/moveit #1487: Speedup Clang Tidy By Running On Changed Files Only

davetcoleman, 2019-05-31:
@rhaschke has done a great job of having clang-tidy run on Travis: https://github.com/ros-planning/moveit_ci/pull/43

However there's been discussion that Clang-Tidy is slowing Travis down too much. @AndyZe :

> Last meeting we talked about how CI builds are taking a really long time and even timing out occasionally. This is just an observation that clang-tidy takes the majority of that time: about an hour. Everything else finishes in ~20 minutes or less. (Example)

@RoboticsYY suggested we only analyze changed code:

> I think one reason that the clang-tidy runs so long time is clang-tidy scan all the code when a PR is submitted. However, it’s reasonable to only analyze the changed codes. I happen to read [someone did this in Jenkins](https://pspdfkit.com/blog/2018/using-clang-tidy-and-integrating-it-in-jenkins/), I tested it on my local computer, seems OK. I haven’t figured this out in Travis, but maybe this is a fix to this issue.

I think this is a great idea. Anyone want to take this on?

rhaschke, 2019-05-31:
Looks easy to implement. Maybe I can have a short look later this evening.

bmagyar, 2019-06-06:
Have you considered the bot-based approach? 

It could be set up to periodically (or on-event-basis (PR merged)) run clang-tidy on the codebase and push a commit.

rhaschke, 2019-06-19:
Addressed in https://github.com/ros-planning/moveit_ci/pull/73.

davetcoleman, 2019-06-27:
Thanks @rhaschke !

closed 2019-06-27 by davetcoleman