moveit/moveit #1491: Build moveit from source error

yunzhe-tao, 2019-06-03:
### Description
Hi,
I was trying to build MoveIt! from source following the instructions: https://moveit.ros.org/install/source/. 
However, I had some errors while doing catkin build.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? Source

### Steps to reproduce
rosdep update
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install python-wstool python-catkin-tools clang-format-3.9
mkdir ~/ws_moveit
cd ~/ws_moveit
source /opt/ros/kinetic/setup.bash
wstool init src
wstool merge -t src https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall
wstool update -t src
rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}
catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build

### Expected behaviour
Build success.

### Actual behaviour
Most packages were abandoned.

### Backtrace or Console output

-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/kinetic
Workspace:                   /home/yuntze/ws_moveit
-----------------------------------------------------------
Build Space:        [exists] /home/yuntze/ws_moveit/build
Devel Space:        [exists] /home/yuntze/ws_moveit/devel
Install Space:      [unused] /home/yuntze/ws_moveit/install
Log Space:         [missing] /home/yuntze/ws_moveit/logs
Source Space:       [exists] /home/yuntze/ws_moveit/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------
[build] Found '31' packages in 0.0 seconds.                                                             
[build] Updating package table.                                                                         
Starting  >>> catkin_tools_prebuild                                                                     
Finished  <<< catkin_tools_prebuild                           [ 1.4 seconds ]                           
Starting  >>> geometric_shapes                                                                          
Starting  >>> moveit_msgs                                                                               
Starting  >>> rviz_visual_tools                                                                         
Finished  <<< moveit_msgs                                     [ 3.8 seconds ]                           
Finished  <<< geometric_shapes                                [ 7.3 seconds ]                           
[... 33 lines cut ...]
Abandoned <<< moveit_ros_planning                             [ Unrelated job failed ]                  
Abandoned <<< moveit_fake_controller_manager                  [ Unrelated job failed ]                  
Abandoned <<< moveit_kinematics                               [ Unrelated job failed ]                  
Abandoned <<< moveit_planners_ompl                            [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_move_group                           [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_manipulation                         [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_robot_interaction                    [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_warehouse                            [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_benchmarks                           [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_planning_interface                   [ Unrelated job failed ]                  
Abandoned <<< moveit_experimental                             [ Unrelated job failed ]                  
Abandoned <<< moveit_planners_chomp                           [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_visualization                        [ Unrelated job failed ]                  
Abandoned <<< moveit_setup_assistant                          [ Unrelated job failed ]                  
Abandoned <<< moveit_simple_controller_manager                [ Unrelated job failed ]                  
Abandoned <<< moveit_ros_control_interface                    [ Unrelated job failed ]                  
Abandoned <<< moveit_visual_tools                             [ Unrelated job failed ]                  
Abandoned <<< moveit_tutorials                                [ Unrelated job failed ]                  
[build] Summary: 4 of 27 packages succeeded.                                                            
[build]   Ignored:   5 packages were skipped or are blacklisted.                                        
[build]   Warnings:  None.                                                                              
[build]   Abandoned: 22 packages were abandoned.                                                        
[build]   Failed:    1 packages failed.                                                                 
[build] Runtime: 51.6 seconds total.                                                                    
[build] Note: Workspace packages have changed, please re-source setup files to use them.

rhaschke, 2019-06-03:
Do you have a non-default octree installation?
On Travis, the master branch [compiles just fine for ROS Kinetic](https://travis-ci.org/ros-planning/moveit/builds/540341299).

yunzhe-tao, 2019-06-03:
I am not sure about that. But I have FCL 0.6.0 installed in my system. Does that affect the installation here? Thanks.

rhaschke, 2019-06-03:
This might be an issue. ROS Melodic / Ubuntu Bionic comes with FCL 0.5 by default. 
Support for FCL 0.6 was added in #1156, but maybe there are still open issues. 
@2scholz / @v4hn, can you comment on this?

BryceStevenWilley, 2019-06-04:
This depends on how you installed FCL 0.6. If you didn't have Octomap installed/if FCL couldn't find octomap when installing then the `OcTreed` type [won't be declared](https://github.com/flexible-collision-library/fcl/blob/9dba579158109c0164bfe0e8b4a75c16bfc788f6/include/fcl/geometry/octree/octree.h#L43). 

A possibly related issue: https://github.com/flexible-collision-library/fcl/issues/210

yunzhe-tao, 2019-06-04:
I (re-)installed Octomap and FCL again, which were successful. Also, I did make test on FCL installation. The tests with Octomap were all passed. However, I had the same issue when I rebuilt the MoveIt. Any further comments? Thanks.

BryceStevenWilley, 2019-06-12:
The only thing I can think is to dive into the installed FCL header files to
* make sure `OcTreed` is declared
* Checkout the FCL version number in `config.h`, and see what the `FCL_VERSION_CHECK` ([here](https://github.com/ros-planning/moveit/blob/898cdc8d8df5689319cd462c2c674954c6134604/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/fcl_compat.h#L42)) is being evaluated to.

Worst case you can dive into the build folder for `moveit_core` and run `make -B VERBOSE=1`, which should list all of the include files being passed into the build, and you should be able to find your install of FCL if that is correct.

yunzhe-tao, 2019-06-12:
Hi @BryceStevenWilley @rhaschke, I removed and re-installed everything (including libccd, OctoMap, fcl, and MoveIt), then everything worked just fine. Thank you guys so much for your help! Sometimes startover is the best way to debug :)

BryceStevenWilley, 2019-06-12:
Thanks for the update yunzhe-tao!

closed 2019-06-12 by BryceStevenWilley