moveit/moveit #1493: Possible Threading error during trajectory execution

LeroyR, 2019-06-07:
### Description

During Execution of one Trajectory part, my controller handle is sometime still reporting status running after the the `handle->waitForExecution();` Already returned.

From looking at the code, waitForExecution should only return after getting the result and setting the status. But i am getting the following logging output:

```
Stopped trajectory monitor
Completed trajectory execution with status RUNNING ...
Controller torso_controller is done with state SUCCEEDED:
Controller torso_controller successfully finished
Controller gripper_controller is done with state SUCCEEDED:
Controller handle gripper_controller reports status RUNNING
Controller gripper_controller successfully finished
Controller arm_controller is done with state SUCCEEDED:
Controller arm_controller successfully finished
```

My bags only record one result message per goal (success) so i am not really sure what is causing the state to be set to RUNNING.

```
topics:      /torso_controller/follow_joint_trajectory/feedback    74 msgs    : control_msgs/FollowJointTrajectoryActionFeedback
             /torso_controller/follow_joint_trajectory/goal         3 msgs    : control_msgs/FollowJointTrajectoryActionGoal    
             /torso_controller/follow_joint_trajectory/result       3 msgs    : control_msgs/FollowJointTrajectoryActionResult  
             /torso_controller/follow_joint_trajectory/status     303 msgs    : actionlib_msgs/GoalStatusArray
```

This "_fix_" lets me work around this completing the whole trajectory https://github.com/LeroyR/moveit/commit/8a99875fab981a73459ababe91911f17ee246d5b 

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build, Master

jschleicher, 2019-11-06:
I get this same issue on melodic and move_group returns `FAILURE` in some of these cases after approx 15 minutes (earlier on high-load): 
[Dockerfile gist](https://gist.github.com/jschleicher/20a9e6dd58e1fb45becc4cad1e1e0e8c)

```
[ WARN] [1573030232.349341327]: Controller handle manipulator_joint_trajectory_controller/ reports status RUNNING
[ INFO] [1573030232.349555675]: Completed trajectory execution with status RUNNING ...
[ INFO] [1573030232.349627577]: Controller manipulator_joint_trajectory_controller/ successfully finished
```

I'll check, if the problem also persists on current master...

rhaschke, 2019-11-06:
No need to verify on master. It's still broken there as well - unfortunately. Any help to fix the issue is highly welcome. We (the maintainers) didn't have time to look into this race condition in detail.

jschleicher, 2019-11-09:
Enabling `ActionBasedController` DEBUG output shows the race condition in its full beauty:

https://github.com/ros-planning/moveit/blob/985e34c2f6e9a91f4fe2bec9168eca366454e238/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1492-L1495

```
[ INFO] [0.7198]: Controller prbt/manipulator_joint_trajectory_controller/ successfully finished
[DEBUG] [0.7199]: Controller prbt/manipulator_joint_trajectory_controller/ is done with state SUCCEEDED: 
[ WARN] [0.7200]: Controller handle prbt/manipulator_joint_trajectory_controller/ reports status SUCCEEDED
[ INFO] [1.0206]: Completed trajectory execution with status SUCCEEDED ...
```

My explanation: `waitForResult` and thus `waitForExecution` returns, the check in line 1492 still returns `RUNNING`, then the done-callback jumps in with SUCCEEDED and in the next line `getLastExecutionStatus()` also prints succeeded, which was false a line before.

I see two possible fixes: 
1. Wait for the done-Callback (probably checking done_ in `waitForExecution`:
https://github.com/ros-planning/moveit/blob/d3b2db31dba1b7f292e0dfb2b92f656e6cc2e555/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/action_based_controller_handle.h#L128

2. get rid of `last_exec_` and query the current status from the actionclient directly in `getLastExecutionStatus`

3. hypothetically: change upstream behaviour in actionlib: `waitForResult()` should return _after_ the doneCb returned?!

What do you think?

v4hn, 2019-11-10:
thank you for debugging this!
I feel it's a design error in actionlib and it would be nice to see a pr there.
it's somewhat unlikely to get merged though, so i also support your first proposed solution. we could even get rid of waitForResult altogether and report on a signal from the done callback.

i somewhat dislike your second proposal.
this would allow getLastExecutionStatus to return ACTIVE.

jschleicher, 2019-11-20:
Upstream PR is in  ros/actionlib#156.

rhaschke, 2019-11-27:
Fixed via #1783.

simonschmeisser, 2021-05-19:
did this regress or is it something new?

```
[ WARN] [1621415987.180401613]: Controller handle  reports status RUNNING
[ INFO] [1621415987.180552492]: Controller  successfully finished
[ INFO] [1621415987.180730883]: Completed trajectory execution with status RUNNING ...
[ERROR] [1621415987.180921196]: Error executing plan:  "RUNNING"
```
on master

v4hn, 2021-06-26:
#1783 was supposed to resolve this and @rhaschke merely forgot to close this (unlinked) issue.

If you see the same issue again, this is most annoying. You will have to debug it though so we can fix it upstream (wherever that might be...)