moveit/moveit #1497: Academic research on vulnerable c++ code snippet

JafarAkhondali, 2019-06-09:
Dear ROS Planning, 
We are a group of Academic researchers. We are analyzing vulnerable C++ code snippets migrated from StackOverflow to GitHub. Our research will be published in Academic publications and will not be used in any Industrial application. 
We noted a vulnerable code snippet in your repository that was most likely copied from Stack Overflow. The vulnerability exists in this source code [file](https://raw.githubusercontent.com/ros-planning/moveit/kinetic-devel/moveit_setup_assistant/src/widgets/setup_assistant_widget.cpp) of your repository.

Please verify our report here with regards to the above vulnerability to assist you.
**[Link to report](https://docs.google.com/forms/d/e/1FAIpQLScyzcHYnBBzXxGmAJV09-qyyOJws7aMGJFmvsJUFX6xvMGvOA/viewform?entry.793621839=2266)** with four questions for you related to the vulnerability (should not take more than 5 minutes to answer).

Here is a summary of the vulnerable code snippet:

Description:
===
Reading variable from ENV is not safe at all, because it can be bypassed easily by setting new value for `PWD` in shell ( like bash ).
**example:**
Suppose current directory is : `/tmp` , now run the program:
```
$ ./prog
Working directory : /tmp
```
Let's change `PWD` value in the environment:
```
$ export PWD=/anywhere
```
Run the program again:
```
$ ./prog
Working directory : /anywhere
```  
 
Mitigation:
===
Use this [answer](https://stackoverflow.com/a/7403075/3686236):
`QDir::currentPath()`
        
References:
===
[https://stackoverflow.com/a/7403075/3686236](https://stackoverflow.com/a/7403075/3686236)
        
        
---
Please verify our report here with regards to the above vulnerability to assist you.
**[Link to report](https://docs.google.com/forms/d/e/1FAIpQLScyzcHYnBBzXxGmAJV09-qyyOJws7aMGJFmvsJUFX6xvMGvOA/viewform?entry.793621839=2266)** with four questions for you related to the vulnerability (should not take more than 5 minutes to answer).

Sincerely yours,
Morteza Verdi, Shiraz university, E-mail: m.verdi@shirazu.ac.ir
Jafar Akhondali, Shiraz university, E-mail: jafar.akhondali@yahoo.com
Ashkan Sami, Shiraz university, E-mail: ashkan.sami@gmail.com
Foutse Khomh, Polytechnique Montreal, E-mail: foutse.khomh@polymtl.ca, website: http://www.khomh.net/
Gias Uddin,  Polytechnique Montreal, E-mail: gias98@gmail.com, website: https://giasuddin.github.io
Alireza Karami motlagh, Shahid Chamran University, E-mail: alireza.karami.m@gmail.com

gavanderhoorn, 2019-06-09:
https://github.com/ros-planning/moveit/blob/898cdc8d8df5689319cd462c2c674954c6134604/moveit_setup_assistant/src/widgets/setup_assistant_widget.cpp#L120

JafarAkhondali, 2019-06-09:
@gavanderhoorn Yes, since `getenv` will read data from shell environment, users can modify `pwd` variable before execution and set a crafted `PWD`. This does not mean that your code is necessarily vulnerable, because our study is on code snippets in Stack Overflow not on the whole projects of Github. The vulnerability might be not exploitable (easily) due to different conditions and flows of programs, but it's a good practice to make the program safe as much as possible.

JafarAkhondali, 2019-06-18:
Hi,
            Looks like you didn't complete the survey, can you finish it up?
            It would take less than 5 minutes.
            Thanks again.
            [Link to report](https://docs.google.com/forms/d/e/1FAIpQLScyzcHYnBBzXxGmAJV09-qyyOJws7aMGJFmvsJUFX6xvMGvOA/viewform?entry.793621839=2266)

davetcoleman, 2019-07-18:
Pretty cool tool, thanks @JafarAkhondali. Looking at ``git blame``, this was introduced by @130s here: https://github.com/ros-planning/moveit/pull/509

@130s would you mind creating a quick patch for this please?

davetcoleman, 2019-08-15:
Addressed by @JafarAbdi

closed 2019-08-15 by davetcoleman