moveit/moveit #1509: Setting goal_constraints from the python interface

Rayman, 2019-06-20:
(I'm not sure if this is the right place to ask) According to the moveit_msgs it is possible to specify goal constraints. I not able to set them from moveit_commander. Can this be done from python? 

I would like to contribute a PR if I can get a few pointers where to start.

felixvd, 2019-06-21:
It sounds like what you are looking for is `set_joint_value_target` and `set_pose_target`. I don't know if `set_position_target` really works, but I am curious.

If you want to add anything, you probably want to do it in [here](https://github.com/ros-planning/moveit/blob/master/moveit_commander/src/moveit_commander/move_group.py) (the Python file) and [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/wrap_python_move_group.cpp) (the C++ functions that the Python ones bind to).

Rayman, 2019-06-21:
Thanks you for the quick answer. I've clarified my question to hopefully make it more clear.

felixvd, 2019-06-21:
> I want to restrict the configuration of the robot, while the end effector is in a certain position.

I am not sure what you mean by this. If you just want to move to a pose you can use `set_pose_target`, and the orientation at the goal will be defined by the pose. Do you want to constrain the orientation of the end effector while the robot is moving? Then you could use `set_path_constraints`.

Rayman, 2019-06-21:
My robot arm has 6 dof, but the arm can be in 4 different configurations (similar to the UR10). The shoulder joint can be to the left or right, and the wrist can be in 2 different directions.

So the end effector is in the same position, but the robot is in a different configuration.

Rayman, 2019-06-21:
[image]
[image]

felixvd, 2019-06-21:
I understand. That's a general property of 6 DOF arms of this type.

I think that "don't do funky unnecessary reorientations" is an expectation that almost all users have when they send a move command, but I don't know what's the systematic way to deal with this out of the box. At the very least I am pretty sure that you cannot set both a joint and pose constraint at the same time. Maybe someone else can comment.

I know that a common workaround is to plan to joint positions and then do linear motions in a small area around those. Another one is to restrict the joint values in the SRDF, so that only one configuration is permissible. 

Another way to deal with it is to fiddle with the IK solvers, e.g. by retrieving all 4 solutions for a given pose and selecting the one you want, but I have not looked into this. I am pretty sure this has been done though, so you would probably find something on Google. Feel free to add your results here if you search for it.

Rayman, 2019-06-24:
I think I found a solution for my problem. 

You can call the [msg/PositionIKRequest](http://docs.ros.org/api/moveit_msgs/html/msg/PositionIKRequest.html) service directly with JointConstraints. I'll first calculate the IK, and then move those joint values. It's not available from the moveit_commander interface, but this is good enough for me.

closed 2019-06-24 by Rayman