moveit/moveit #1521: "import moveit_commander" fails on python3

kalyanvasudev, 2019-06-25:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Kinetic
* OS Version: 16.04
* Source or Binary build? Binary

I am trying to get moveit to work with python3. I get the following error when I do 
"import moveit_commander" in a python3 console. 

Can some please help me on how to fix it?

Thank!

[image]

gavanderhoorn, 2019-06-26:
The `moveit_commander` is a hybrid C++/Python module. If you've installed the Python 2.7 version (using `apt` fi) you cannot just use it in Python 3 (by changing your `PYTHONPATH` fi).

If you want to do this, you'll have to compile it from source.

closed 2019-06-26 by kalyanvasudev

huangjiancong1, 2020-06-12:
> The `moveit_commander` is a hybrid C++/Python module. If you've installed the Python 2.7 version (using `apt` fi) you cannot just use it in Python 3 (by changing your `PYTHONPATH` fi).
> 
> If you want to do this, you'll have to compile it from source.
The best way to compile from source:
https://moveit.ros.org/install/source/

YeeCY, 2020-07-08:
Any update? I still got the same error above after compiling the moveit from scratch like tf2_py.

https://github.com/ros/geometry2/issues/259
https://answers.ros.org/question/326226/importerror-dynamic-module-does-not-define-module-export-function-pyinit__tf2/

rhaschke, 2020-07-17:
When compiling from source, you need to ensure that the whole ROS workspace is compiled against python3 as well.
This is not at all a trivial task and not recommended for non-experts. Only ROS Noetic (for Ubuntu 20.04) supports python3.
For older releases you should stick with python2. In Noetic, moveit_commander works without any issues:
```
$ python
Python 3.8.2 (default, Apr 27 2020, 15:53:34) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import moveit_commander
>>> 
```

seann999, 2020-08-26:
I'm trying to use moveit_commander in Python 3 with Noetic.
I'm using the noetic-source docker image [here](https://hub.docker.com/r/moveit/moveit/tags) but I'm running into an error:
`_moveit_move_group_interface.so: undefined symbol: _ZN3tf27fromMsgIN13geometry_msgs11Quaternion_ISaIvEEEN5Eigen10QuaternionIdLi0EEEEEvRKT_RT0_`
[This issue](https://github.com/ros-planning/moveit/issues/1785) seems relevant but I'm not sure what exactly I'm supposed to do.
Any hints would be appreciated.

rhaschke, 2020-08-30:
Hi @seann999, indeed #1785 is related to your issue. While #1785 was fixed in the upstream geometry2 package for Melodic a while ago, a corresponding fix was pending for Noetic (https://github.com/ros/geometry2/pull/471) and only recently released (https://github.com/ros/rosdistro/pull/26309).
To work around those issues, we manually have setup a Noetic docker image for MoveIt comprising the corresponding fixes.
After the new tf2 release those patches were overridden.
Long story short: I recreated the MoveIt docker images and it should work now.

rhaschke, 2023-09-03:
It is not sufficient to link moveit_commander into your conda environment. Ideally, you build MoveIt for your conda env to have all python dependencies correctly resolved.