moveit/moveit #1522: Duration is out of dual 32-bit range

sshazly, 2019-06-26:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Kinetic and Melodic]
* OS Version: [Ubuntu 16.04 Ubuntu 18.04]
* Source or Binary build? [Source]

* If source, which git commit or tag?
kinetic-devel (unkown git commit, .git files deleted by predeccesor)
melodic-devel
4df31882dd67da4ee5169f28d53b29d8c47c4821

### Steps to reproduce
Unfortunately I work in a government facility and cannot provide the code, but I have seen it repeatedly fail under the same step in the process, under the same conditions.

When you perform an async execute (in my case it was using a moveToJointPosition method for the base pan joint of a UR3) and you do not explicitly stop the motion before it reaches the goal position, the move_group node will crash and cannot respawn.
----------------------------------------------------------------------------------------------------------------
[ERROR] [1561565513.125394532, 1020.648000000]: Planning pipeline threw an exception: Duration is out of dual 32-bit range
terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range
[ur3_0/move_group-1] process has died [pid 23974, exit code -6, cmd /home/lab/catkin_ws/devel/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/lab/.ros/log/a3383dd2-982c-11e9-8a3d-80fa5b51c525/ur3_0-move_group-1.log].
log file: /home/lab/.ros/log/a3383dd2-982c-11e9-8a3d-80fa5b51c525/ur3_0-move_group-1*.log

(log file included below)
-------------------------------------------------------------------------------------------------------------

So in the demo it searches for an aruco marker while moving, and if it finds the aruco marker and execute a stop command everything will continue without error. But if it does not find the aruco marker and it reaches the goal position the move_group will crash with the error above

I realize the more traditional way is to implement an action server for this, but I was pressed for time and this seemed to be the quickest way.

Either way it seems like a weak point that needs to be fixed. This error is reproduced across kinetic 16.04 and melodic 18.04 and with multiple computers (most system76s but some velocity computers as well)
I know this won't mean much to you guys since you don't have my manipulator class that I create but this is essentially what happens:
-----------------------------------------------------------------
bool is_found = false;
while(!is_found){
            ur.SetAsyncMode(true);
            ur.MoveToJointPosition(MP.manipulator_prefix + "shoulder_pan_joint",-2.25);
            ur.SetAsyncMode(false);
            
           //Search for aruco marker
           if(is_found){
                 ur.Stop();
           }
          // Do other stuff
};
-----------------------------------------------------------------

### Expected behaviour
The robot should reach its goal position and the demo node should continue normally.

### Actual behaviour
The move_group node crashes and the demo node cannot continue since all the services crash with it.

### Backtrace or Console output
[ur3_0-move_group-1-stdout.log](https://github.com/ros-planning/moveit/files/3331075/ur3_0-move_group-1-stdout.log)

rhaschke, 2019-06-27:
Please have a look here: https://github.com/ros-planning/moveit/issues/1062.

sshazly, 2019-06-27:
Mr Haschke,

I saw that issue, and I don't believe it applies to this since I have re-installed Ubuntu 16.04 and ROS-kinetic over 6 times in the past 6 months on multiple computers and the problem still persists.

In addition I have freshly installed 18.04 and ROS-melodic 2 days ago and I have replicated this issue with both the binary and source installs.

Thanks,

Sherif Shazly

rhaschke, 2019-06-27:
In this case you should provide a minimal source code example to replicate your issue, e.g. with the panda robot used in the MoveIt tutorials. Otherwise we cannot really help.

sshazly, 2020-04-06:
My solution for this was to bypass the planning_interface::MoveGroup and use the execute action server directly.
When I need to do asynchronous movements I just send a request without using the SimpleActionClient::waitForResult  method.

closed 2020-04-06 by sshazly