moveit/moveit #1524: Moveit building from source fails

kalyanvasudev, 2019-06-26:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? Ans: Source 
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Followed instructions here to download source code - 
https://moveit.ros.org/install/source/
then ran "catkin build" in the catkin work space "ws_moveit"

After doing so, the build fails with the following error.
[image]

Can someone please help me on how to fix this?

rhaschke, 2019-06-27:
Looks like you are missing the package `python-numpy`. Did you run the `rosdep install` command to install all required dependencies? This [works on Travis...](https://travis-ci.org/ros-planning/moveit/builds/548781867)

closed 2019-06-27 by rhaschke

knorth55, 2019-07-05:
I have exact same issue on my kinetic pc.
I'm looking into this issue.

knorth55, 2019-07-05:
This issue occurs when you install upper version `numpy` via `pip` (i.e. `sudo pip install numpy -U`)
I installed `1.16.3` numpy, so I uninstall it (`sudo pip uninstall numpy`) and reinstall debian numpy `sudo apt install python-numpy --reinstall`.
Now I can correctly build `moveit_ros_planning_interface`.